espressif/arduino-esp32 #8623: Cannot set and cannot get i2c frequency

tomy983, 2023-09-11:
### Board

ESP32-S3-WROOM-1-N16R8CT-ND

### Device Description

It's a custom PCB with the ESP32-S3-WROOM-1-N16R8CT-ND module and an i2c sensor

### Hardware Configuration

GPIO 1 and 13 are SDA and SCL

### Version

v2.0.11

### IDE Name

VScode PlatformIO 

### Operating System

Win11

### Flash frequency

80000000

### PSRAM enabled

yes

### Upload speed

921600

### Description

Cannot set and cannot get i2c frequency using Wire.h

Returns: [Wire.cpp:381] setClock(): could not acquire lock

### Sketch

```cpp
#include <Arduino.h>
#include <Wire.h>

uint32_t frequency = 400000;

void setup() {
    Serial.begin();   
    log_i("Setting i2c freq at %uHz", frequency);
    Wire.setClock(frequency);
    frequency = Wire.getClock();
    log_i("I2c freq is set at: %uHz", frequency);
}

void loop() {}
```

### Debug Message

```plain
Serial Output:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201fc2e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[    95][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   109][I][main.cpp:8] setup(): Setting i2c freq at 400000Hz
[   109][E][Wire.cpp:381] setClock(): could not acquire lock
[   110][E][Wire.cpp:359] getClock(): could not acquire lock
[   114][I][main.cpp:13] setup(): I2c freq is set at: 0Hz
```

### Other Steps to Reproduce

Build info and board json file attached
[esp32-s3-devkitc-1-n16r8.json.txt](https://github.com/espressif/arduino-esp32/files/12577195/esp32-s3-devkitc-1-n16r8.json.txt)
[build.txt](https://github.com/espressif/arduino-esp32/files/12577198/build.txt)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-09-11:
Did you forget to call `Wire.setPins(SDA, SCL);` and `Wire.begin();`?

me-no-dev, 2023-09-11:
```cpp
#include <Arduino.h>
#include <Wire.h>

uint32_t frequency = 400000;

void setup() {
    Serial.begin();
    delay(1000);
    Wire.setPins(1, 13);
    Wire.begin();
    log_i("Setting i2c freq at %uHz", frequency);
    Wire.setClock(frequency);
    frequency = Wire.getClock();
    log_i("I2c freq is set at: %uHz", frequency);
}

void loop() {}
```

```
[  1096][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=1 scl=13 freq=100000
[  1097][I][sketch_sep11a.ino:11] setup(): Setting i2c freq at 400000Hz
[  1100][I][sketch_sep11a.ino:14] setup(): I2c freq is set at: 400000Hz
```

tomy983, 2023-09-11:
Unfortunately still does not work for me... Also tried on a freshly installed system...
```#include <Arduino.h>
#include <Wire.h>

uint32_t frequency = 400000;

#define BNO086_PS0 42
#define BNO086_PS1 41
#define BNO086_RST 12
#define BNO086_ADD 0x4A

#define BNO086_SDA 1
#define BNO086_SCL 13

void setup() {
    Serial.begin();
    pinMode(BNO086_PS1, OUTPUT);
    pinMode(BNO086_PS0, OUTPUT);
    pinMode(BNO086_RST, OUTPUT);
    log_i("Setting BNO086 to i2c output");
    digitalWrite(BNO086_PS1, LOW);
    digitalWrite(BNO086_PS0, LOW);
    log_i("BNO086 hardware reset");
    digitalWrite(BNO086_RST, HIGH);
    delay(10);
    digitalWrite(BNO086_RST, LOW);
    delay(10);
    digitalWrite(BNO086_RST, HIGH);
    delay(500);

    log_i("Setting i2c pins (SDA: %d SCL: %d)", BNO086_SDA, BNO086_SCL);
    Wire.setPins(BNO086_SDA, BNO086_SCL);

    log_i("Wire begin");
    Wire.begin();
     
    Wire.setClock(frequency);
    log_i("Setting i2c freq at %uHz", frequency);
    frequency = Wire1.getClock();
    log_i("I2c freq is set at: %uHz", frequency);

    log_i("Check i2c device @ 0x%x", BNO086_ADD);
    Wire.beginTransmission(BNO086_ADD);
    if (Wire.endTransmission() == 0)
        log_i("i2c device @ 0x%x OK", BNO086_ADD);
    else
        log_i("i2c device @ 0x%x not found", BNO086_ADD);
}

void loop() {}
```

```
[    99][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   114][I][main.cpp:19] setup(): Setting BNO086 to i2c output
[   114][I][main.cpp:22] setup(): BNO086 hardware reset
[   634][I][main.cpp:30] setup(): Setting i2c pins (SDA: 1 SCL: 13)
[   634][I][main.cpp:33] setup(): Wire begin
[   634][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=1 scl=13 freq=100000
[   641][I][main.cpp:37] setup(): Setting i2c freq at 400000Hz
[   646][E][Wire.cpp:359] getClock(): could not acquire lock
[   652][I][main.cpp:39] setup(): I2c freq is set at: 0Hz
[   657][I][main.cpp:41] setup(): Check i2c device @ 0x4a
[   662][I][main.cpp:44] setup(): i2c device @ 0x4a OK
```

tomy983, 2023-09-11:
> Did you forget to call `Wire.setPins(SDA, SCL);` and `Wire.begin();`?

No, I've just put here a short version showing the issue.

me-no-dev, 2023-09-11:
`Wire1.getClock()`? Wrong Wire. And you did forget to start Wire the first time or you would have gotten a message in Serial ;)

tomy983, 2023-09-11:
Yes, I did not started it. I thought I could set the frequency before starting it. Made sense to me. 
Wire1 is a leftover of test on both and got me in the debug.
Anyway, in esp32-hal-i2c.c (I noticed sometimes the error comes from there as: `[E][esp32-hal-i2c.c:255] i2cSetClock(): could not acquire lock` )I see that debug messages are present for both `could not acquire lock` and `bus is not initialized`, the second being the actual reason in this case 🤦‍♂️ but not what is the log actually saying... Maybe inverting the checks could avoid another poor guy all this :-)
Aaaand probably just time to have a walk outside... it works. My bad. Thank you