espressif/arduino-esp32 #8625: SD size() returns random value until file has non-zero bytes of content

frankcohen, 2023-09-12:
### Board

ESP32-S3 Dev Module

### Device Description

I use my Reflections board, it is an open-source Arduino compatible ESP32-S3 board built from the ESP32-S3 Dev Module. Details at https://github.com/frankcohen/ReflectionsOS

### Hardware Configuration

// SPI Bus
#define SPI_MOSI      35
#define SPI_MISO      37
#define SPI_SCK       36

// NAND Storage
#define NAND_SPI_CS   15

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.1.0 nightly

### Operating System

MacOS 13.5.1

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm using an external SD/NAND with an ESP32-S3 to create and save log values to a file. NAND is a surface mount version of an SD card. I connect the NAND over SPI using pin 15 for the CS. This is part of my Reflections open-source project https://github.com/frankcohen/ReflectionsOS.

File mylog = SD.open( "/log1", FILE_WRITE );
if( !mylog )
{
Serial.print( F( "Unable to create log file " ) );
}
Serial.print( "mylog.size() = " );
Serial.println( (long) mylog.size() );

I see random size values in the Serial monitor. For example mylog.size = 1070537564. When I write the the file first, then the size values return correctly.

File mylog = SD.open( "/log1", FILE_WRITE );
if( !mylog )
{
Serial.print( F( "Unable to create log file " ) );
}
int logval2 = mylog.print( "Starting log" );
Serial.print( "mylog.size() = " );
Serial.println( (long) mylog.size() );

size returns 14, which is correct.

-Frank

### Sketch
[... 341 lines cut ...]
      uploadcount++;
    }
    else
    {
      Serial.print( "Uploading complete, log sent to service " );
      Serial.println( uploadcount );

      myupload.close();
      myuploadopen = 0;
      SD.remove( uploadfilename );
      uploading = 0;
    }
  }
}
```

### Debug Message

```plain
Opened log file, mylog.size()=1828101982
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-09-12:
@P-R-O-C-H-Y PTAL

everslick, 2023-09-13:
Probably related: https://github.com/espressif/arduino-esp32/issues/7759

closed 2025-01-16 by Parsaabasi