espressif/arduino-esp32 #8628: Camera probe failed with error 0x20004

Abhay-2412, 2023-09-12:
### Board

AI Thinker Esp32-cam

### Device Description

just attached with eso32-cam-mb

### Hardware Configuration

OV5640 camera

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Actually I'm searching for a simple code to test my camera which can be ported to ESP-IDF later (Arduino as component) came across a code but whenever i try to access it through my ip address i get below error constantly
```

16:57:55.361 -> E (418544) i2c: i2c driver install error
16:57:55.392 -> [E][camera.c:1113] camera_probe(): Detected camera not supported.
16:57:55.439 -> [E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
```

i tried traditional esp32 cam web server code its working fine but iam unable to port it to ESP-IDF since it showing some files are missing related to `app_httpd.cpp` 

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include "esp_camera.h"

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22
// Replace with your network credentials
const char* ssid = "Abhay iphone13";
const char* password = "12345678";

// Create an instance of the web server
WebServer server(80);

// Camera configuration
const int cameraWidth = 320;
const int cameraHeight = 240;

void handleRoot() {
  camera_fb_t* frameBuffer = NULL;

  // Start the camera
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  // Initialize the camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera initialization failed with error 0x%x", err);
    return;
  }

  // Set the content type to multipart/x-mixed-replace
  server.sendHeader("Content-Type", "multipart/x-mixed-replace; boundary=ESP32CAM");

  while (true) {
    // Capture a new frame from the camera
    frameBuffer = esp_camera_fb_get();

    // Send the frame as a response
    if (frameBuffer) {
      server.sendHeader("Content-Type", "image/jpeg");
      server.sendHeader("Content-Disposition", "inline; filename=capture.jpg");
      server.sendContent_P(reinterpret_cast<const char*>(frameBuffer->buf), frameBuffer->len);
      esp_camera_fb_return(frameBuffer);
    }

    // Yield to the server for handling other requests
    server.handleClient();
  }
}

void setup() {
  // Start the serial communication
  Serial.begin(115200);

  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  // Start the web server and handle the root request
  server.on("/", handleRoot);
  server.begin();
  Serial.println("Web server started");
  Serial.println(WiFi.localIP());
}

void loop() {
  // Handle incoming client requests
  server.handleClient();
}
```

### Debug Message

```plain
16:50:57.500 -> t Jun  06 002:5

rst:0x10 (RTWD_T_EET),boot:0x13(P_AT_FLASH_BOOT)
16:50:57.500 -> cnisp 0, SPIWP:0xe
ckdr:0x00,q_drv00,_r:000,cs0_drv:00,d_r:00,pdrv:000
16:50:57.500 -> oe:IO, clock di:
16:50:57.500 -> od03fff0018,len:
la:x3fff001cln11
h 0tail 12 room4
16:50:57.500 -> oa:x4078000,len:094
16:50:57.500 -> od:0x40000,e:38
16:50:57.500 -> etry 0x4008064
16:50:59.782 -> Connecting to WiFi...
16:51:00.775 -> Connecting to WiFi...
16:51:01.756 -> Connecting to WiFi...
16:51:02.773 -> Connecting to WiFi...
16:51:03.779 -> Connecting to WiFi...
16:51:04.784 -> Connecting to WiFi...
16:51:04.784 -> Connected to WiFi
16:51:04.784 -> Web server started
16:51:04.784 -> 172.20.10.10
16:51:17.043 -> [E][camera.c:1113] camera_probe(): Detected camera not supported.
16:51:17.043 -> [E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
16:51:17.089 -> E (20286) i2c: i2c driver install error
16:51:17.175 -> [E][camera.c:1113] camera_probe(): Detected camera not supported.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-09-12:
The problem is that you are trying to init the camera every request. It should be inited once in setup.

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include "esp_camera.h"

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22
// Replace with your network credentials
const char* ssid = "Abhay iphone13";
const char* password = "12345678";

// Create an instance of the web server
WebServer server(80);

// Camera configuration
const int cameraWidth = 320;
const int cameraHeight = 240;

void handleRoot() {
  camera_fb_t* frameBuffer = NULL;

  // Capture a new frame from the camera
  frameBuffer = esp_camera_fb_get();

  // Send the frame as a response
  if (frameBuffer) {
    // Set the content type to multipart/x-mixed-replace
    server.sendHeader("Content-Type", "multipart/x-mixed-replace; boundary=ESP32CAM");

    server.sendHeader("Content-Type", "image/jpeg");
    server.sendHeader("Content-Disposition", "inline; filename=capture.jpg");
    server.sendContent_P(reinterpret_cast<const char*>(frameBuffer->buf), frameBuffer->len);
    esp_camera_fb_return(frameBuffer);
  } else {
    server.send(500);
  }
}

void setup() {
  // Start the serial communication
  Serial.begin(115200);

  // Start the camera
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  // Initialize the camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera initialization failed with error 0x%x", err);
    return;
  }

  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  // Start the web server and handle the root request
  server.on("/", handleRoot);
  server.begin();
  Serial.println("Web server started");
  Serial.println(WiFi.localIP());
}

void loop() {
  // Handle incoming client requests
  server.handleClient();
}
```

me-no-dev, 2023-09-12:
also not clear to me which camera board you are using?

Abhay-2412, 2023-09-12:
> also not clear to me which camera board you are using?

Greetings! I am currently utilizing the **OV5640** camera module, which is labeled as "0465-HZF" on the physical camera itself.

closed 2023-09-15 by Abhay-2412