espressif/arduino-esp32 #8635: Where to find the source file (spi_master.c) corresponding to spi_master.h

schen4116, 2023-09-15:
### Related area

core source file edit

### Hardware specification

ESP32

### Is your feature request related to a problem?

For ESP32 arduino, I can see the definition of spi_master.h, but I cannot find its course file spi_master.c. For example, I want to find the implementation of spi_device_queue_trans, I dont know where is the source file.

### Describe the solution you'd like

I want to find the path of spi_master.c (or source file with some other name).

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-09-15:
Source files are in ESP-IDF. We precompile them for Arduino and provide them as blobs + headers. https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/spi_master.c

closed 2023-09-15 by me-no-dev

schen4116, 2023-09-15:
> Source files are in ESP-IDF. We precompile them for Arduino and provide them as blobs + headers. https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/spi_master.c

Thanks for your answer. Can we edit the source file in Arduino? I need to do this for advanced usage.

me-no-dev, 2023-09-15:
for such advanced usage, I suggest you use Arduino as ESP-IDF component. Else you can not edit the file directly and test the results

schen4116, 2023-09-17:
> for such advanced usage, I suggest you use Arduino as ESP-IDF component. Else you can not edit the file directly and test the results

How to use Arduino as ESP-IDF component please? Could you provide some tutorial website?

me-no-dev, 2023-09-17:
It often pays off to try and solve things yourself. The link for the tutorial is in the README of the repo. https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

schen4116, 2023-09-18:
> It often pays off to try and solve things yourself. The link for the tutorial is in the README of the repo. https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

Thanks. I have successfully get Arduino as ESP-IDF component done. In this case, how can I modify spi_master.c and use it in my Arduino .ino sketch please?

me-no-dev, 2023-09-18:
you edit the file in your local ESP-IDF and recompile the project