espressif/arduino-esp32 #8641: esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void

deltaford, 2023-09-16:
### Board

ESP32-S2

### Device Description

ESP32-S2 Mini

### Hardware Configuration

No

### Version

v2.0.13

### IDE Name

Arduino IDE 2.2.1

### Operating System

Windows 10

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Compile warning on 2.0.13 due to missing return value. Seem related to #8629  

### Sketch

```cpp
N/A
```

### Debug Message

```plain
_C:\Users\Stefan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.13\cores\esp32\esp32-hal-uart.c: In function 'uartSetPins':
C:\Users\Stefan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.13\cores\esp32\esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
C:\Users\Stefan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.13\cores\esp32\esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)_
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-09-16:
It doesn't prevent to build the firmware, but prints a warning message. 
 With this fix, the warning message won't be displayed anymore.

This issue doesn't affect the `uartSetPins()` function because it only happens in the case that the HAL level is used with a UART number higher than the limit, which is very improbable.

@me-no-dev - I see no need for a new release.

SuGlider, 2023-09-16:
@me-no-dev / @P-R-O-C-H-Y 
I was wondering if we need to increase the warning-error check setting in CI.
These messages should be picked up in the CI.

mikrocoder, 2023-12-10:
This should definitely be rectified. Basically, this is not a warning but a serious error.

lbernstone, 2023-12-10:
It has been fixed in master, and will be released soonish.  If you want a fix today, you can patch in the pull request above (or just edit the file).

SuGlider, 2023-12-13:
The PR #8644 fixed this issue (just a warning message).
It is already in the [`release/v2.x`](https://github.com/espressif/arduino-esp32/blob/release/v2.x/cores/esp32/esp32-hal-uart.c#L153) branch and in [`master`](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L295) branch:

This is not in the 2.0.14 release package, but if the user set the 3.0.0-Alpha Core in the Arduino IDE, the fix will be there already.

Therefore, the issue can be closed.

SuGlider, 2023-12-13:
In case we release a final `Arduino Core 2.0.15` version with all the fixes from `release/v2.x` branch, it will be in the release package for the Arduino IDE users.

closed 2023-12-13 by SuGlider

AntosDR, 2025-07-23:
I was just looking for a version that fixes this warning and i saw this response... Please, the following snippets are WRONG, just make sure to return the right value for the return instruction:

> 
> ## Function minimally modified to return ESP_FAIL on uart_num out of range error:
> ```
> // Valid pin UART_PIN_NO_CHANGE is defined to (-1)
> // Negative Pin Number will keep it unmodified, thus this function can set individual pins
> bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
> {
>     if(uart_num >= SOC_UART_NUM) {
>         log_e("Serial number is invalid, please use numers from 0 to %u", SOC_UART_NUM - 1);
>         return ESP_FAIL;
>     }
> 
>     // IDF uart_set_pin() will issue necessary Error Message and take care of all GPIO Number validation.
>     bool retCode = uart_set_pin(uart_num, txPin, rxPin, rtsPin, ctsPin) == ESP_OK; 
>     return retCode;
> }
> ```
> 
> One of these should be the return value based on the uart_set_pin() function.
> 
> ```
> #define ESP_OK          0       /*!< esp_err_t value indicating success (no error) */
> #define ESP_FAIL        -1      /*!< Generic esp_err_t code indicating failure */
> ```
> 
> ## Function modified for single return value:
> ```
> // Valid pin UART_PIN_NO_CHANGE is defined to (-1)
> // Negative Pin Number will keep it unmodified, thus this function can set individual pins
> bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
> {
>     if(uart_num < SOC_UART_NUM) {
>         // IDF uart_set_pin() will issue necessary Error Message and take care of all GPIO Number validation.
>         bool retCode = uart_set_pin(uart_num, txPin, rxPin, rtsPin, ctsPin) == ESP_OK; 
>     } else {
>     // uart_num Out of range
>         bool retCode = ESP_FAIL
>         log_e("Serial number is invalid, please use numers from 0 to %u", SOC_UART_NUM - 1);
>     }
> 
>     return retCode;
> }
> ```

The ESP_FAIL is a constant value equal to -1, which is a "truthy" value and it is casted to a true boolean value! Please, the fix is a simple "false" constant. Nothing else. The original code uses the `resp == ESP_OK` instruction that verifies the returning code to be equal to ZERO, in that case a "true" boolean value is returned. Numerically speaking, using `ESP_*` constants instead of boolean values is an inverted logic.

The only fix required is the "false" at line 153:

```
if(uart_num >= SOC_UART_NUM) {
    log_e("Serial number is invalid, please use numers from 0 to %u", SOC_UART_NUM - 1);
    return false;
}
```
In any case, my version of the espressif 32 platform was the 6.6.0 and the warning was still there. Updating to 6.11.0 resolved the issue.

PR #8644 (2023-10-05): Fixes set pins return value in 2.0.13
## Description of Change
Fixes a lack of return value in the `uartSetPins()` function. It doesn't prevent to build the firmware, but prints a warning message. With this fix, the warning message won't be displayed anymore.

This issue doesn't affect the `uartSetPins()` function because it only happens in the case that the HAL level is used with a UART number higher than the limit, which is very improbable.

``` cpp
    if(uart_num >= SOC_UART_NUM) {
        log_e("Serial number is invalid, please use numers from 0 to %u", SOC_UART_NUM - 1);
        return f [... cut]