espressif/arduino-esp32 #8642: timeout: ArduinoOTA over VPN

micha6554, 2023-09-16:
### Board

ESP32

### Device Description

ESP32

### Hardware Configuration

none

### Version

v2.0.13

### IDE Name

Arduino IDE 2.2.1

### Operating System

Windows

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

 I am writing to report an issue with
the ArduinoOTA library when using it **over a VPN** connection.
Specifically, the OTA update process tends to fail approximately 90% of
the time when initiated through a VPN connection.

To address this problem, I made a modification in the **ArduinoOTA.cpp**
file. I increased the OTA timeout value from 1000 to 2000 milliseconds
by applying the following patch:

```diff from ArduinoOTA.cpp:
-, _ota_timeout(1000)
+, _ota_timeout(2000)
```

This change has resulted in a significant improvement in the OTA
functionality, even when used over a VPN connection. The increased
timeout appears to have provided the necessary buffer to ensure
successful updates under these conditions.

Best regards,

Michael Süß

### Sketch

```cpp
none
```

### Debug Message

```plain
OTA Error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-09-16:
Thanks for submitting here, Michael :)

VojtechBartoska, 2024-05-23:
how we can test this @micha6554?

SuGlider, 2024-05-23:
Why not using `void ArduinoOTAClass::setTimeout(int timeoutInMillis)` to adjust the timeout before starting the OTA?

like this: `ArduinoOTA.setTimeout(2000);`

https://github.com/espressif/arduino-esp32/blob/6bfcd6d9a9a7ecd07d656a7a88673c93e7d4d537/libraries/ArduinoOTA/src/ArduinoOTA.cpp#L383-L386

me-no-dev, 2024-05-31:
Closing as answered by @SuGlider

closed 2024-05-31 by me-no-dev