espressif/arduino-esp32 #8645: OTA Process understanding and Mistakes

meakashrao, 2023-09-18:
### Board

ESP32-WROOM-32E-N4 

### Device Description

I ended up using ESP32-WROOM-32E-N16 for the development Prototype for our Product which uses both BLE for normal Data Communication and WiFi while OTA

### Hardware Configuration

Nothing

### Version

v2.0.7

### IDE Name

Platform IO

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

For production I wanted to try ESP32-WROOM-32E-N4

and the configurations used on N16 were
with these configurations in .ini file
[image]

Now, There were some stable development done on Arduino and then I shifted to PlatfromIO, now while doing OTA N16 jumps properly between the firmware versions of Arduino and Platform IO Nicely

but N4 only jumps to the ones developed using Platform IO, although for the ones developed through Arduino, N4 says its successful but then the firmware version stays the same. 

I use esptool.py and espota.py for the OTA Process

The VS Code build is 
[image]

The Arduino Build for the same is
[image]

### Sketch

```cpp
/* 
The device Enters into OTA Mode if the Dial Switch is pressed for 5 Seconds
then the firmware restarts the device, then again at setup the device is setup into Access Point Mode
and after connecting to the ESP-32 Access point the firmware is updated 

This File contains all the definitions for OTA
1) The SSID and Password for OTA
2) OTAFlag is stored at EEPROM address 0, if the flag value is 1->OTA Mode or 
else if value is 0->Device Operation Mode
3) To Enter into OTA Mode, press the Dial Switch for 5 seconds
4) To Enter into Normal Operation Mode form OTA Mode press dial Switch for 5 Seconds
Note: Once OTA update is done, the device Enters into Normal Operation Mode by design
*/

/*
First OTA Method: 
WiFI -> Socket creation 
Look @ 
a) https://lastminuteengineers.com/esp32-ota-updates-arduino-ide/
b) https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino
*/

/*
ToDo: 
1) Find why int OTA_Flag is required, why can't I read it directly as Mohit said(figure out what the hell BackTrace addresses mean)
*/
#include <Arduino.h>
#include "main.h"
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <Update.h>
#include <EEPROM.h>
#include <ArduinoOTA.h>
#include <ESPmDNS.h>

#include "OTA_Wrapper.h"
#include "config.h"
#include "Timer_Wrapper.h"

Timer_Handler ota_timer_handler;
Timer_Handler ota_timer_handler_after_ota;

unsigned long otaTimerPeriod = 5000;
unsigned long otaTimerPeriodafterOTA = 500;
bool otaTaskCounter = false; 
int otaButtonCounter = 0;
int deviceInOTA = 0;
bool deviceToGoinOTA = false;

int OTA_Status = 0;
const char* host = "esp32";
const char *ssid_ap = "ReactHardwareCB";
const char *password_ap = "IfTzeitelMarriesLazarwolf";

unsigned long dialLongPressDatum = 0;

int OTA_Handler::read_OTA_Status()
{
    EEPROM.begin(EEPROM_SIZE);
    OTA_Status = EEPROM.read(0);
    return OTA_Status;
}

void OTA_Handler::begin()
{
    IPAddress local_ip(192, 168, 1, 1);
    IPAddress local_mask(255, 255, 255, 0);
    IPAddress gateway(192, 168, 1, 1);
    Serial.begin(9600);
    WiFi.softAP(ssid_ap, password_ap);
    WiFi.softAPConfig(local_ip, gateway, local_mask);
    Serial.println("");
    Serial.print("AP set to ");
    Serial.println(ssid_ap);
    Serial.print("IP address: ");
    Serial.println(WiFi.softAPIP());

    ArduinoOTA
        .onStart([]()
                 {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type); })
        .onEnd([]()
               {
      Serial.println("\nEnd");
      updateEEPROMValue(0);
      ESP.restart(); })
        .onProgress([](unsigned int progress, unsigned int total)
                    { Serial.printf("Progress: %u%%\r", (progress / (total / 100))); })
        .onError([](ota_error_t error)
                 {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed"); });

    ArduinoOTA.begin();
    ArduinoOTA.setPort(3232);
    ArduinoOTA.setPassword("admin");
}

void OTA_Handler::loop()
{
    if (digitalRead(sw) == LOW)
    {
        dialLongPressDatum = millis();
    }
    if (digitalRead(sw) == HIGH)
    {
        if ((millis() - dialLongPressDatum) > OTA_Mode_Switch_Time)
        {
            updateEEPROMValue(0);
            ESP.restart();
        }
    }
    ArduinoOTA.handle();
}

void OTA_Handler::setToOTA()
{
  if(otaTaskCounter)
  {
    ota_timer_handler.start(otaTimerPeriod);
    otaTaskCounter = false;
    OTA_Count_Start_Flag = true;
  }
  if(ota_timer_handler.isRunning())
  {
    if(otaButtonCounter == 10)
    {
      ota_timer_handler.stop();
      deviceInOTA = 1;
      updateEEPROMValue(1);
      char dialArray[8];  
      dialArray[0] = 'o';
      dialArray[1] = 't';   
      dialArray[2] = 'a';
      dialArray[3] = 'm';
      dialArray[4] = 'o';
      dialArray[5] = 'd';
      dialArray[6] = 'e';
      dialArray[7] = '\0';
      BLECharacteristic *notifyCharacteristic7 = myBLE.getNotifyCharacteristic("DIAL");
      if (notifyCharacteristic7 != nullptr)
      {
        myBLE.sendNotification(notifyCharacteristic7, dialArray);
      }
      OTA_Count_Start_Flag = false;
      deviceToGoinOTA = true;
      ota_timer_handler_after_ota.start(otaTimerPeriodafterOTA);
    }
    if(deviceToGoinOTA)
    {
      if(ota_timer_handler_after_ota.isFinished())
      {
        ESP.restart();
      }
    }
    if(ota_timer_handler.isFinished())
    {
      if(otaButtonCounter != 10)
      {
        OTA_Count_Start_Flag = false;
        otaButtonCounter = 0;
        deviceInOTA = 0;
      }
    }
  }
}
```

### Debug Message

```plain
Nothing
```

### Other Steps to Reproduce

NA

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

meakashrao, 2023-09-18:
The N4 chip is original and was confirmed using 
[image]

Also, wanted to ask how does 

board_build.partitions = min_spiffs.csv
differ for N16 vs N4

meakashrao, 2023-09-18:
I think this is the one I am using 
[image]
which I obtained from (https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv?plain=1)

As far as I understand, 
First hex is the offset, second hex is the size therefore, 1E0000 is 1,966,080 Bytes

and i use the same setting for both N4 which has 4MB of Flash and N16 of 16Mb of Flash, so what is the difference

SuGlider, 2023-09-21:
>am i right here? and i use the same setting for both N4 which has 4MB of Flash and N16 of 16Mb of Flash, so what is the difference

Yes, you are right.

You can change the Partition Table for the N16 in order to use all the 16MB Flash, if you need so.
You can also delete the spiffs partition and use this space for other purpose or even add it to the app0/app1 partitions.
More information about partitions: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html

meakashrao, 2023-09-21:
I have been unknowingly using the Partition table for 4MB of Flash for N16 as well and it has worked.

I will also remove Spiffs and try to use this space for N4(better usage of Flash memory)
but my problem isn't that 

I want to use N4 but the OTA always say's successful but doesnt update for Arduino based builds and for Platfrom IO based builds it works, N16 used to work correctly all the time and I havent changed any major settings just added new features which are well within the limits of app0 and app1 partition

I want some direction as to where am I going wrong.

lbernstone, 2023-09-21:
Arduino IDE has an option in the Tools menu for flash size.  This should only apply when you lay down a full image (ie, the partition table), but perhaps OTA is comparing something in the firmware to the setting in the flash.  Make sure you have this set to 4M when you build the firmware.

closed 2024-01-17 by meakashrao