espressif/arduino-esp32 #8651: ESP32C3-MINI-1 reboots when Wndows JTAG driver access it for the first time

evansgl, 2023-09-19:
### Board

ESP32-C3-MINI-1

### Device Description

ESP32-C3-MINI-1 dev board.

### Hardware Configuration

Nothing else

### Version

latest master (checkout manually)

### IDE Name

Platformio

### Operating System

WIndows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

It seems that when you open the serial JTAG/DEBUG USB port with Windows OS, the device reboots.
The strange thing is that this happens only once. After that you can open and close the serial port as many times as you want without an ESP32 reboot. 

Important note is that when you reboot Windows OS and you try to connect again to a already running ESP32C3-MINI-1 device , it reboots again.

This does not happen with other OS. e.g using platformio on OSX.
 

### Sketch

```cpp
sketch does not matter. Just open the serial port.
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

evansgl, 2023-09-19:
When you plug the USB cable directly on the JTAG pins and open the USBSerial, the device resets.
**This happens only on WIndows.** OSX and Linux does not have this behaviour

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038cdae
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710

SuGlider, 2023-09-19:
@evansgl - Yes, I can see that when the USB CDC JTAG Windows terminal is closed, it resets the ESP32-C3.

  Is this the issue?

evansgl, 2023-09-19:
I discovered that if you open the serial with handshake RTS & XON/XOFF settings , the device does not reset.
A simple c# serial software needs: this.serialPort.Handshake = Handshake.RequestToSend;

SuGlider, 2023-09-21:
RTS and DTS signals are used by the software layer to reset the board.

SuGlider, 2023-09-21:
I will close it as this seems solved by fixing the serial software windows application. Feel free to reopen it if necessary.

closed 2023-09-21 by SuGlider