espressif/arduino-esp32 #8652: Cannot receive UDP multicast packets

HX2003, 2023-09-19:
### Board

ESP32, ESP32-S3

### Device Description

-

### Hardware Configuration

Nil

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am running [AsyncUDPMulticastServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPMulticastServer/AsyncUDPMulticastServer.ino) on both a ESP32 and ESP32-S3.  On my mobile phone, I can receive the "Anyone here?" message from both ESP32 and ESP32-S3. However, no packets are being received by the ESP devices, I want the device to act as a multicast receiver. Sending unicast packets to the ESP devices works.

### Sketch

```cpp
[AsyncUDPMulticastServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPMulticastServer/AsyncUDPMulticastServer.ino)
```

### Debug Message

```plain
Nil
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-09-19:
@HX2003 - Try listeninig to any IP using special symbol `IP_ANY_TYPE` on the example port `1234`:

Line 20 of the example  [AsyncUDPMulticastServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPMulticastServer/AsyncUDPMulticastServer.ino):
`    if(udp.listenMulticast(IP_ANY_TYPE, 1234)) {`

mrengineer7777, 2023-09-19:
Also try example https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPServer/AsyncUDPServer.ino.  That's basically how my project listens for discovery packets.

HX2003, 2023-09-19:
> @HX2003 - Try listening to any IP, using the special symbol `IP_ANY_TYPE`, on the example port `1234`:
> 
> Line 20 of the example [AsyncUDPMulticastServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPMulticastServer/AsyncUDPMulticastServer.ino): ` if(udp.listenMulticast(IP_ANY_TYPE, 1234)) {`

Your suggestion resulted in a Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Basically the end result is that each of the ESP device should receive the "Anyone here?" message from the other device.

SuGlider, 2023-09-19:
Multicast IPv4 addresses go from 224.0.0.0 to 239.255.255.255.
239.1.2.3 is a possible address in this space.

All nodes shall write to this addess in order to Multicasting.
In order to receive from this Multicasting address, each node must register to listen from it.
`udp.listenMulticast(IPAddress(239,1,2,3), 1234))` does register for listening. It returns `true` if successful.
`udp.onPacket([](AsyncUDPPacket packet) {.... }` sets a call back when a UDP packet arrives. This is the place where it shall receive the Multicast Message.

`udp` is the object that binds all of it.

SuGlider, 2023-09-19:
@HX2003 - I have successfuly created a multicasting WLAN with 2 ESP32.

### Output of ESP32_#1:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
UDP Listening on IP: 192.168.15.49
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
UDP Packet Type: , From: 192.168.15.16:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#2 :: Anyone here?
```

### Output of ESP32_#2:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
UDP Listening on IP: 192.168.15.16
UDP Packet Type: , From: 192.168.15.49:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#1 :: Anyone here?
UDP Packet Type: , From: 192.168.15.49:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#1 :: Anyone here?
UDP Packet Type: , From: 192.168.15.49:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#1 :: Anyone here?
UDP Packet Type: , From: 192.168.15.49:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#1 :: Anyone here?
UDP Packet Type: , From: 192.168.15.49:1234, To: 239.1.2.3:1234, Length: 21, Data: ESP#1 :: Anyone here?
```

### Sketch for both ESP32:
``` cpp
#include "WiFi.h"
#include "AsyncUDP.h"

// Add your WiFi Credentials here
const char * ssid = "***********";
const char * password = "***********";

AsyncUDP udp;

void setup()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("WiFi Failed");
    while (1) {
      delay(1000);
    }
  }
  if (udp.listenMulticast(IPAddress(239, 1, 2, 3), 1234)) {
    Serial.print("UDP Listening on IP: ");
    Serial.println(WiFi.localIP());
    udp.onPacket([](AsyncUDPPacket packet) {
      Serial.print("UDP Packet Type: ");
      if (packet.isMulticast()) {  // <<<====   Only listen to Multicasting Messages
        Serial.print(", From: ");
        Serial.print(packet.remoteIP());
        Serial.print(":");
        Serial.print(packet.remotePort());
        Serial.print(", To: ");
        Serial.print(packet.localIP());
        Serial.print(":");
        Serial.print(packet.localPort());
        Serial.print(", Length: ");
        Serial.print(packet.length());
        Serial.print(", Data: ");
        Serial.write(packet.data(), packet.length());
        Serial.println();
      }
    });
  }
}

void loop()
{
  delay(2000);
  //Send multicast -- CHANGE ESP#1 by ESP#2 for each ESP32
  udp.print("ESP#1 :: Anyone here?");
}

```

HX2003, 2023-09-20:
Thank you all for the help. I think somethings up with my wifi router, I have tried multicast with several phones and computers, and wireshark, it has been a hit and miss, I'm facing very strange behaviour not limited to the ESP32 devices. I guess I will give up on multicasting.

lbernstone, 2023-09-20:
multicast is highly dependent on the routing/switching equipment between devices.  If you can't guarantee a good path between devices, multicast will be problematic.

SuGlider, 2023-09-21:
I have never tried it, but, maybe, the ESP32 in AP mode can route Multicasting traffic.

closed 2023-09-21 by HX2003