espressif/arduino-esp32 #8656: TWAI data frame is worng some times

FedericoAndrades, 2023-09-20:
### Board

ESP32-S2-Wrover

### Device Description

Is a custom board with CAN driver , USB , MicroSD, and RS485 driver.

### Hardware Configuration

GPIO1 - is a button
GPIO2,3,4 - are some leds
GPIO15 - is RTS485 channel change
GPIO6 - is CAN TX
GPIO7 - is CAN RX
GPIO38 - is SD detect
GPIO34 - is SD SS
Uart0 - RS485
Uart1 - TLL RS232
USB - USB
SPI - SD

### Version

latest master (checkout manually)

### IDE Name

PlatFormIO

### Operating System

Windows11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

TWAI communication works well almost all the time, there are no problems with the reception of messages in any case, but if it happens that some messages, at random, are sent (TX) with the correct ID but the data frame changed, That is, the ID is the one that is configured but the data is apparently garbage.
I tried restricting interrupts when loading "twai_transmit" but it doesn't help.
I tried printing the parameters sent to "twai_transmit" through the console before calling the function, printing them from the "twai_message_t" variable itself and they appear correct, but they are not sent and instead it sends garbage data.

The CAN protocol is its own, and has been working on other hardware for 20 years. 
Placing a protocol analyzer on the sent data, I find the error by seeing a message packet that sometimes sends both ID and data correctly:

```
---- ID ----[  DATA               ]
<ip|6|D|20|7[20,61, 0, 0, F, 0, F,]
<ip|6|C|20|7[20,61, 0, 0,64,1C,E8,]
<ip|6|E|20|7[20,61, 0, 0, 0,30,D4,]
<ip|6|5|20|3[20,61, 0,]
```

and other times it sends the ID correctly, and the data incorrectly (it is always the same data that is sent for the purposes of this test).:
```
---- ID ----[  DATA               ]
<ip|6|D|20|7[FE,AF,39,7F,46,27,7F,]
<ip|6|C|20|7[ D,51,7F,7F, 3,2B,7F,]
<ip|6|E|20|7[61,9F,7F,7F, 0,C3,50,]
<ip|6|5|20|3[77,3B,7F,]
```
What catches my attention the most is the fact that the ID remains correct and only the data is modified.

The attached code is the original TWAI message generator. I have tried printing the data from the tx_msg variable before calling the twai_transmit function and there is no problem with them, they are always correct, although then garbage data is sent.

### Sketch

```cpp
// Generadora de Mensajes NCP0x ----------------------------------------------------------------------
int NCP0x::gen_msg(  uint8_t inmediato , uint8_t prioridad , uint8_t grupo , uint8_t comando , uint8_t *buffer , uint8_t buff_len )
{
  uint8_t interrupt;
  msgNCP0x temp;
  int ret = true;
  uint8_t i;
  canID id_mensaje;
  uint8_t prn[128];
  
#if NCP0x_DEBUG_LEVEL == 2
  sprintf( prn , "\n -TX--------------------------------- ");Serial.print(prn);
#endif
  id_mensaje.reg = 0xFFFFFFFF;
  id_mensaje.NCP0xl.INMEDIATO = !inmediato;
  id_mensaje.NCP0xl.PRIORIDAD = !prioridad;
  id_mensaje.NCP0xl.GRUPO     = grupo;
  id_mensaje.NCP0xl.COMANDO   = comando;
  id_mensaje.NCP0xl.ID_NODO   = (ThisDeviceID); 
#if NCP0x_DEBUG_LEVEL == 2
  sprintf( prn , "\n ID : %0X " , id_mensaje.reg );Serial.print(prn);
  sprintf( prn , "| ID_NODO : %02X " , id_mensaje.NCP0xl.ID_NODO);Serial.print(prn);
  sprintf( prn , "| ID_GRUPO : %02X  " , id_mensaje.NCP0xl.GRUPO);Serial.print(prn);
  sprintf( prn , "| ID_COMANDO : %02X  " , id_mensaje.NCP0xl.COMANDO );Serial.print(prn);
  sprintf( prn , "| ID_DATA : %02X,",(ThisDeviceID));Serial.print(prn);
#endif
  twai_message_t tx_msg;
  tx_msg.identifier = id_mensaje.reg;
  tx_msg.data_length_code = buff_len + 1;
  tx_msg.extd = 1;
  tx_msg.data[0] = (ThisDeviceID);
  for( i = 0 ; i < buff_len ; i++ ) {
#if NCP0x_DEBUG_LEVEL == 2
    sprintf( prn , "%02X,",*buffer);Serial.print(prn);
#endif
    tx_msg.data[i+1] = *(buffer++) ;
  }
#if NCP0x_DEBUG_LEVEL == 2
  sprintf( prn , "\n ------------------------------------ \n");Serial.print(prn);
#endif
    if( twai_transmit(&tx_msg, pdMS_TO_TICKS(10)) == ESP_OK) {
#if NCP0x_DEBUG_LEVEL == 1
        Serial.print("Mensaje Enviado\n");
#endif
    } else {
        Serial.print("Failed to queue message for transmission\n");
    }
#if NCP0x_DEBUG_LEVEL >= 1
  Serial.println("Send CAN TX ----- ");
#endif  
  return( ret );
}
// ---------------------------------------------------------------------------------------------------
```

### Debug Message

```plain
nothing to show
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

FedericoAndrades, 2023-09-21:
I found the problem, the twai_message_t tx_msg varable has a register called "flags" that includes some message configurations, and when defining it local in the function this register takes any value, this causes the configuration to vary from message to message.
Error that I did not see when generating the function based on the Espressif example, which is very basic and I did not take into account its generic nature.

SuGlider, 2023-09-21:
Thanks for reporting the solution to the issue!