espressif/arduino-esp32 #8657: ESP32-S3 Touch Problem

weiwi, 2023-09-20:
### Board

Qt Py ESP32-S3 AND Seeed XIAOesp32S3

### Device Description

Nothing else. 

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHZ

### PSRAM enabled

no

### Upload speed

115200

### Description

I was trying the QT Py ESP32-S3 touch examples and looks like the readings will freeze after the touch pin is grounded and then touched without grounding. I tried both Arduino IDE examples and IDF examples, same problem. Seeed Xiaoesp32S3 has similar problem. 

Is there a fix to this? Or is there a way to “reset” the touch function without resetting the whole system, so that the touch function could still be used after ground is detached from touch pin?

I have tried other ESP32 boards(WROOM) and this does not seem to happen.

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Touch
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-09-20:
#include <driver/touch_sensor.h>
  touch_pad_deinit();

weiwi, 2023-09-21:
Thank you for the reply. I tried this, but was not able to initialize the touch function using touch_pad_init(), so touch still does not work after touch_pad_deinit(). Is there anything else needed? Thanks again.

```
void setup()
{
Serial.begin(115200);
delay(1000); // give me time to bring up serial monitor
Serial.println("ESP32 Touch Test");
}

void loop()
{
touch_pad_init();
Serial.println(touchRead(T1)); // get value using T1
delay(1000);
touch_pad_deinit();
}
```

jdeltoft, 2024-03-10:
I've been checking if any of the [APIs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/touch_pad.html) help.  

I tried:

-   meas = touch_pad_meas_is_done();
-   status = touch_pad_get_status();
-   touch_pad_clear_status();
-   touch_pad_io_init(TOUCH_PAD_NUM1);
-   touch_pad_io_init(TOUCH_PAD_NUM2);

When it's working, the get_status works with bit fields for the pins.  But when it's stuck those are stuck.  And calling io_init or clear_status does not help.  

The meas_is_done returns 1 or 0 depending on when it's called, but doesn't give any insight either when stuck

Or doing this if it gets stuck:
-   touch_pad_deinit();  
-  wait 1-2 seconds 
-   touch_pad_init();

I'm not seeing any way to reset just the touch sensor, but will continue digging into the sensor tech docs.

jdeltoft, 2024-03-10:
Maybe a slightly different question.  I also don't see a great way to know for sure they touch sensor is locked up.  Does anyone have a direct way to query this failure state (rather than just trying reads and assuming no movement is "stuck")

The reason I ask is because this reset is very fast and so far fully recovers for me if I have grounded a touch sensor input.

> ESP.restart()

VojtechBartoska, 2024-09-10:
Hello folks, it looks this issue is expired and it is quite old. To keep the issue tracker of this project manageable, I am closing this ticket. If needed, you can reopen it. Thanks for understanding.

closed 2024-09-10 by VojtechBartoska

rupraph, 2024-10-22:
I also experience a similar issue. Anyone found a solution to restart the touch when it stopped ? 

In my case I am reading multiple touch pins, and I get the freeze situation when I am touching multiple inputs at the same time, and all are reading very high values. 
I am wondering if this would not be due to the system timing out when values gets too big. 

Any lead welcome.

VojtechBartoska, 2024-10-23:
@rupraph on which Arduino version?

rupraph, 2024-10-31:
I am running on platformIO with:
PLATFORM: Espressif 32 (6.9.0)
framework-arduinoespressif32 @ 3.20017.0 (2.0.17)

Note the issue is the same whether I use the arduino API or the lower level espressif API functions. It really is the hardware stopping to measure I think. I just have no idea how to prevent that to happen on esp32-s3 or restart when stopped.

I see we can set a time out for the touch channels, but could not find an easy way to check when it actually timed out nor how to prevent it.