espressif/arduino-esp32 #8661: assertion "Invalid mbox" failed

Abhay-2412, 2023-09-22:
### Board

ESP32 wrover

### Device Description

In my project, I successfully utilized several ICs that communicate via I2C, as well as a camera. Additionally, the project incorporated both Wi-Fi and Bluetooth functionality.

### Hardware Configuration

In my project, I successfully utilized several ICs that communicate via I2C, as well as a camera. Additionally, the project incorporated both Wi-Fi and Bluetooth functionality.

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

im trying to implement a project using BLE  and WIFI features and some ICs i encounterd following error 

```
14:21:45.597 -> E (2504) cam_hal:  not in cam_obj->jpeg_mode
14:21:45.597 -> E (2504) esp32 ll_cam: node_size: 1280, nodes_per_line: 4, lines_per_node: 1, dma_half_buffer_min:  5120, dma_half_buffer: 40960,lines_per_half_buffer:  8, dma_buffer_size: 81920, image_size: 307200
14:21:45.597 -> E (2511) cam_hal: buffer_size: 81920, half_buffer_size: 40960, node_buffer_size: 1280, node_cnt: 64, total_cnt: 7
14:21:45.644 -> OV5640 camera is connected
14:22:01.373 -> assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 374, function: tcpip_send_msg_wait_sem
14:22:01.406 -> abort() was called at PC 0x400f9e1f on core 1
14:22:01.406 -> 
14:22:01.406 -> ELF file SHA256: 0000000000000000
14:22:01.406 -> 
14:22:01.406 -> Backtrace: 0x40094f1c:0x3ffccbb0 0x40095195:0x3ffccbd0 0x400f9e1f:0x3ffccbf0 0x4018a66f:0x3ffccc20 0x4019e519:0x3ffccc50 0x4019e6cd:0x3ffccc70 0x4018c654:0x3ffcccb0 0x400d5529:0x3ffcccd0 0x400d4f05:0x3ffccd20 0x401d5069:0x3ffccd50 0x400d21d1:0x3ffccd70 0x400d265b:0x3ffccd90 0x400d26ef:0x3ffccdd0 0x400df654:0x3ffcce10 0x40096e4a:0x3ffcce30
14:22:01.406 -> 
14:22:01.406 -> Rebooting...
```

### Sketch

```cpp
i can share the code privately i just want to know reason behind the error this is first time for me to encounter some error like this
```

### Debug Message

```plain
14:21:45.597 -> E (2504) cam_hal:  not in cam_obj->jpeg_mode
14:21:45.597 -> E (2504) esp32 ll_cam: node_size: 1280, nodes_per_line: 4, lines_per_node: 1, dma_half_buffer_min:  5120, dma_half_buffer: 40960,lines_per_half_buffer:  8, dma_buffer_size: 81920, image_size: 307200
14:21:45.597 -> E (2511) cam_hal: buffer_size: 81920, half_buffer_size: 40960, node_buffer_size: 1280, node_cnt: 64, total_cnt: 7
14:21:45.644 -> OV5640 camera is connected
14:22:01.373 -> assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 374, function: tcpip_send_msg_wait_sem
14:22:01.406 -> abort() was called at PC 0x400f9e1f on core 1
14:22:01.406 -> 
14:22:01.406 -> ELF file SHA256: 0000000000000000
14:22:01.406 -> 
14:22:01.406 -> Backtrace: 0x40094f1c:0x3ffccbb0 0x40095195:0x3ffccbd0 0x400f9e1f:0x3ffccbf0 0x4018a66f:0x3ffccc20 0x4019e519:0x3ffccc50 0x4019e6cd:0x3ffccc70 0x4018c654:0x3ffcccb0 0x400d5529:0x3ffcccd0 0x400d4f05:0x3ffccd20 0x401d5069:0x3ffccd50 0x400d21d1:0x3ffccd70 0x400d265b:0x3ffccd90 0x400d26ef:0x3ffccdd0 0x400df654:0x3ffcce10 0x40096e4a:0x3ffcce30
14:22:01.406 -> 
14:22:01.406 -> Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-09-22:
This typically means you are trying to set up a socket before you start the networking.  Call `WiFi.mode(WIFI_STA)` (or whatever mode you will use) early in your code.

Abhay-2412, 2023-09-25:
i followed your suggestion it got resolved , Thanks!!