espressif/arduino-esp32 #8663: SD card mount failed

zhekch, 2023-09-22:
### Board

ESP 32 dev module

### Device Description

Plain module on breadboard with micro-sd card reader attached.

### Hardware Configuration

MISO -> 19
MOSI -> 23
CS -> 5
SCK -> 18

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.2.1

### Operating System

Mac OS 13.5

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Trying to mount an SD card from esp32 fails.
Note, i read #6081, and tried all fixes described there but still did not achieve any positive results. 

### Sketch

```cpp
Default SD card example
```

### Debug Message

```plain
entry 0x400805f0
[    15][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   535][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   535][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[   537][W][sd_diskio.cpp:180] sdCommand(): crc error
[   649][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1249][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1249][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1749][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1749][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1749][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x29
[  1753][W][sd_diskio.cpp:547] ff_sd_initialize(): APP_OP_COND failed: 255
[  1760][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2269][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2269][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2023-09-27:
I've just tested it and everything is working as expected. Please make sure you are using the `SD_Test` example if you are using the SPI interface.

closed 2023-10-30 by VojtechBartoska

ihsan728, 2024-07-14:
esp323 
SD card mount failed,
why this error