espressif/arduino-esp32 #8664: BLE - Advertising of two Service UUID advertises first one two times

savejeff, 2023-09-23:
### Board

ESP32-S3

### Device Description

Plain Board

### Hardware Configuration

no extra hardware

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have observed that when i try to advertise two Serivce UUIDs only the first one is advertised but two times. 

### Sketch

```cpp
BLEAdvertising* pAdvertising = BLEDevice::getAdvertising();
pAdvertising->addServiceUUID(SERVICE_UUID_0);
pAdvertising->addServiceUUID(SERVICE_UUID_1);
pServer->getAdvertising()->start();
```

### Debug Message

```plain
Using an App like LightBlue you can look into the Advertised Data. 
only the SERVICE_UUID_0 is advertised but two times

I have tried the NimBLE-Arduino library but here it limits the Advertised Service length to 31, which is 1 short of the length of two UUIDs, might this be a general limitation where only one 128bit Servcice UUID can be advertised at the same time?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2023-09-25:
I just tested it and I was able to advertise both UUIDs (although they are indeed duplicated, probably related to #6750)

SuGlider, 2023-09-26:
@savejeff - what kind of UUID are you using? 16bits or 128bits?
I tested it and it can advertise two 16bits UUIDs, but just one 128bits UUID.
Using Arduino IDE with Arduino Core 2.0.13.

Sketch:

``` cpp
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// Short 16bits UUID
//#define SERVICE_UUID        "110A"
//#define SERVICE_UUID2        "1200"

// Long 128bits UUID
#define SERVICE_UUID        "4a4a4a4a-1fb5-459e-8fcc-c5c9c331914b"
#define SERVICE_UUID2       "5f5f5f5f-1fb5-459e-8fcc-c5c9c3319111"

void setup() {
  //BLEDevice::init("issue_8664_UUID_16bits");
  BLEDevice::init("issue_8664_UUID_128bits");

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->addServiceUUID(SERVICE_UUID2);
  BLEDevice::startAdvertising();
}

void loop() {
  delay(1000);
}
```

Advertising two 16bits UUIDs:

[image]

Advertising just one 128bits UUID:

[image]

SuGlider, 2023-09-26:
I was also able to advertise one 128bits UUID plus another 16bits UUID:

[image]

SuGlider, 2023-09-26:
@savejeff - Using BLE 4 Legacy Advertising, limited to 32 bytes data, you can advertise only one single 128 bit UUID.

Using the ESP32-S3 you can also use BLE 5 extended advertising that allows for 255 bytes data. I think that it is possible to add two 128 bit UUIDs in the Advertising Service List.

Check the Arduino BLE 5 Multi Advertising example. You can change it to just use the 1M params, with its data using the RAW ADV DATA with field 0x07 (complete 128 bit UUID service), twice, with both Service UUIDs.

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE5_multi_advertising/BLE5_multi_advertising.ino

From the Bluetooth SIG specification:
[image]

SuGlider, 2023-09-26:
Using the BLE 5 example, I have modified the Lines 96 to 99:

``` cpp
static uint8_t raw_scan_rsp_data_coded[] = {
        0x37, 0x09, 'V', 'E', 'R', 'Y', '_', 'L', 'O', 'N', 'G', '_', 'D', 'E', 'V', 'I', 'C', 'E', '_', 'N', 'A', 'M', 'E', '_',
        'S', 'E', 'N', 'T', '_', 'U', 'S', 'I', 'N', 'G', '_', 'E', 'X', 'T', 'E', 'N', 'D', 'E', 'D', '_', 'A', 'D', 'V', 'E', 'R', 'T', 'I', 'S', 'I', 'N', 'G', 0x00,
        // Adds TWO 128 bit UUID to the Complete List of 128-bit Service UUIDs in the Advertising Ext Data
        0x21, 0x07, 
              // UUID 1: "4a4a4a4a-1fb5-459e-8fcc-c5c9c331914b"
              0x4B, 0x91, 0x31, 0xC3, 0xC9, 0xC5, 0xCC, 0x8F, 
              0x9E, 0x45, 0xB5, 0x1F, 0x4A, 0x4A, 0x4A, 0x4A,
              // UUID 2: "5a5a5a5a-1fb5-459e-8fcc-c5c9c331914b"
              0x4B, 0x91, 0x31, 0xC3, 0xC9, 0xC5, 0xCC, 0x8F, 
              0x9E, 0x45, 0xB5, 0x1F, 0x5A, 0x5A, 0x5A, 0x5A               
};
```

This will solve the issue as shown in the screenshot below:

[image]

SuGlider, 2023-09-26:
@savejeff - It seems that the only way to add two 128-bit UUIDs to the Advertising data is by using BLE5 as above.
The BLE 4 Legacy 32 bytes Advertising package will only allow to add two 16-bit UUIDs, if necessary.

This is the only way to do achieve it. I think that it shall solve this issue.

savejeff, 2023-09-27:
that for looking into it @SuGlider. i was using 128-bit UUIDs. 
i was already thought it might be a protocol limitation.
I found it weird that it took both ids but then advertised the first on twice. 

i think it would be sufficient to add a comment in the addServiceUUID function description about the limitation of only a single UUID with the (i think most common) 128-bit UUIDs. just to stop users from running blindly into the same problem.

should i mark this as closed ?

closed 2023-09-27 by lucasssvaz

SuGlider, 2023-09-27:
@savejeff - a last comment:

BLE 4.2 has about 62 bytes for advertising information.
31 bytes for the main Advertising data block and another 31 for the Advertising Response data block (complementary). 

The application can create a customized Advertising Data and/or Advertising Response Data, to add more information, like a more UUIDs or Manufacturer data, for instance, .

`void BLEAdvertising::setScanResponseData(BLEAdvertisementData& advertisementData)` can be used to set a complementary advertising information. 

This example seems to work fine for advertising two 128-bit UUIDs:
SERVICE_UUID0        "4a4a4a4a-1fb5-459e-8fcc-c5c9c331914b"
SERVICE_UUID1        "5a5a5a5a-1fb5-459e-8fcc-c5c9c331914b"
You can change it to whatever may be needed.

``` cpp
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  
  uint8_t advData[] = { // MAX 31 bytes!!
    0x02, 0x01, 0x06, // Advertising Flags
    0x11, 0x06,   // adds information of incomplete list of 128-bit UUIDS
      //SERVICE_UUID0        "4a4a4a4a-1fb5-459e-8fcc-c5c9c331914b"
      0x4B, 0x91, 0x31, 0xC3, 0xC9, 0xC5, 0xCC, 0x8F,
      0x9E, 0x45, 0xB5, 0x1F, 0x4A, 0x4A, 0x4A, 0x4A,
    // Min, Max Interval
    0x05, 0x12, 0x20, 0x00, 0x40, 0x00,
    // PowerTx
    0x02, 0x0A, 0x09
    // TOTAL: 0x02 + 0x11 (17) + 0x05 + 0x02 + 4 bytes (length data) = 30 bytes, OK
  };

  BLEAdvertisementData BLEAdvData;
  BLEAdvData.addData(std::string((const char*)advData, sizeof(advData)));
  pAdvertising->setAdvertisementData(BLEAdvData);

  uint8_t scanRspData[] = { // MAX 31 bytes!!
    0x11, 0x06,   // adds information of incomplete list of 128-bit UUIDS
      //SERVICE_UUID1        "5a5a5a5a-1fb5-459e-8fcc-c5c9c331914b"
      0x4B, 0x91, 0x31, 0xC3, 0xC9, 0xC5, 0xCC, 0x8F,
      0x9E, 0x45, 0xB5, 0x1F, 0x5A, 0x5A, 0x5A, 0x5A,
    // Device Name "ESP32-2UUID"  // Device Name MAX Length is 12 bytes (31 - 17 - 2)
    0xC, 0x09, 0x45, 0x53, 0x50, 0x33, 0x32, 0x2D, 0x32, 0x55, 0x55, 0x49, 0x44
    // TOTAL: 0x11 (17) + 0x0C (12) + 2 bytes (length data) = 31 bytes, OK.
  };

  BLEAdvertisementData BLEScanRspData;
  BLEScanRspData.addData(std::string((const char*)scanRspData, sizeof(scanRspData)));
  pAdvertising->setScanResponseData(BLEScanRspData);

```

[image]

[image]