espressif/arduino-esp32 #8669: External flash write

savagerex, 2023-09-25:
### Board

ESP32 Wrover e

### Device Description

External flash

### Hardware Configuration

NONE

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

i use "fwrite"API to write flash about 1.7M.

it need 22 seconds.

I think it spends too much time.

Is there any other API that can be faster?

### Sketch

```cpp
// Create a file in FAT FS
    ESP_LOGI(TAG, "Opening file");
    FILE *f = fopen("/extflash/hello.txt", "wb");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    //fprintf(f, "Written using ESP-IDF %s\n", esp_get_idf_version());
    for (i = 0; i < 1781999; i++) {
                DataBufWrite[i] = '1';
                //Serial.print(i);
    }
    StartTestTime = millis();                  
    fwrite (DataBufWrite, sizeof(uint8_t), 1782000, f);
    
    fclose(f);
    //ESP_LOGI(TAG, "File written");
    Serial.println("File written");
    EndTestTime = millis();
    Serial.print("**************Flash Write Time*************");
    Serial.println(EndTestTime - StartTestTime);
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2023-09-27:
@savagerex which flash chip are you connecting?

For example, GD25Q32 datasheet indicates a typical 50ms erase time for a single sector. So even without data transfer, write operation is limited by 4kB/50ms = 80kB/s. Which is very close to the speed you are getting.

SuGlider, 2023-09-27:
Based on the last related issue https://github.com/espressif/arduino-esp32/issues/8479 - it seems that the Flash chip is the **W25Q256JV**

SuGlider, 2023-09-27:
Datasheet: https://www.winbond.com/hq/support/documentation/?__locale=en&line=/product/code-storage-flash-memory/index.html&family=/product/code-storage-flash-memory/serial-nor-flash/index.html&category=/.categories/resources/datasheet/&pno=W25Q256JV

[image]

savagerex, 2023-09-28:
OK, Thanks

VojtechBartoska, 2023-10-04:
Closing this as answered.

closed 2023-10-04 by VojtechBartoska