espressif/arduino-esp32 #8670: Crash whet twai driver started after 2 Wire instances

AcuarioCat, 2023-09-25:
### Board

ESP32

### Device Description

ESP32 dev board but also on WROOM (no difference)

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Visual Mocro

### Operating System

Windows

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

ESP crashes if twai driver is started AFTER 2 Wire drivers. 
With a single Wire driver it works fine starting after the Wire driver.. 
If the twai driver is started BEFORE the Wire drivers there is no problem.

The problem only occurs when using both Wire instances (Wire and Wire1).

Problem occurs on 2.0.5 and latest 2.0.13

### Sketch

```cpp
#include <Wire.h>
#include "driver/twai.h"

#define CAN_TX_GPIO    GPIO_NUM_12
#define CAN_RX_GPIO    GPIO_NUM_13
#define WIRE_SDA        GPIO_NUM_14
#define WIRE_SCL        GPIO_NUM_17
#define WIRE1_SDA        GPIO_NUM_21
#define WIRE1_SCL        GPIO_NUM_22

static const twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(CAN_TX_GPIO, CAN_RX_GPIO, TWAI_MODE_NO_ACK);
static const twai_timing_config_t t_config = TWAI_TIMING_CONFIG_25KBITS();
static const twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();
void i2c_scan(byte bus);

void setup()
{
    Serial.begin(115200);
    while (!Serial) {}
    Serial.println("__ OK __");

    Wire.setPins(WIRE_SDA, WIRE_SCL);
    if (!Wire.begin())
        Serial.println("Wire failed to start");											// Initialize I2C SDA/SCL
    i2c_scan(0);
    Wire1.setPins(WIRE1_SDA, WIRE1_SCL);
    if (!Wire1.begin())
        Serial.println("Wire1 failed to start");											// Initialize I2C SDA/SCL
    i2c_scan(1);

    //Install TWAI driver, trigger tasks to start
    twai_driver_install(&g_config, &t_config, &f_config);

    Serial.println("TWAI Driver installed");
    twai_start();
    Serial.println("TWAI Driver started");

}

void loop()
{
    delay(1);
}

void i2c_scan(byte bus)
{
    byte error, address;
    int nDevices;
    Serial.printf("Scanning bus %d...\n", bus);
    nDevices = 0;
    for (address = 1; address < 127; address++)
    {
        // The i2c_scanner uses the return value of the Write.endTransmisstion to see if a device did acknowledge to the address.
        if (bus == 0)
        {
            Wire.beginTransmission(address);
            error = Wire.endTransmission();
        }
        else
        {
            Wire1.beginTransmission(address);
            error = Wire1.endTransmission();
        }

        if (error == 0)
        {
            Serial.print("I2C device found at address 0x");
            if (address < 16)
                Serial.print("0");
            Serial.print(address, HEX);
            Serial.println("  !");
            nDevices++;
        }
        else if (error == 4)
        {
            Serial.print("Unknown error at address 0x");
            if (address < 16)
                Serial.print("0");
            Serial.println(address, HEX);
        }
    }
    if (nDevices == 0)
        Serial.println("No I2C devices found\n");
    else
        Serial.println("done\n");
}
```

### Debug Message

```plain
ELF file SHA256: fe561b0ff47ec610

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
__ OK __
Scanning bus 0...
No I2C devices found

Scanning bus 1...
No I2C devices found

ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x400883a8
file: "IDF/components/driver/twai.c" line 447
func: twai_driver_install
expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)

abort() was called at PC 0x400883ab on core 0

Backtrace: 0x40083769:0x3ffb2150 0x400883b5:0x3ffb2170 0x4008d351:0x3ffb2190 0x400883ab:0x3ffb2210 0x400d5aea:0x3ffb2230 0x400d1609:0x3ffb2260 0x400d231e:0x3ffb2290
```

### Other Steps to Reproduce

Move the twai install and start to before the Wire section and it does not crash.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2023-09-25:
I couldn't get it to crash when testing with an ESP32 DevKitC V4:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1416
load:0x40078000,len:14804
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
__ OK __
Scanning bus 0...
No I2C devices found

Scanning bus 1...
No I2C devices found

TWAI Driver installed
TWAI Driver started
```

AcuarioCat, 2023-09-26:
As far as I´m aware it is all up to date - in my ide I have ESP32 Dev Module selected which selects the esp32 bootloader.
I tried with a more recent ESP32 device but still the same.

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
__ OK __
Scanning bus 0...
No I2C devices found

Scanning bus 1...
No I2C devices found

ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x400883a8
file: "IDF/components/driver/twai.c" line 447
func: twai_driver_install
expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)

abort() was called at PC 0x400883ab on core 0

Backtrace: 0x40083769:0x3ffb2150 0x400883b5:0x3ffb2170 0x4008d351:0x3ffb2190 0x400883ab:0x3ffb2210 0x400d5aea:0x3ffb2230 0x400d1609:0x3ffb2260 0x400d231e:0x3ffb2290

atanisoft, 2023-09-26:
This crash is due to running out of interrupt levels. You can fix this in your code by removing `const` from `g_config` and adding the following before initializing TWAI:
```
g_config.intr_flags |= ESP_INTR_FLAG_SHARED;
```

The default value for this field is to request exclusive use of a given interrupt level, the above code allows the interrupt level to be shared.

AcuarioCat, 2023-09-27:
I modified the code as suggested but it still crashes but with a same error.

lucasssvaz, 2023-09-27:
@AcuarioCat The issue might be related to your development environment. We currently do not support any IDEs beside the Arduino IDE. Please try using it with the latest ´arduino-esp32´.

AcuarioCat, 2023-09-27:
@lucasssvaz I have now found how to fix the problem and what will cause it.
I dusted off the Arduino IDE and built the code and it didn´t crash. 
Comparing the VM and Arduino settings I noticed in the Arduino IDE both Arduino and Events are set to run on Core 1. On VM Arduino was set to run on Core 0 and events on core 1.

Changing VM to Core 1 for both stopped the crash. Also changing the Arduino IDE to run on Core 0 made the compiled code in the Arduino IDE crash. Using Core 0 for both and it crashes when Twai is started after Wire.

So it seems the following:
- With Arduino and Events both on Core 1 Twai can start after Wire and Wire1
- With Arduino on Core 1 and Events on core 0  Twai can start after Wire and Wire1
- With Arduino and Events both on Core 0 Twai has to start before 2 Wire instances or it crashes
- With Arduino on Core 0 and Events on core 1 Twai has to start before 2 Wire instances or it crashes

Extrapolating further it seems:
- To start Twai AFTER Wire and Wire1 Arduino MUST run on core 1
- If Arduino runs on Core 0 Twai MUST be started BEFORE Wire and Wire1

If only 1 Wire instance is used (Wire or Wire1) there is no problem with cores or order of starting the drivers.

lucasssvaz, 2023-09-28:
@AcuarioCat Please check this comment: https://github.com/espressif/esp-idf/issues/6709#issuecomment-801176126
In this case the initialization order will indeed matter.

lucasssvaz, 2023-09-29:
@AcuarioCat This was fixed in IDF 5.1: https://github.com/espressif/esp-idf/commit/3a79e3694c85475756fd99adaca3ba0cd1573aa7

Release 3.0.0 will include this fix.

closed 2024-01-26 by lucasssvaz