espressif/arduino-esp32 #8673: SimpleFOC InlineCurrentSense.init block

mailonghua, 2023-09-26:
### Board

ESP32S3

### Device Description

Board drawn by myself using ESP32S3 FN8 chip

### Hardware Configuration

InlineCurrentSense current_sense = InlineCurrentSense((float)0.01, (float)20, (int)A8, (int)A9);

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Win10

### Flash frequency

NULL

### PSRAM enabled

yes

### Upload speed

115200

### Description

When using SimpleFOC, the program will stay at the current_sense.init(); function location

### Sketch

```cpp
#include <SimpleFOC.h>

MagneticSensorI2C sensor = MagneticSensorI2C(AS5600_I2C);
static BLDCMotor motor = BLDCMotor(7,7.8); 
static BLDCDriver3PWM driver = BLDCDriver3PWM(5, 6, 7, 4);

// 设置目标电压
float target_voltage = 2;
// instantiate the commander
static Commander command = Commander(Serial);
void doTarget(char* cmd) { command.scalar(&motor.target, cmd); }

//电流检测
InlineCurrentSense current_sense = InlineCurrentSense((float)0.01, (float)20, (int)A8, (int)A9);

void setup() {
  // monitoring port
  Serial.begin(115200);

  // configure i2C
  
  Wire.begin(1,0,(uint32_t)400000);
  Wire.setClock(400000);//400000
  // initialise magnetic sensor hardware
  sensor.init(&Wire);
  motor.linkSensor(&sensor);//1.链接传感器和电机
  //驱动器
  driver.voltage_power_supply = 24;
  driver.init();
  // 连接电机和驱动器
  motor.linkDriver(&driver);//2.链接电机驱动和电机

  //初始化电流检测传感器
  Serial.println("Start init current_sense");
  current_sense.init();
  Serial.println("End init current_sense");
  motor.linkCurrentSense(&current_sense);

  motor.torque_controller = TorqueControlType::foc_current;//foc电流力矩控制
  motor.controller = MotionControlType::torque;
  // motor.foc_modulation = FOCModulationType::SpaceVectorPWM;
  // motor.torque_controller = TorqueControlType::voltage;
  // motor.controller = MotionControlType::torque;
  //电流环的q和d的PID
  motor.PID_current_q.P = 5;
  motor.PID_current_q.I= 300;
  motor.PID_current_d.P= 5;
  motor.PID_current_d.I = 300;

  //低通滤波器
  motor.LPF_current_q.Tf = 0.01; 
  motor.LPF_current_d.Tf = 0.01; 

   // 校准电压
  //motor.voltage_sensor_align = 3;
  motor.current_limit = 1.5;

  motor.useMonitoring(Serial);
   // 初始化电机
  motor.init();
  // 校准编码器，启用FOC
  motor.initFOC();

  //设置电机初目标

  //command.add('T', doTarget, "target velocity");
  command.add('T', doTarget, "target current");
  Serial.println("Sensor ready");
  _delay(1000);
}

void loop() {
  motor.loopFOC();
  // 开环速度运动
  // 使用电机电压限制和电机速度限制
  motor.move();
  // iterative function updating the sensor internal variables
  // it is usually called in motor.loopFOC()
  // this function reads the sensor hardware and 
  // has to be called before getAngle nad getVelocity
  command.run();
  //sensor.update();
 
}
```

### Debug Message

```plain
Executing task: C:\Users\mailonghua\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
Start init current_sense
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-09-26:
You will need to ask for help on 3rd party modules at their repo/website.

lucasssvaz, 2023-09-26:
@mailonghua As @lbernstone said, it appears that the issue is related to a third-party library rather than our project itself. While I'd like to assist you as much as possible, the nature of this issue falls outside the scope of our direct control.
Please open an issue on https://github.com/simplefoc/Arduino-FOC to get the desired help.

closed 2023-09-27 by mailonghua