espressif/arduino-esp32 #8675: Audio streaming over Bluetooth

podaen, 2023-09-27:
### Related area

BT

### Hardware specification

ESP32-S

### Is your feature request related to a problem?

Hi,

I have the classic ESP32 working with the A2DP profile and I want to make the step to S3 and I just want to ask if it's possible to stream audio from an android device using for example Spotify and stream it to the ESP S series? If yes how could that been done? I don't need the details, just some direction to look for. Thanks!

Kr
Podaen

### Describe the solution you'd like

?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lucasssvaz, 2023-09-27:
Unfortunately I don't think that is possible. A2DP is part of the Bluetooth classic spec only and ESP32-S3 does not support BT classic.

VojtechBartoska, 2023-10-04:
Closing this as answered, there is nothing we can do.

closed 2023-10-04 by VojtechBartoska