espressif/arduino-esp32 #8678: ESP32-S3-WROOM-2-N16R8V does not start Arduino Sketch

BornaBiro, 2023-09-27:
### Board

ESP32-S3-WROOM-2-N16R8V module

### Device Description

Custom setup (module directly wired to USB UART programmer)

### Hardware Configuration

USB UART connected to RX and TX lines, 10k pull-up on EN GPIO.

### Version

v2.0.13

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm having weird issue. The Arduino Sketch simply does not start and ESP32 keeps rebooting. The reboot happens every 9 seconds. The sketch successfully gets uploaded on ESP32 but it does not start. The sketch is pretty simple, it just sends something on the serial. My Arduino Board settings are shown here
[image]

Module in question is shown below:
[image]

Is there is something wrong with my Arduino Board settings or with my module? What is interesting, when using ESP32-S3-WROOM-2-N32R8V with the same Arduino Board settings, same sketch and same hardware setup, Arduino Sketch starts without any problems.

### Sketch

```cpp
int i = 0;

void setup()
{
    // Start Serial communication at 115200 baud.
    Serial.begin(115200);
}

void loop()
{
    // Print message on Serial.
    Serial.printf("test %d\r\n", i++);

    // Wait a little bit.
    delay(1000);
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x292c
entry 0x403c98bc
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x292c
entry 0x403c98bc
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x292c
entry 0x403c98bc
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x292c
entry 0x403c98bc
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-09-28:
Flash Mode is not OPI. The module has QIO Flash. PSRAM OPI is correct

BornaBiro, 2023-09-28:
But the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) of the modules on page 3 shows that both 16MB and 32MB versions have Octal SPI (OPI) on Flash memory.

[image]

Jason2866, 2023-09-28:
The bootloop you see is typical for wrong used Flash mode.

BornaBiro, 2023-09-28:
I understand that, but the datasheet clearly points out that both of the modules use OPI Flash memory. Could be an error in the datasheet, but I highly doubt that. Even on Espressif [page ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#ordering-information) states that 16MB and 32MB versions use OPI Flash.

What is more interesting, if I disable PSRAM (but still keep Flash Mode to OPI) it runs normally.

SuGlider, 2023-09-28:
I have the ESP32-S3-WROOM-2-N32R8 devkit here (ESP32-S3-DevkitC-1 v1.1).
It has both, Flash and PSRAM, in OPI mode.

If I set PSRAM to QIO, it displays an error:
`E (114) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PStest 0`

Therefore, it must use all in OPI  mode or set PSRAM disabled.

When PSRAM is OPI I see the message that says it has been initialized:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x292c
entry 0x403c98bc
[   114][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
test 0
test 1
test 2
```

SuGlider, 2023-09-28:
There is a caveat with PSRAM and ESP32-S3 documented:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html#hardware
`Some PSRAM chips are 1.8 V devices and some are 3.3 V. The working voltage of the PSRAM chip must match the working voltage of the flash component. Consult the datasheet for your PSRAM chip and ESP32-S3 device to find out the working voltages. For a 1.8 V PSRAM chip, make sure to either set the MTDI pin to a high signal level on bootup, or program ESP32-S3 eFuses to always use the VDD_SIO level of 1.8 V. Not doing this can damage the PSRAM and/or flash chip.`

SuGlider, 2023-09-28:
About the 1.8 vs 3.3 volts for Flash/PSRAM ==> ESP32-S3 has a strapping GPIO for that:

Flash voltage (VDD_SPI) determined by GPIO45 on reset (GPIO45=High: VDD_SPI pin is powered from internal 1.8V LDO
GPIO45=Low or NC: VDD_SPI pin is powered directly from VDD3P3_RTC_IO via resistor Rspi. Typically this voltage is 3.3 V).

Try pulling up/down GPIO45 to verify if it can work with PSRAM.

BornaBiro, 2023-09-28:
@SuGlider Thank you for the help. But unfortunately neither pulling GPIO45 to the GND nor to the 3.3V at boot time did not help. Same behaviour.

For me, it looks like it's trying to initialize PSRAM but it keeps failing and then after 9 seconds WDT triggers and resets the ESP32. GPIO0 is pulled to 3.3V and other two strapping pins are for ROM messages and JTAG.

Also, I tested with the QIO on PSRAM and got exactly the same result as @SuGlider.

SuGlider, 2023-09-28:
It sounds like the module has a dead PSRAM chip...

BornaBiro, 2023-09-28:
Tried second ESP32-S3-WROOM-2-N16R8V module, same thing. But it could be that both have faulty PSRAM IC...

SuGlider, 2023-09-28:
It is hard to tell. I have just the full devkit board here. It works fine with PSRAM enabled.

You may try to see the devkit schematic and try to get as close as possible to its circuit.

It could be something related to it.

lbernstone, 2023-09-29:
Do you have capacitors wired in somewhere, or is that a 3v3 hooked directly to EN/VCC?

Seenov, 2023-10-27:
After many tries over a few months, this post got me on track and I got my ESP32S3 dev kit to work.
I am using Arduino 2.2.1 on a Linux Min machine.
I updated the Espressif files as outlined here: [](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)
I now put a header on my ESP32 code to save the Arduino configuration. A better tool would be greatly appreciated.
`/*******************************************************
 Seenov inc. OCT 27 2023  rev 1 
 Arduino 2  board setup ESP32-S3-DevKitC-1 v1.1 
 ESP32S3  WROOM 2 MON32R8V

Board: "ESP32S3 Dev Module" Port: "/dev/ttyUSB0"
Get Board Info
USB CDC On Boot: "Disabled"
CPU Frequency: "240MHz (WiFi)"
Core Debug Level: "Debug"
USB DFU On Boot: "Disabled"
Erase All Flash Before Sketch Upload: "Disabled"
Events Run On: "Core 1"
Flash Mode: "OPI 80MHz"
Flash Size: "32MB (256Mb)" JTAG Adapter: "Disabled"
Arduino Runs On: "Core 1"
USB Firmware MSC On Boot: "Disabled"
Partition Scheme: "32M Flash (4.8MB APP/22MB LittleFS)"
PSRAM: "OPI PSRAM"
Upload Mode: "UARTO / Hardware CDC"
Upload Speed: "921600"
USB Mode: "Hardware CDC and JTAG"

========================================

****************************************/

#define LED_PIN     38  // RGB LED

void setup() {
delay(100);
 Serial.begin(115200);
 delay(1000);
  Serial.println();
  Serial.println("configuring..."); 
  
}

void loop() {
  
  Serial.print("It works!  ");
  delay(5000);
}

`

Good luck.

lbernstone, 2023-10-27:
> After many tries over a few months, this post got me on track and I got my ESP32S3 dev kit to work. I am using Arduino 2.2.1 on a Linux Min machine. I updated the Espressif files as outlined here: I [...]

I have added an entry for octal S3 boards. It will have the necessary memory settings pre-selected. https://github.com/espressif/arduino-esp32/commit/e8f82286b5cff0b1f979b873ede61d8e7e693cd5

VojtechBartoska, 2023-12-12:
I think we can close this ticket, thanks for your PR @lbernstone. If needed, you can reopen.

closed 2023-12-12 by VojtechBartoska