espressif/arduino-esp32 #8681: WiFiClient::flush() crashes due to Load Prohibit since 2.0.12, earlier versions work fine.

fredlcore, 2023-09-29:
### Board

ESP32 Olimex EVB and other ESP32 boards

### Device Description

None.

### Hardware Configuration

Nothing

### Version

v2.0.13

### IDE Name

Arduino IDE 2.2.1

### Operating System

MacOS Sonoma

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

This change leads to a Guru meditation due to Load Prohibited when calling `client.flush()` on a WiFiClient instance:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e3847  PS      : 0x00060f30  A0      : 0x800e3950  A1      : 0x3ffb1f50  
A2      : 0x00000000  A3      : 0x3ffbfd38  A4      : 0x00000384  A5      : 0x00000001  
A6      : 0x000000ff  A7      : 0x3f4059d4  A8      : 0x8008fccc  A9      : 0x3ffb1f40  
A10     : 0x00000001  A11     : 0x0002710d  A12     : 0x00027385  A13     : 0x00060f23  
A14     : 0x00060f20  A15     : 0x00000001  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4008ad85  LEND    : 0x4008ad95  LCOUNT  : 0xfffffff5  
Backtrace: 0x400e3844:0x3ffb1f50 0x400e394d:0x3ffb1f80 0x400dc1fd:0x3ffb1fa0 0x400ee93d:0x3ffb2290
ELF file SHA256: 4726c43adf41a5ad
```
ESP Stack Trace decoder gives me this:
```
PC: 0x400e387f: WiFiClientRxBuffer::r_available() at /Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/src/WiFiClient.cpp line 46
EXCVADDR: 0x00000010

Decoding stack results
0x400e387c: WiFiClientRxBuffer::r_available() at /Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/src/WiFiClient.cpp line 44
0x400e3985: WiFiClient::flush() at /Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/src/WiFiClient.cpp line 158
0x400df411: loop() at /Users/frederik/Downloads/BSB-LAN-master/BSB_LAN/BSB_LAN.ino line 7073
0x400ee1f5: loopTask(void*) at /Users/frederik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/cores/esp32/main.cpp line 50
```
The trace narrows the culprit down to line 46:
```
if(_fd < 0){
```
If I do a `Serial.println(_fd);` right before that, the ESP32 crashes.
It seems not to happen all the time, but the reason either seems to be if there are still characters in the incoming client buffer (for lack of a better word) or if there are no more characters coming in.
This problem only occurs in 2.0.12 and 2.0.13, once I revert back to 2.0.11 the problem is gone.
Can you please investigate this further and if possible revert to the state from 2.0.11?
Thank you!

### Sketch

```cpp
client.flush();
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e3847  PS      : 0x00060f30  A0      : 0x800e3950  A1      : 0x3ffb1f50  
A2      : 0x00000000  A3      : 0x3ffbfd38  A4      : 0x00000384  A5      : 0x00000001  
A6      : 0x000000ff  A7      : 0x3f4059d4  A8      : 0x8008fccc  A9      : 0x3ffb1f40  
A10     : 0x00000001  A11     : 0x0002710d  A12     : 0x00027385  A13     : 0x00060f23  
A14     : 0x00060f20  A15     : 0x00000001  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4008ad85  LEND    : 0x4008ad95  LCOUNT  : 0xfffffff5  
Backtrace: 0x400e3844:0x3ffb1f50 0x400e394d:0x3ffb1f80 0x400dc1fd:0x3ffb1fa0 0x400ee93d:0x3ffb2290
ELF file SHA256: 4726c43adf41a5ad
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2023-09-29:
Could you please provide a sample sketch that causes this error ?

fredlcore, 2023-10-02:
Here's a minimalistic reduced version:
```
#include <Arduino.h>
#include <WiFi.h>
#include <Ethernet.h>

EthernetClient client1;
WiFiClient client2;

void loop() {
  delay(1);
  Serial.print(".");
}

void setup() {
  Serial.begin(115200);
  Serial.println("READY");
  Serial.println("Flush EthernetClient client...");
  client1.flush();
  delay(5000);
  Serial.println("Flush WiFiClient client...");
  client2.flush();
  delay(5000);
  Serial.println("Setup complete");
}
```
You can see that client1.flush() goes through, client2.flush() crashes. If you revert back to 2.0.11 or earlier, both work fine.

Jason2866, 2023-10-02:
@me-no-dev Please take a look. We run in this issue with Tasmota too. Older core does work as for @fredlcore 
Probably caused from https://github.com/espressif/arduino-esp32/pull/8541

Jason2866, 2023-10-02:
Reverting does solve. Checked with this sketch

```
#include <Arduino.h>
#include <WiFi.h>

WiFiClient client;

void loop() {
  delay(1);
  Serial.print(".");
}

void setup() {
  Serial.begin(115200);
  Serial.println("READY");
  delay(1000);
  Serial.println("Flush WiFiClient client...");
  client.flush();
  delay(2000);
  Serial.println("Setup complete");
}
```

Jason2866, 2023-10-03:
@lucasssvaz Solved? The reason is found for the issue. Not solved. Should i open a revert PR?

lucasssvaz, 2023-10-03:
Ah, I thought that PR was in this repo as it had the same name, my bad 😓. Yes, please open a revert PR as this won't break anything.

lucasssvaz, 2023-10-03:
As this affects the WiFi it will probably lead to version 2.0.14 ASAP

lucasssvaz, 2023-10-03:
@me-no-dev @SuGlider Should we fix this in 3.0.0 or create a 2.0.14 release ?

fredlcore, 2023-10-03:
Please don't wait until 3.0.0 - Our project has several hundred installs and when the users install the next update, it will crash all the time, unless I remove all client.flush() instances.

SuGlider, 2023-10-03:
> @me-no-dev @SuGlider Should we fix this in 3.0.0 or create a 2.0.14 release ?

I think that we should target 3.0.0.

fredlcore, 2023-10-03:
May I ask why? 2.0.12 and 2.0.13 have been released relatively shortly after one another, partly because of bug fixes, so what is different here?

lucasssvaz, 2023-10-04:
I've just submitted a patch to fix it. 2.0.14 should be released soon including it.

Jason2866, 2023-10-04:
@lucasssvaz  CI fails

lucasssvaz, 2023-10-04:
It was just a reorder warning. I forgot to enable my compilation warnings locally.

PR #8699 (2023-10-04): WiFiClient - Properly initialize and check _rxBuffer
## Description of Change
Fixes `LoadProhibited` exception (introduced in #8541) by properly initializing and checking the `_rxBuffer` pointer in the WiFiClient.

## Tests scenarios
Tested on ESP32-DevKitC using the following sketch:

```
#include <Arduino.h>
#include <WiFi.h>

WiFiClient client;

void loop() {
  delay(1000);
  Serial.println(".");
}

void setup() {
  Serial.begin(115200);
  Serial.println("READY");
  delay(1000);
  Serial.println("Flush WiFiClient client...");
  client.flush();
  delay(2000);
  Serial.println("Setup complete [... cut]
closed 2023-10-04 by me-no-dev