espressif/arduino-esp32 #8682: Problem with SPI connecting vs1053 codec?

CelliesProjects, 2023-09-29:
### Board

esp32 s3 box

### Device Description

The board is connected to a vs1053 codec chip. (the codec chip is working on all other boards I tested on including a LOLIN s3 mini)

### Hardware Configuration

The pins are defined as follows:
```
platform = espressif32
board = esp32s3box
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
; board_build.flash_mode = qio
build_flags =
    ${env.build_flags}
    -D VS1053_CS_PIN=10
    -D VS1053_DCS_PIN=21
    -D VS1053_DREQ_PIN=42                    
    -D SPI_CLK_PIN=12
    -D SPI_MISO_PIN=13
    -D SPI_MOSI_PIN=11
    ; XRST from VS1053 connects to pin 41 that has to be HIGH to enable the codec
    -D VS1053_RST_PIN=41
```

And SPI is started like this:
```
    SPI.setHwCs(true);
    SPI.begin(SPI_CLK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN);

    static ESP32_VS1053_Stream audio;

    if (!audio.startDecoder(VS1053_CS_PIN, VS1053_DCS_PIN, VS1053_DREQ_PIN) || !audio.isChipConnected())
    {
        log_e("VS1053 board could not init\nSystem halted");
        while (true)
            delay(100);
    }
```

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Mint Linux

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

The board just does not finish starting the SPI bus. No error, nothing but an unresponsive system.

The problem happens in this code in the vs1053 lib:
```
void VS1053::begin() {
    pinMode(dreq_pin, INPUT); // DREQ is an input
    pinMode(cs_pin, OUTPUT);  // The SCI and SDI signals
    pinMode(dcs_pin, OUTPUT);
    digitalWrite(dcs_pin, HIGH); // Start HIGH for SCI en SDI
    digitalWrite(cs_pin, HIGH);
    delay(100);
    LOG("\n");
    LOG("Reset VS1053...\n");
    digitalWrite(dcs_pin, LOW); // Low & Low will bring reset pin low
    digitalWrite(cs_pin, LOW);
    delay(500);
    LOG("End reset VS1053...\n");
    digitalWrite(dcs_pin, HIGH); // Back to normal again
    digitalWrite(cs_pin, HIGH);
    delay(500);
    // Init SPI in slow mode ( 0.2 MHz )
    VS1053_SPI = SPISettings(200000, MSBFIRST, SPI_MODE0);  <- This call never returns 
    // printDetails("Right after reset/startup");
    delay(20);
    // printDetails("20 msec after reset");
    if (testComm("Slow SPI,Testing VS1053 read/write registers...\n")) {
        //softReset();
        // Switch on the analog parts
        writeRegister(SCI_AUDATA, 44101); // 44.1kHz stereo
        // The next clocksetting allows SPI clocking at 5 MHz, 4 MHz is safe then.
        writeRegister(SCI_CLOCKF, 6 << 12); // Normal clock settings multiplyer 3.0 = 12.2 MHz
        // SPI Clock to 4 MHz. Now you can set high speed SPI clock.
        VS1053_SPI = SPISettings(4000000, MSBFIRST, SPI_MODE0);
        writeRegister(SCI_MODE, _BV(SM_SDINEW) | _BV(SM_LINE1));
        testComm("Fast SPI, Testing VS1053 read/write registers again...\n");
        delay(10);
        await_data_request();
        endFillByte = wram_read(0x1E06) & 0xFF;
        LOG("endFillByte is %X\n", endFillByte);
        //printDetails("After last clocksetting") ;
        delay(100);
    }
}
```
The problem occurs with this call:
```    VS1053_SPI = SPISettings(200000, MSBFIRST, SPI_MODE0); ```

### Sketch

```cpp
#include <Arduino.h>
#include <VS1053.h>               /* https://github.com/baldram/ESP_VS1053_Library */
#include <ESP32_VS1053_Stream.h>

#define SPI_CLK_PIN 18
#define SPI_MISO_PIN 19
#define SPI_MOSI_PIN 23

#define VS1053_CS 5
#define VS1053_DCS 21
#define VS1053_DREQ 22

ESP32_VS1053_Stream stream;

const char* SSID = "xxx";
const char* PSK = "xxx";

void setup() {
#if defined(CONFIG_IDF_TARGET_ESP32S2) && ARDUHAL_LOG_LEVEL != ARDUHAL_LOG_LEVEL_NONE
    delay(3000);
    Serial.setDebugOutput(true);
#endif

    Serial.begin(115200);

    WiFi.begin(SSID, PSK);

    Serial.println("\n\nSimple vs1053 Streaming example.");

    while (!WiFi.isConnected())
        delay(10);
    Serial.println("wifi connected - starting decoder");

    SPI.setHwCs(true);
    SPI.begin(SPI_CLK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN);  /* start SPI before starting decoder */

    if (!stream.startDecoder(VS1053_CS, VS1053_DCS, VS1053_DREQ) || !stream.isChipConnected())
    {
        Serial.println("Decoder not running");
        while (1) delay(100);
    };

    Serial.println("decoder running - starting stream");

    stream.connecttohost("http://espace.tekno1.fr/tekno1.m3u");

    Serial.print("codec: ");
    Serial.println(stream.currentCodec());

    Serial.print("bitrate: ");
    Serial.print(stream.bitrate());
    Serial.println("kbps");

}

void loop() {
    stream.loop();
    //Serial.printf("Buffer status: %s\n", stream.bufferStatus());
    delay(5);
}

void audio_showstation(const char* info) {
    Serial.printf("showstation: %s\n", info);
}

void audio_showstreamtitle(const char* info) {
    Serial.printf("streamtitle: %s\n", info);
}

void audio_eof_stream(const char* info) {
    Serial.printf("eof: %s\n", info);
}
```

### Debug Message

```plain
No debug messages, just an unresponsive system.
```

### Other Steps to Reproduce

I can not reproduce the issue on any of my other boards including a LOLIN S3 mini.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2023-09-29:
Could you please post the output when setting the core debug level to "Debug" ?

lucasssvaz, 2023-09-29:
@SuGlider @me-no-dev Do you guys have a S3 Box to test this, please ?

me-no-dev, 2023-09-30:
You sure your SPI Frequency is OK? It's set to 200KHz currently, which is very very low.

CelliesProjects, 2023-09-30:
Debug does not add a lot.

```c++
E (207) esp_core_dump_flash: Incorrect size of core dump image: [  1160][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1184][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1185][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.0.105, MASK: 255.255.255.0, GW: 192.168.0.1
[  1218][I][eStreamPlayer32_VS1053.ino:432] setup(): WiFi connected - IP 192.168.0.105
[  1219][I][eStreamPlayer32_VS1053.ino:440] setup(): Waiting for NTP sync...
[  1222][I][eStreamPlayer32_VS1053.ino:445] setup(): Synced
[  1228][D][percentEncode.ino:141] percentEncode(): Input str: Search API provided by: <a href="https://www.radio-browser.info/" target="_blank"><span style="white-space:nowrap;">radio-browser.info</span></a>
[  1245][D][percentEncode.ino:142] percentEncode(): Returning html encoded str: Search API provided by: <a href="https://www.radio-browser.info/" target="_blank"><span style="white-space:nowrap;">radio-browser.info</span></a>
[  1265][D][eStreamPlayer32_VS1053.ino:653] audio_showstreamtitle(): streamtitle
Search API provided by: <a href="https://www.radio-browser.info/" target="_blank"><span style="white-space:nowrap;">radio-browser.info</span></a>
[  1285][D][eStreamPlayer32_VS1053.ino:644] audio_showstation(): showstation
eStreamPlayer 2.1.1
FILE
[  1294][I][eStreamPlayer32_VS1053.ino:597] setup(): Webserver started
[  1300][I][eStreamPlayer32_VS1053.ino:51] playerTask(): Starting VS1053 codec...
[  1408][D][VS1053.cpp:156] begin(): [ESP_VS1053] 

[  1408][D][VS1053.cpp:157] begin(): [ESP_VS1053] Reset VS1053...

[  1909][D][VS1053.cpp:161] begin(): [ESP_VS1053] End reset VS1053...

[  2429][D][VS1053.cpp:119] testComm(): [ESP_VS1053] VS1053 not properly installed!
```

CelliesProjects, 2023-09-30:
@me-no-dev Just tried starting SPI with the default 1Mhz but gave the exact same results.

CelliesProjects, 2023-09-30:
Just wired and checked another codec board and getting the same result.

SuGlider, 2023-10-01:
@CelliesProjects - My suggestion would be to try another library: https://github.com/adafruit/Adafruit_VS1053_Library , 
at least to verify where could be the problem. Run some of the examples to check HW and SW.

CelliesProjects, 2023-10-03:
@me-no-dev @SuGlider Thanks for looking into this.
Already tested the AdaFruit lib a couple of days ago.

Just re-installed the adafruit library and triple-tested the wiring.
Also does not work.
Unfortunately there is the same lack of output in debug mode:
```
...
Adafruit VS1053 Simple Test
Couldn't find VS1053, do you have the right pins defined?
```
Also the ada lib runs fine on al my other test boards except the s3 box.

me-no-dev, 2023-10-03:
can you send me your project? I have 1053 and S3-Box here and can give it a test. Settings look OK to me.

me-no-dev, 2023-10-03:
[image]

Here is the problem. The diode prevents the S3-Box to power the external 5V lines. They are only for power-in.

CelliesProjects, 2023-10-03:
@me-no-dev Yes, just found out that there is no power on those pins.
Thanks. You at least saved me finding out if it was by design.

@all Thanks for the effort.

closed 2023-10-03 by CelliesProjects