espressif/arduino-esp32 #8683: ESP32 CAM better settings?

zekageri, 2023-09-29:
### Board

ESP32_CAM

### Device Description

5MP ESP32-CAM OV5640
ESP32-S?

### Hardware Configuration

-

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

So i have bought an ESP32-S with an OV5640 camera module.
I got a really dissapointing picture quality but i have seen others make a lot more out of it.
I can't find any settings to adjust more.

Any recommendation for better settings for this camera?

This is the best i could set so far

[image]

### Sketch

```cpp
#include <Arduino.h>
#include "esp_camera.h"
#include "ESP32_OV5640_AF.h"

OV5640 ov5640 = OV5640();

void enableAutoFocus(sensor_t* sensor){
    ov5640.start(sensor);

    // Download Firmware for AF
    if (ov5640.focusInit() == 0) {
        Serial.println("OV5640_Focus_Init Successful!");
    }

    // Set Continuous AF MODE
    if (ov5640.autoFocusMode() == 0) {
        Serial.println("OV5640_Auto_Focus Successful!");
    }
}

void setConfig() {
    config.ledc_channel = LEDC_CHANNEL_0;
    config.ledc_timer = LEDC_TIMER_0;
    config.pin_d0 = Y2_GPIO_NUM;
    config.pin_d1 = Y3_GPIO_NUM;
    config.pin_d2 = Y4_GPIO_NUM;
    config.pin_d3 = Y5_GPIO_NUM;
    config.pin_d4 = Y6_GPIO_NUM;
    config.pin_d5 = Y7_GPIO_NUM;
    config.pin_d6 = Y8_GPIO_NUM;
    config.pin_d7 = Y9_GPIO_NUM;
    config.pin_xclk = XCLK_GPIO_NUM;
    config.pin_pclk = PCLK_GPIO_NUM;
    config.pin_vsync = VSYNC_GPIO_NUM;
    config.pin_href = HREF_GPIO_NUM;
    config.pin_sccb_sda = SIOD_GPIO_NUM;
    config.pin_sccb_scl = SIOC_GPIO_NUM;
    config.pin_pwdn = PWDN_GPIO_NUM;
    config.pin_reset = RESET_GPIO_NUM;

    config.xclk_freq_hz = 20000000;
    config.frame_size = FRAMESIZE_XGA; //FRAMESIZE_SVGA;
    config.pixel_format = PIXFORMAT_JPEG;  // for streaming
    config.grab_mode = CAMERA_GRAB_LATEST;
    config.fb_location = CAMERA_FB_IN_PSRAM;
    config.jpeg_quality = 9;
    config.fb_count = 2;

    esp_err_t err = esp_camera_init(&config);
    if (err != ESP_OK) {
        Serial.printf("[Camera] - Init failed with error 0x%x", err);
        return;
    }

    sensor_t* s = esp_camera_sensor_get();
    // s->set_contrast(s, INITIAL_CONTRAST);
    // s->set_brightness(s, INITIAL_BRIGHTNESS);

    enableAutoFocus(s);
}
```

### Debug Message

```plain
There is no crash but poor quality.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2023-09-29:
I don't have this module to test nor have I ever used it to give advice. If anyone has experience with it, tips are appreciated :)

zekageri, 2023-09-29:
Yeah, I'm waiting. For some reasonn my picture's left side is brighter. :/

lucasssvaz, 2023-09-29:
With the little experience I have with DSLR cameras, this looks like some artifact from the lens/sensor rather than software configuration (I might be wrong though)

zekageri, 2023-09-29:
Yeah, it's an aliexpress cam. It might be faulty. :/

VojtechBartoska, 2023-12-12:
Closing as wont do, there is nothing we can do here. :)

closed 2023-12-12 by VojtechBartoska

stringandstickytape, 2024-09-21:
> Yeah, I'm waiting. For some reasonn my picture's left side is brighter. :/

Many months late to the party, but in case it helps anyone - I have two ESP32S3CAMs which demonstrably do exactly this while they rapidly overheat.  If I point a fan at them, I can watch the bright white effect on one side of the image, fade and then disappear.  Turn the fan off, it fades back in.