espressif/arduino-esp32 #8686: How add support for ethernet chip (KSZ8851)

FrancoFerrara, 2023-10-01:
### Related area

Ethernet peripheral support 

### Hardware specification

KSZ8851 10/100 Base-T/TX Ethernet Controller with SPI interface

### Is your feature request related to a problem?

The Ethernet KSZ8851 cannot be use with the ETH.h. https://github.com/espressif/arduino-esp32/blob/02e31b4001892535602bb0c0cc9b42e14d0c4901/libraries/Ethernet/src/ETH.h#L56C1-L56C9

### Describe the solution you'd like

I want to add support for this chip but i don't know how to implement a new ethernet controller in ETH. Is there documentation or procedure to how to add a new chip ?

### Describe alternatives you've considered

Using another chip like KSZ8081 that is already supported. 

### Additional context

https://www.microchip.com/en-us/product/KSZ8851

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2023-10-01:
Not possible to add SPI driven PHY Ethernet chips to current Arduino core 2.0.x.
This will be possible with the upcoming Arduino 3.0.0

me-no-dev, 2023-10-02:
as @Jason2866 said. Coming in 3.0.0: https://github.com/espressif/arduino-esp32/pull/8575

closed 2023-10-02 by me-no-dev

ewpa, 2024-06-15:
Hi @FrancoFerrara refer to this if you need this in 2.0.x: https://github.com/ewpa/arduino-esp32/commit/85f06c982e5c70af414029853e38dfe6303e9bdd

FrancoFerrara, 2024-06-17:
> Hi @FrancoFerrara refer to this if you need this in 2.0.x: [ewpa@85f06c9](https://github.com/ewpa/arduino-esp32/commit/85f06c982e5c70af414029853e38dfe6303e9bdd)

Hi thank you very much.