espressif/arduino-esp32 #8688: unable to write efuse

Endorfin35, 2023-10-02:
### Board

ESP32 Dev Module

### Device Description

Devkit

### Hardware Configuration

No connection

### Version

v2.0.11

### IDE Name

Platformio & Arduino Framework

### Operating System

win 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I want to read/write 3 byte data to Efuse memory with the program code. After writing the data, when I restart and read again, the data is lost. I'm adding the links I got reference below.

I thought I should call the esp_efuse_burn_new_values() function after the writing process, but when I call this function, I cannot read the data I wrote. 

I don't understand what should I do? I am requesting your support.

https://esp32.com/viewtopic.php?t=5742
https://github.com/espressif/arduino-esp32/issues/2556
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#efuse

### Sketch

```cpp
#include "Arduino.h"
#include <soc/efuse_reg.h>
#include "esp_efuse.h"

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("EFUSE TEST");

  uint32_t data_1; 
  uint32_t data_2 = 0xF1F2F3F4;
  uint32_t data_3;

  Serial.print("WILL BE WRTIE   : "); 
  Serial.println(data_2, HEX);

  data_1 = REG_READ(EFUSE_BLK1_WDATA7_REG);
  Serial.print("[OLD] BLK DATA  : "); 
  Serial.println(data_1, HEX);

  //esp_efuse_reset();
  REG_WRITE(EFUSE_BLK1_WDATA7_REG,data_2);
  //esp_efuse_burn_new_values();

  data_3 = REG_READ(EFUSE_BLK1_WDATA7_REG);
  Serial.print("[NEW] BLK1 DATA : "); 
  Serial.println(data_3, HEX);

}
```

### Debug Message

```plain
// Serial Output
EFUSE TEST
WILL BE WRTIE   : F1F2F3F4
[OLD] BLK DATA  : 0
[NEW] BLK1 DATA : F1F2F3F4

// Serial Output Second attemp with esp_efuse_burn_new_values()
WILL BE WRTIE   : F1F2F3F4
[OLD] BLK DATA  : 0
[NEW] BLK1 DATA : 0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Endorfin35, 2023-10-02:
I made the following additions to the sdkconfig file.

CONFIG_EFUSE_CUSTOM_TABLE=y
CONFIG_EFUSE_VIRTUAL=y

But, There was no change...
I can read efuse records. For example, I can read data such as MAC address without any problems. It seems possible to write. But there is a problem with the commit process. If you have any other advice, I want to hear it.

I completed my project with Arduino framework. I have no IDF experience. This is the last option for me. So I want to try the possibilities. In the link below, it is mentioned that the OPTIONAL_UNIQUE_ID efuse record is used by rainmaker. Can I write Efuse indirectly with Arduino? All I need is just one recording.

Thanks.

https://github.com/espressif/esp-idf/issues/8819

SuGlider, 2023-10-02:
Each eFuse is a one-bit field which can be programmed to 1 after which it cannot be reverted back to 0.
Therefore, if it went out from the manufacturer already written, it can't be changed anymore, except by turning it into 1.

https://github.com/espressif/esp-idf/issues/8819#issuecomment-1481163019

Endorfin35, 2023-10-03:
Thank you for your help.

I will use the Flash section.

closed 2023-10-03 by Endorfin35