espressif/arduino-esp32 #8689: HTTPS_OTA_Update.ino Fails with mbedtls_ssl_handshake returned -0x0050

orbian, 2023-10-02:
### Board

ESP32 Dev Module

### Device Description

HiLetgo ESP-WROOM-32

### Hardware Configuration

None

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying to run the HTTPS_OTA_Update sketch and it is failing. The only parts I have changed are the server certificate and the URL. But it always fails with mbedtls_ssl_handshake returned -0x0050

I have pulled the certificate from my site using the command:
openssl s_client -connect www.bluemummy.com:443

I am assuming the returned certificate is the one I am supposed to embed.

### Sketch

```cpp
// This sketch provide the functionality of OTA Firmware Upgrade
#include "WiFi.h"
#include "HttpsOTAUpdate.h"
// This sketch shows how to implement HTTPS firmware update Over The Air.
// Please provide your WiFi credentials, https URL to the firmware image and the server certificate.

static const char *ssid     = "ssid";  // your network SSID (name of wifi network)
static const char *password = "pwd"; // your network password

static const char *url = "https://www.bluemummy.com/firmware.bin"; //state url of your firmware image

static const char *server_certificate = R"(-----BEGIN CERTIFICATE-----
MIIGMjCCBRqgAwIBAgIQC52eketMaq0FP6EVdUaVFjANBgkqhkiG9w0BAQsFADBg
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMR8wHQYDVQQDExZSYXBpZFNTTCBUTFMgUlNBIENBIEcx
MB4XDTIzMTAwMTAwMDAwMFoXDTI0MDkzMDIzNTk1OVowHDEaMBgGA1UEAxMRd3d3
LmJsdWVtdW1teS5jb20wggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDS
Ar/v1XTULaUnLaKuxNhusaNxprpSh6/1MimT2QGabmPYlXvr40XnWFz58hUhGPwa
HsVPhLNR2ifkT3M4+olESV4JqzS3xANLBnJYR3ADot3A0ZmGvU/O6n7s+eMh9jT0
aIRJmW3wg5rIWWLPoNHwdrWz4VJg3f/6PGEJXA0HT1PQkClIMamSzcs9ky+OpQX+
n7aS58qA8AHAXL50FlgHRS1R+9tv8xPTKRe7WxGAWFFfpUAuqgFtbmQ8YaL3Nk3s
YzRKrbl17//k6FBnXG11/0nc6cAXuADAKm4IjImNkYh/BFAyFczVat43mCitdL43
KcUXeEnHOFIAAqXnxkcNAgMBAAGjggMqMIIDJjAfBgNVHSMEGDAWgBQM22yCSQ9K
Zwq4FO56xEhSiOtWODAdBgNVHQ4EFgQUz5Vxs7aJhxCtnpdSmek1fehF0/wwKwYD
VR0RBCQwIoIRd3d3LmJsdWVtdW1teS5jb22CDWJsdWVtdW1teS5jb20wPgYDVR0g
BDcwNTAzBgZngQwBAgEwKTAnBggrBgEFBQcCARYbaHR0cDovL3d3dy5kaWdpY2Vy
dC5jb20vQ1BTMA4GA1UdDwEB/wQEAwIFoDAdBgNVHSUEFjAUBggrBgEFBQcDAQYI
KwYBBQUHAwIwPwYDVR0fBDgwNjA0oDKgMIYuaHR0cDovL2NkcC5yYXBpZHNzbC5j
b20vUmFwaWRTU0xUTFNSU0FDQUcxLmNybDB2BggrBgEFBQcBAQRqMGgwJgYIKwYB
BQUHMAGGGmh0dHA6Ly9zdGF0dXMucmFwaWRzc2wuY29tMD4GCCsGAQUFBzAChjJo
dHRwOi8vY2FjZXJ0cy5yYXBpZHNzbC5jb20vUmFwaWRTU0xUTFNSU0FDQUcxLmNy
dDAMBgNVHRMBAf8EAjAAMIIBfwYKKwYBBAHWeQIEAgSCAW8EggFrAWkAdwDuzdBk
1dsazsVct520zROiModGfLzs3sNRSFlGcR+1mwAAAYrr4fQkAAAEAwBIMEYCIQCU
Q4q6ABafz5nw7qYKLI+yySMO2QuKKBJAOnbT0ESttQIhAOsCCJYMI2ksnoYWfN7x
VuEVF/SfZqNh1IEXp3gdiq3LAHYASLDja9qmRzQP5WoC+p0w6xxSActW3SyB2bu/
qznYhHMAAAGK6+Hz0QAABAMARzBFAiEAyYm5uod7Sga2b4+gLmw/i2XRwisCiIeS
k8yuhK4rBvYCIGQ69WO79cHDgZmLCiOXTljbDMjG6vifSMagpJd8jdraAHYA2ra/
az+1tiKfm8K7XGvocJFxbLtRhIU0vaQ9MEjX+6sAAAGK6+HzygAABAMARzBFAiEA
5+KdUUNpbNY3UrEHXPJhsNOZYBngGrUcvtkwO0YFIXcCIAZ6upTYkn6d8zPKA//O
a6tNoCkjNU0HhJKriFyjAGHGMA0GCSqGSIb3DQEBCwUAA4IBAQAg1gDmN4sDPqCh
152+biabdE6tujr+xZrKZd0d5GFzdrUCQNKoAys5ukgeLn4rSG2lfyr2rJxVY0CD
n19/Qpa6Y2jAWC9P/KmEAbiWgXhziO21Kh3aB8QJQPs1WGOs34rrNV7eoreEl/fq
eCLhs7QXzLLuklARCkNq1401XZv4BUBXae/s+IFZYA+o5dF5LOE1Cl1IY0r8nnBl
EInv2wWl5jZJN+BtbA0WUs2At16e5SbUFJqr4DPbzIVpPah+gI/0yMlOQWc7+HwH
mTp+SIaDnw6QOOjYFuFnr0WYbhPGHrNm/tGsmhf2bLNJI9eIzcH7E3rYX5ucxoQI
5SpdXYIN
-----END CERTIFICATE-----)";

static HttpsOTAStatus_t otastatus;

void HttpEvent(HttpEvent_t *event)
{
    switch(event->event_id) {
        case HTTP_EVENT_ERROR:
            Serial.println("Http Event Error");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            Serial.println("Http Event On Connected");
            break;
        case HTTP_EVENT_HEADER_SENT:
            Serial.println("Http Event Header Sent");
            break;
        case HTTP_EVENT_ON_HEADER:
            Serial.printf("Http Event On Header, key=%s, value=%s\n", event->header_key, event->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            break;
        case HTTP_EVENT_ON_FINISH:
            Serial.println("Http Event On Finish");
            break;
        case HTTP_EVENT_DISCONNECTED:
            Serial.println("Http Event Disconnected");
            break;
    }
}

void setup(){

    Serial.begin(115200);
    Serial.println(server_certificate);
    Serial.print("Attempting to connect to SSID: ");
    WiFi.begin(ssid, password);

    // attempt to connect to Wifi network:
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(1000);
    }

    Serial.print("Connected to ");
    Serial.println(ssid);
    
    delay(1000);
    HttpsOTA.onHttpEvent(HttpEvent);
    Serial.println("Starting OTA");
    HttpsOTA.begin(url, server_certificate,false); 

    Serial.println("Please Wait it takes some time ...");
}

void loop(){
    otastatus = HttpsOTA.status();
    if(otastatus == HTTPS_OTA_SUCCESS) { 
        Serial.println("Firmware written successfully. To reboot device, call API ESP.restart() or PUSH restart button on device");
    } else if(otastatus == HTTPS_OTA_FAIL) { 
        Serial.println("Firmware Upgrade Fail");
    }
    delay(1000);
}
```

### Debug Message

```plain
Attempting to connect to SSID: ..Connected to SSID
Starting OTA
Please Wait it takes some time ...
E (3676) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x0050
E (3676) esp-tls: Failed to open new connection
E (3676) TRANSPORT_BASE: Failed to open a new connection
E (3681) HTTP_CLIENT: Connection failed, sock < 0
Http Event Error
E (3695) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (3695) esp_https_ota: Failed to establish HTTP connection
Http Event Disconnected
Http Event Disconnected
Firmware Upgrade Fail
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

orbian, 2023-10-02:
Tried with skip_cert_common_name_check set to true and false. false gives mbedtls_ssl_handshake returned -0x2700

lbernstone, 2023-10-02:
For a chain you should include all CA certificates.

orbian, 2023-10-02:
Got it. Thanks! But it only works with skip_cert_common_name_check=false (the default is true).

lbernstone, 2023-10-02:
That entirely depends on how the cert is generated and how you are accessing it (IP address vs domain name)

orbian, 2023-10-02:
> That entirely depends on how the cert is generated and how you are accessing it (IP address vs domain name)

Thanks.

closed 2023-10-02 by orbian