espressif/arduino-esp32 #8690: Possible serial noise or corruption error when trying to upload code to ESP32 wroom32U custom board

walidbmd, 2023-10-02:
### Board

ESP32 WROOM32U

### Device Description

Custom PCB board with uses esp32-wroom32U

### Hardware Configuration

custom board

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

windows 10

### Flash frequency

i tried 40 and 80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When i try to upload any simple code , i get this corruption error , i have removed the IO12 as recommnded, changed the baud rate, added 1uF between enable pin and GND, tried both manuel and auto upload methods, changed the pc, changed the cables and i still have nothing. Tho it worked for like 3 4 times and then stopped. i would appreciate your help !!!!

### Sketch

```cpp
i tried empty codes
```

### Debug Message

```plain
A fatal error occurred: Invalid head of packet (0x65): Possible serial noise or corruption.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-10-03:
Not enough information in order to help this issue.

VojtechBartoska, 2023-12-12:
Closing as expired

closed 2023-12-12 by VojtechBartoska