espressif/arduino-esp32 #8692: WPS PBC Method don't work with WPA3

khlabg22, 2023-10-03:
### Board

ESP32 PICO D4

### Device Description

Plain ESP32 PICO D4 on a breadboard.

### Hardware Configuration

Blank device only the USB Port is connected

### Version

v2.0.13

### IDE Name

Arduino IDE

### Operating System

Ubuntu 20.04 LTS

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The Board can't get a WiFi conection to my Router with the WPS PBC Method. After 2 Minutes the WPS is timed out.

### Sketch

```cpp
The WPS example provided with the Arduino IDE
packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/examples/WPS/WPS.ino
```

### Debug Message

```plain
13:43:04.902 -> ets Jun  8 2016 00:22:57
13:43:04.902 -> 
13:43:04.902 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:43:04.902 -> configsip: 188777542, SPIWP:0xee
13:43:04.902 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:43:04.935 -> mode:DIO, clock div:1
13:43:04.935 -> load:0x3fff0030,len:1184
13:43:04.935 -> load:0x40078000,len:13260
13:43:04.935 -> load:0x40080400,len:3028
13:43:04.935 -> entry 0x400805e4
13:43:05.200 -> [    22][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
13:43:05.233 -> 
13:43:05.266 -> [    71][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
13:43:05.365 -> Starting WPS
13:43:05.365 -> [   158][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
13:43:05.365 -> [   158][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
13:43:05.365 -> Station Mode Started
13:43:06.061 -> sm BSSid: 00:00:00:00:00:00 scan BSSID 1e:ed:6f:9c:fc:5d
13:43:07.914 -> [  2725][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: forguests, BSSID: 1e:ed:6f:9c:fc:5d, Reason: 202
13:43:07.914 -> [  2726][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[... 454 lines cut ...]
13:45:05.632 -> [120433][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 205 - CONNECTION_FAIL
13:45:05.632 -> [120441][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
13:45:05.665 -> [120462][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
13:45:05.665 -> Disconnected from station, attempting reconnection
13:45:08.081 -> [122877][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: forguests, BSSID: 00:00:00:00:00:00, Reason: 15
13:45:08.081 -> [122878][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:45:08.081 -> [122885][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
13:45:08.081 -> [122893][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
13:45:08.114 -> [122915][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
13:45:08.114 -> Disconnected from station, attempting reconnection
13:45:10.531 -> [125330][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: forguests, BSSID: 00:00:00:00:00:00, Reason: 15
13:45:10.531 -> [125331][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:45:10.531 -> [125338][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
13:45:10.531 -> [125346][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
13:45:10.565 -> [125368][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
13:45:10.565 -> Disconnected from station, attempting reconnection
```

### Other Steps to Reproduce

I also tried with ESP32-S2 dev board. Same behavior.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-10-03:
The WPA3 standard dropped WPS support, as it's considered insecure.

mrengineer7777, 2023-10-03:
https://en.wikipedia.org/wiki/Wi-Fi_Protected_Access

khlabg22, 2023-10-03:
Hello David, for WPS with PIN I agree, because it can't disabled on some routers. 
The WPS push button is only enabled for 2 minutes. My router FRITZ!Box 6591 Cable supports WPS PBC together with WPA3.
I did a Test with ESP-IDF and the provided WPS.ino to get more detailed debug output. The result is very interesting. 
With esp-idf v4.4.5 the result is the same as with Arduino IDE. The WPS example  compiled with esp-idf 5.1.1 works like charm.
See the attended log, please.

[esp32-wps-wpc-wpa3-espidf-445-debug.log.txt](https://github.com/espressif/arduino-esp32/files/12794641/esp32-wps-wpc-wpa3-espidf-445-debug.log.txt)
[esp32-wps-wpc-wpa3-espidf-511-debug.log.txt](https://github.com/espressif/arduino-esp32/files/12794642/esp32-wps-wpc-wpa3-espidf-511-debug.log.txt)

Jason2866, 2023-10-04:
Support in IDF is not relevant. The IDF settings choosen to compile the Arduino libs are without support for WPS.

khlabg22, 2023-10-05:
Hello Jason, the WPS support is enabled in arduino-esp32.
I did a test with the brandnew arduino-esp32 2.0.14. With this update the WPS PBC Method works with WPA3 as expected.
See the Log below, please:

20:43:28.930 -> ets Jun  8 2016 00:22:57
20:43:28.930 -> 
20:43:28.930 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:43:28.930 -> configsip: 188777542, SPIWP:0xee
20:43:28.930 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:43:28.930 -> mode:DIO, clock div:1
20:43:28.930 -> load:0x3fff0030,len:1184
20:43:28.930 -> load:0x40078000,len:13260
20:43:28.930 -> load:0x40080400,len:3028
20:43:28.930 -> entry 0x400805e4
20:43:29.227 -> [    22][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
20:43:29.459 ->  
20:43:29.459 ->  
20:43:29.459 -> ------------ 
20:43:29.459 ->  
20:43:29.459 ->  Source: /home/esp32/arduino-1.8.19/portable/sketchbook/wifi-test/WPS/WPS.ino
20:43:29.459 ->  compiled at: Oct  5 2023, 20:42:15
20:43:29.459 ->  SDK Version: v4.4.6-dirty
20:43:29.459 -> 
20:43:29.459 ->  Arduino ESP32 SDK: 2_0_14-44da992b
20:43:29.492 -> 
20:43:29.492 -> ESP32 Chip model = ESP32-PICO-D4 Rev 1
20:43:29.492 -> This chip has 2 cores
20:43:29.492 -> Chip ID: xxxxxxxx
20:43:29.492 -> PSRAM size:      0
20:43:29.492 -> Free PSRAM size: 0
20:43:29.492 -> Total heap size: 336916
20:43:29.492 -> Available heap : 311436
20:43:29.492 -> Min free heap:   305796
20:43:29.492 -> Largest block of heap: 110580
20:43:29.492 -> ---> I2C:  SDA: 21, SCL: 22
20:43:29.492 -> ---> SPI:  SS: 5, SCK: 18, MOSI: 23, MISO: 19
20:43:29.492 -> 
20:43:29.525 -> [   322][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
20:43:29.592 -> Starting WPS
20:43:29.592 -> [   387][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
20:43:29.592 -> [   388][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
20:43:29.592 -> Station Mode Started
20:43:55.388 -> sm BSSid: 00:00:00:00:00:00 scan BSSID 1e:ed:6f:11.22.33
20:43:58.236 -> [ 29022][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: wpa3wifi, BSSID: 1e:ed:6f:11.22.33, Reason: 23
20:43:58.236 -> [ 29023][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
20:43:58.236 -> [ 29030][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 23 - 802_1X_AUTH_FAILED
20:43:58.236 -> [ 29038][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
20:43:58.269 -> [ 29046][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
20:43:58.269 -> Disconnected from station, attempting reconnection
20:43:59.229 -> [ 30021][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 24 - WPS_ER_SUCCESS
20:43:59.229 -> WPS Successfull, stopping WPS and connecting to: 
20:43:59.229 -> [ 30035][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: wpa3wifi, BSSID: 1e:ed:6f:11.22.33, Reason: 8
20:43:59.229 -> [ 30045][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
20:43:59.262 -> [ 30054][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
20:43:59.262 -> [ 30054][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 8 - ASSOC_LEAVE
20:43:59.262 -> Disconnected from station, attempting reconnection
20:44:00.024 -> [ 30805][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: wpa3wifi, BSSID: 1e:ed:6f:11.22.33, Reason: 202
20:44:00.024 -> [ 30806][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
20:44:00.024 -> [ 30814][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
20:44:00.024 -> Disconnected from station, attempting reconnection
20:44:00.752 -> [ 31560][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: wpa3wifi, BSSID: 1e:ed:6f:11.22.33, Reason: 202
20:44:00.785 -> [ 31561][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
20:44:00.785 -> [ 31569][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
20:44:00.785 -> Disconnected from station, attempting reconnection
20:44:03.202 -> [ 33995][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: wpa3wifi, BSSID: 00:00:00:00:00:00, Reason: 205
20:44:03.202 -> [ 33996][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
20:44:03.202 -> [ 34003][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 205 - CONNECTION_FAIL
20:44:03.290 -> [ 34011][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
20:44:03.290 -> [ 34019][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
20:44:03.290 -> Disconnected from station, attempting reconnection
20:44:04.593 -> [ 35394][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: wpa3wifi, BSSID: 1e:ed:6f:11.22.33, Channel: 6, Auth: WPA2_WPA3_PSK
20:44:05.999 -> [ 35396][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
20:44:05.999 -> [ 35560][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.179.3
20:44:05.999 -> [ 35560][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
20:44:05.999 -> [ 35563][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.179.3, MASK: 255.255.255.0, GW: 192.168.179.1
20:44:05.999 -> Connected to :wpa3wifi
20:44:05.999 -> Got IP: 192.168.179.3

khlabg22, 2023-10-06:
With Arduino-ESP32 I have also ESP32S2 and ESP32S3 successfull tested. 
From my point of view this Issue can be closed.

closed 2023-10-06 by khlabg22