espressif/arduino-esp32 #8698: Custom Partition Table - Failed to execute script gen_esp32part

Oleg-Perevyshin, 2023-10-04:
### Board

ESP32 Dev Module

### Device Description

-

### Hardware Configuration

-

### Version

v2.0.13

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I needed my own partition table.
I made it according to the example.
When compiling a project I constantly see an error.

I calculated everything on a calculator and double-checked everything. All characters are in Latin, but the project is not assembled.
What could be the problem?

### Sketch

```cpp
# Partition Table for 4M
# Name,    Type, SubType,  Offset,   Size
nvs,      data, nvs,        0x9000, 0x4000,
otadata,  data, ota,        0xD000, 0x2000,
phy_init, data, phy,        0xF000, 0x1000,
app0,     app,  ota_0,     0x10000, 0x1A0000,
app1,     app,  ota_1,    0x1B0000, 0x1A0000,
spiffs,   data, spiffs,   0x350000, 0xA0000,
coredump, data, coredump, 0x3F0000, 0x10000,
```

### Debug Message

```plain
Traceback (most recent call last):
  File "gen_esp32part.py", line 534, in <module>
  File "gen_esp32part.py", line 480, in main
UnicodeDecodeError: 'ascii' codec can't decode byte 0xef in position 0: ordinal not in range(128)
Failed to execute script gen_esp32part
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-04:
the first byte of the file seems to be 0xef, which is non-printable, so your editor probably acts, like it's not there. There should be an option in your editor to display the hidden characters or similar. You could open it in a hex editor and verify as well.

Oleg-Perevyshin, 2023-10-04:
Yes, you are right, I deleted 2 characters and everything worked.