espressif/arduino-esp32 #8700: esptool and ESP.getPsramSize() report different psram size for the same board

CelliesProjects, 2023-10-04:
### Board

wemos s2 mini

### Device Description

Nothing just the plain Wemos board.

### Hardware Configuration

Nothing

### Version

v2.0.13

### IDE Name

PlatformIO

### Operating System

Mint Linux

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

Esptool and `ESP.getPsramSize()` give different values for the same board.
 
If I use `esptool.py flash_id` on the board I get:

```bash
esptool.py v4.5.1
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
...
```

So 2MB PSRAM.

If I then run this sketch on the same board:

```bash
#include <Arduino.h>
#include <esp_chip_info.h>

void setup()
{
    delay(3000);
    Serial.setDebugOutput(true);
    // Serial.begin(115200);

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    log_i("This is ESP32 chip with %d CPU cores, WiFi%s%s",
          chip_info.cores,
          (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
          (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    log_i("embedded PSRAM: %s", (chip_info.features & CHIP_FEATURE_EMB_PSRAM) ? "yes" : "no");

    log_i("%iMB psram", ESP.getPsramSize() / (1024 * 1024));

    log_i("silicon revision %d", ESP.getChipRevision());

    log_i("%dMB %s flash", spi_flash_get_chip_size() / (1024 * 1024),
          (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");
}

void loop()
{
    // log_i("test %s", soc_has_cache_lock_bug() ? "bug" : "bugfree");
}
```

I get this output:

```bash
         Connected!
[  3724][I][main.cpp:16] setup(): This is ESP32 chip with 1 CPU cores, WiFi
[  3731][I][main.cpp:18] setup(): embedded PSRAM: no
[  3735][I][main.cpp:20] setup(): 1MB psram
[  3739][I][main.cpp:22] setup(): silicon revision 0
[  3743][I][main.cpp:25] setup(): 4MB external flash
```

### Sketch

```cpp
#include <Arduino.h>
#include <esp_chip_info.h>

void setup()
{
    delay(3000);
    Serial.setDebugOutput(true);
    // Serial.begin(115200);

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    log_i("This is ESP32 chip with %d CPU cores, WiFi%s%s",
          chip_info.cores,
          (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
          (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    log_i("embedded PSRAM: %s", (chip_info.features & CHIP_FEATURE_EMB_PSRAM) ? "yes" : "no");

    log_i("%iMB psram", ESP.getPsramSize() / (1024 * 1024));

    log_i("silicon revision %d", ESP.getChipRevision());

    log_i("%dMB %s flash", spi_flash_get_chip_size() / (1024 * 1024),
          (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");
}

void loop()
{
    // log_i("test %s", soc_has_cache_lock_bug() ? "bug" : "bugfree");
}
```

### Debug Message

```plain
//OUTPUT from esptool:

esptool.py v4.5.1
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
...
```

### Other Steps to Reproduce

Tested only PIO

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

CelliesProjects, 2023-10-04:
You are right.

Stupid computer doing what I say instead of what I mean.

It should be:
`log_i("%.2f MB psram", ESP.getPsramSize() / (1024.0 * 1024));`

Which outputs:
```
         Connected!
[  3724][I][main.cpp:16] setup(): This is ESP32 chip with 1 CPU cores, WiFi
[  3730][I][main.cpp:18] setup(): embedded PSRAM: no
[  3734][I][main.cpp:20] setup(): 2.00 MB psram
[  3739][I][main.cpp:22] setup(): silicon revision 0
[  3743][I][main.cpp:25] setup(): 4MB external flash
```
Thanks again!

closed 2023-10-04 by CelliesProjects

me-no-dev, 2023-10-04:
you did not say how many bytes does `ESP.getPsramSize()` report. It's enough to be one off and the int devision will return 1 instead of 2