espressif/arduino-esp32 #8703: wifi.h, WifiServer  authentification example for website? (Webserver authentification example is not compatible)

dsyleixa, 2023-10-04:
### Related area

ESP32

### Hardware specification

ESP32 Adafruit Feather

### Is your feature request related to a problem?

based on 
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino

I would need an authentification example for wifi.h (not webserver.h)

### Describe the solution you'd like

based on 
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino
I would need an authentification example for wifi.h ! without ! webserver.h
The only authenification I found is unfortunately for only webserver but which is not suitable.

Authenification by WebServer example  
https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/HttpBasicAuth/HttpBasicAuth.ino
is not compatible for Wifi.h / wifiserver (can't figure it out)

my current solution for esp8266Wifi.h
```
if (!authorized) {
      handleNotAuthorized();
      delay(100);
   }

   if (authorized) {
      handleWebsite();
      delay(10);
   }

```
 works fine for esp8266 but fails for esp32, can't port it correctly, none of the HTML websites are displayed,  so a working wifi.h authentification example is needed.

### Describe alternatives you've considered

I tried code which works for esp8266 but that is not compatible: 
using this esp8266 code for esp32 then none of the HTML websites are displayed,  so a working wifi.h authentification example is needed.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-10-05:
It is open source code.  You can see what it is doing.  Nobody owes you an example for something a library does.
https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.cpp#L238-L253

dsyleixa, 2023-10-05:
your example code is from the webserver lib, but as stated:
an example vor wifiserver is needed, NOT webserver!

dsyleixa, 2023-10-05:
your example code is from the WEBSERVER lib, but as stated:
an example vor WIFISERVER is needed, NOT webserver!
Also, I would have no idea what all the letters in the webserver class method mean and do.

dsyleixa, 2023-10-05:
your example code is from the WEBSERVER class lib, but as stated:
an example vor WIFISERVER is needed, suitable for the [SimpleWiFiServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino),  NOT webserver!
Also, I would have no idea what all the letters in the webserver class method mean and do.

dsyleixa, 2023-10-05:
your example code is from the WEBSERVER class lib, but as stated:
an example vor WIFIserver / WIFIclient is needed, suitable for the [SimpleWiFiServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino),  NOT WEBserver!
Also, I would have no idea what all the letters in the webserver class method mean and do at all.

dsyleixa, 2023-10-05:
your example code is from the WEBSERVER class lib, but as stated:
an example vor WIFIserver / WIFIclient is needed, suitable for the [SimpleWiFiServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino),  NOT WEBserver!

Also, I would have no idea what all the letters in the webserver class method mean and do at all.

In addition, the Arduino philosophy lives from the tons of simple, understandable and practical example codes supplied, and finally these are precisely the reasons for the legendary success of Arduino.

dsyleixa, 2023-10-08:
I meanwhile found a Serial  error message when my html website tries to build up - what does that mean?
(html script.length() is  about4000bytes)

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). `

I never get this message by my esp8266!

OTOH, when I cut down a test code to just 1/10 of the original html code size then it runs ok...

closed 2025-01-16 by Parsaabasi