espressif/arduino-esp32 #8704: ESP32S3 no longer recognized after setting up `ARDUINO_USB_CDC_ON_BOOT=1`

nguterresn, 2023-10-04:
### Board

Adafruit ESP32-S3 Feather

### Device Description

Adafruit ESP32-S3 Feather

### Hardware Configuration

No connections yet.

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

macOS 13.5.2

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Esp32s3 is no longer recognized by the system after being flashed once with the following flags:

build_flags = 
	-D ESP32
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1

I had flashed successfully a couple of times before I added the flags. I found an [ticket](https://github.com/espressif/arduino-esp32/discussions/6871#discussioncomment-2955119) about these flags and decided to give it a chance, big mistake.

Also, I found this [issue](https://esp32.com/viewtopic.php?t=28326) which seems related to the one I'm having.

### Sketch

```cpp
build_flags = 
	-D ESP32
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
```

### Debug Message

```plain
There is no debug message.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-10-04:
Not recognized means exactly what?  USB-CDC is going to use a different device, so it may just show up as a different port (and likely needs a hard reset after the first flashing).  Can you test this on Arduino IDE?  It is better about seeing these changes than platformio.

nguterresn, 2023-10-05:
> Not recognized means exactly what?

Not recognised as a serial device.

[image]

The usbmodem is my monitor, not the esp32 board.

nguterresn, 2023-10-05:
> likely needs a hard reset after the first flashing

I've tried to:

1. press BOOT
2. keep pressing BOOT and press RESET
3. release RESET 
4. release BOOT

These are the instructions in the ESP-IDF troubleshooting guide, but it still doesn't work. In fact, I was never able to reset the device, even when I was able to read the serial port.

P-R-O-C-H-Y, 2023-10-05:
@nguterresn I don't own Adafruit ESP32-S3 Feather board, but I tested on MacOS with ESP32-S3 Devkit and after flashing any sketch with `ARDUINO_USB_MODE=1` and `ARDUINO_USB_CDC_ON_BOOT=1` both ports on board (UART + USB) are recognised.

nguterresn, 2023-10-05:
@P-R-O-C-H-Y Thanks for checking.

Ok, so maybe that might not be the issue. I've also tried to flash the spiffs partition before it was no longer recognised, but the platformIO always failed to recognise the board — always had the `Connecting ....` log running.

Could be that some fuses were written while trying to flash the spiffs partition?

nguterresn, 2023-10-07:
> If I upload the code without uploading the filesystem, the device gets grumpy.

Can the ESP32 still recover from it?

According to the adafruit docs, if the board is not recognized, it is still possible to make it recognizable by going into ROM bootloader mode:

> Reset board into ROM bootloader with DFU/BOOT0 + Reset buttons

However, for some reason, my board doesn't reset. I've already check the connectivity of the pins and they are correctly soldered. The BOOT and the RESET button also pull down to ground whenever they are pressed.

Furthermore, I took the time to measure the voltage across the main pins, such the `D+`, the `D-` and the `RESET`.

D+ = 0.4V
D- = 2.5V

The reset pin is read as ~3.3V and slowly decreases to 0V — I think the issue is here. Without pressing the RESET button, it shouldn't pull down.

nguterresn, 2023-10-07:
> If they're really not recognized, you can't reprogram them. Do you just
have a growing pile that quit working and aren't recoverable? There must be
some recovery path that's workign for you if you're saying the documented
one doesn't work.

I decided to unsolder the esp32 and solder manually only the pins I need to make it work.
I kid you not, for a moment it worked — the esp32 went to bootloader mode and the device was once again recognize by my Mac. Not even 10 seconds later, it stopped being recognized again. I didn't touch anything.

[image]

I can tell it is going to bootloader mode since it current draw is only 20mA, whereas in the normal mode is around 180mA. 
I have a couple of standalone esp32s3 SoCs with me, I try to do the same test and maybe replace the faulty one.

nguterresn, 2023-10-07:
Ignore my comment above. I've found the issue.

For the same project, I was using before a normal esp32, not an esp32s3. For some reason I thought it would be just fine to port the same code to the new esp32s3. It was not. Silly me couldn't tell some of the pins weren't the same and they were being incorrectly set up.

I do want to thank your help @robertlipe. Great input and somehow felt like I was not debugging this alone. Thanks!

closed 2023-10-07 by nguterresn