espressif/arduino-esp32 #8705: Backtrace: 0x40084679:0x3ffbf2ec |<-CORRUPTED ESP32

Stroly04, 2023-10-04:
### Board

ESP32 DEVKIT V1

### Device Description

ESP32 DEVKIT V1 

### Hardware Configuration

Sin configuración de GPI/0

### Version

latest master (checkout manually)

### IDE Name

ARDUINO IDE

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am making a control system for a SEIG, for this it is necessary to make the reading of 7 sinusoidal analog signals of 60hz, ideally sampled at about 400us.

For this I am working interruptions.
To test the code I am generating sine signals directly with the ESP32 in the void loop, and in the interruption I sample the signal and make the respective calculations. 

However, when I visualize the calculations in the serial plotter, I see this error:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008467c  PS      : 0x00050031  A0      : 0x800852f2  A1      : 0x3ffbf2ec  
A2      : 0x00000008  A3      : 0x00018008  A4      : 0x000637ff  A5      : 0x3ffbf2cc  
A6      : 0x00000000  A7      : 0x3ffbdbc4  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x000003e8  A11     : 0x80000001  A12     : 0x800811e9  A13     : 0x3ffbf2ac  
A14     : 0x3ffc2158  A15     : 0xfffffbff  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800852fe  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40084679:0x3ffbf2ec |<-CORRUPTED

I am new to ESP32 and I don't know how to approach this problem, I appreciate any idea or correction of the code, I attach the code as well:

### Sketch

```cpp
#define pi 3.1415926535897932384626433832795
float measure_va, measure_vb, measure_vc;
////- Corrientes
volatile float measure_ia, measure_ib, measure_ic, measure_in;
const float voltageConversionFactor = 5.0 / 1023.0;

//- Variables para escala de las señales sensadas
//- Voltajes 
  float va, vb, vc          = 0;
//- Corrientes
  float ia, ib, ic, in      = 0;
// --------------------------
  float sensibilidad        = 0.185;
  float alpha = 0;
  float beta  = 0;
  float zero  = 0;
  float teta = 0;
  float d = 0;
  float q = 0;
hw_timer_t *My_timer = NULL;

void IRAM_ATTR onTimer(){

  va = measure_va*180;
  vb = measure_vb*180;
  vc = measure_vc*180;
// Clack 
  alpha = va*2/3 - vb*1/3 - vc*1/3;
  beta =  (1/(sqrt(3)))*(vb-vc);
  zero = (1/3)*(va+vb+vc);
  teta = atan2(beta,alpha);
// Park
  d =  alpha*cos(teta)+beta*sin(teta);
  q = -alpha*sin(teta)+beta*cos(teta);
}

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
  Serial.println("A,B,C");
  pinMode(LED, OUTPUT);
  My_timer = timerBegin(0, 80, true);
  timerAttachInterrupt(My_timer, &onTimer, true);
  timerAlarmWrite(My_timer, 1000, true);
  timerAlarmEnable(My_timer); //Just Enable
}

// the loop routine runs over and over again forever:
void loop() {

 measure_va = sin(millis()*0.001);  
 measure_vb = sin(millis()*0.001+2*pi/3);   
 measure_vc = sin(millis()*0.001-2*pi/3); 

  Serial.print(va);
  Serial.print(" ");
  Serial.print(vb);
  Serial.print(" ");
  Serial.print(vc);
  Serial.println(" ");

}
```

### Debug Message

```plain
Core  1 register dump:
PC      : 0x4008467c  PS      : 0x00050031  A0      : 0x800852f2  A1      : 0x3ffbf2ec  
A2      : 0x00000008  A3      : 0x00018008  A4      : 0x000637ff  A5      : 0x3ffbf2cc  
A6      : 0x00000000  A7      : 0x3ffbdbc4  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x000003e8  A11     : 0x80000001  A12     : 0x800811e9  A13     : 0x3ffbf2ac  
A14     : 0x3ffc2158  A15     : 0xfffffbff  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800852fe  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40084679:0x3ffbf2ec |<-CORRUPTED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-10-04:
You can decode your backtrace with the [exceptiondecoder](https://github.com/espressif/arduino-esp32#decoding-exceptions).
Using float in an ISR is a bad idea.  You likely don't need an interrupt here anyhow.  Look at [Ticker](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ticker/examples/Blinker/Blinker.ino) which will run in a FreeRTOS stack.

P-R-O-C-H-Y, 2023-10-05:
@Stroly04 can you please decode the backtrace?

Stroly04, 2023-10-05:
> @Stroly04 can you please decode the backtrace?

[image]

these are the results

Stroly04, 2023-10-05:
> You can decode your backtrace with the [exceptiondecoder](https://github.com/espressif/arduino-esp32#decoding-exceptions). Using float in an ISR is a bad idea. You likely don't need an interrupt [...]

I was reading and ticker works in seconds, can it also work for microseconds?

lbernstone, 2023-10-06:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ticker/src/Ticker.h

lbernstone, 2023-10-06:
> > @Stroly04 can you please decode the backtrace?
> 
> [image]
> 
> these are the results

Try putting just the backtrace line in the decoder.  You should get one line that shows where in the code it failed.

fabianoriccardi, 2023-10-14:
I'm experimenting on timers and interrupts, I have seen the same result about exception decoders: if the Backtrace contains 
`|<-CORRUPTED` you don't see any info in the Exception Decoder.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-15 by Parsaabasi

MeY0u, 2025-04-08:
so what is solution for Backtrace: 0x40055448:0x3fca4080 |<-CORRUPTED?
i tried ESP-IDF CMD with no success