espressif/arduino-esp32 #8707: ESP32-S3-PICO-1-N8R8 PSRAM not found - works on N8R2

UnexpectedMaker, 2023-10-05:
### Board

Internal development boards using both PICO-1-N8R2 and N8R8

### Device Description

Trialing both S3-PICO-1 chips on my BLING! boards
https://twitter.com/unexpectedmaker/status/1708986440208773258

### Hardware Configuration

Lots of HW on the board, but not relevant to this issue.

### Version

v2.0.14

### IDE Name

Arduino IDE, PlatformIO

### Operating System

MacOS

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

1155200

### Description

When initialising PSRAM `psramInit()` on the N8R2, it all works fine, but when I call that on the N8R8, I get the following error:
```psram: PSRAM ID read error: 0x00ffffff```

I've checked this with 2.0.12 up to 2.0.14.

### Sketch

```cpp
Not specific to any sketch - happened when any code boots where PSRAM is enabled.
```

### Debug Message

```plain
No debug message, no crash, just fails to initialise and use PSRAM.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-05:
for R8 you need PSRAM=opi

UnexpectedMaker, 2023-10-05:
> for R8 you need PSRAM=opi

For the PICO-1-N8R8 ? I thought `opi` was for octal? This is not octal.

me-no-dev, 2023-10-05:
[image]

UnexpectedMaker, 2023-10-05:
> [image]

Oh wowza, holy moly!!!! Did not read that bit.... thanks mate !!!
I'll go hide in my corner now and pretend this never happened.....

me-no-dev, 2023-10-05:
Rule of thumb: If it's not a custom job (like M5Core-S3) on ESP32-S3 8MB PSRAM is always OPI and 2MB is always QIO