espressif/arduino-esp32 #8733: USBHID GetReport?

OrhanYigitDurmaz, 2023-10-06:
### Related area

USB HID getReport support to get data from the computer

### Hardware specification

mainly ESP32-S3 and S2

### Is your feature request related to a problem?

USBHID library misses one important function: getReport.  With getReport you can get info from the computer without opening any additional serial ports.

### Describe the solution you'd like

one function that will be called as USBHID.getReport .

### Describe alternatives you've considered

TinyUSB supports this, but esp32 port of TinyUSB is bad right now

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-10-07:
it is implemented and it comes as callback. You can not get anything from the host, unless the host gives it to you. USB devices just sit there and answer to the Host, they themselves can not initiate communication.

What is the TinyUSB API that you are missing?

OrhanYigitDurmaz, 2023-10-07:
> it is implemented and it comes as callback. You can not get anything from the host, unless the host gives it to you. USB devices just sit there and answer to the Host, they themselves can not initiate communication.
> 
> What is the TinyUSB API that you are missing?

well, i think TinyUSB is not usable with Arduino Environment, am i wrong?

me-no-dev, 2023-10-07:
I guess you figured out that we do actually use TinyUSB in Arduino :)

tanakamasayuki, 2023-10-10:
https://github.com/tanakamasayuki/EspUsbHost/blob/master/src/EspUsbHost.cpp#L478

This is a sample.
You have to send it yourself using usb_host_transfer_submit_control.
I tried until I got it.

closed 2025-01-15 by Parsaabasi