espressif/arduino-esp32 #8736: WiFi.scanNetworks and WiFi.softAP not working but station mode does

lonerzzz, 2023-10-06:
### Board

ESP32-C3-12F

### Device Description

The module is used on a product PCB.

### Hardware Configuration

None

### Version

v2.0.11

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

Not relevant

### PSRAM enabled

yes

### Upload speed

Not relevant

### Description

We are encountering the following WiFi issues in release 2.0.11.
1) WiFi.scanNetworks() returns no SSIDs
2) WiFi.softAP() and WiFi.softAPConfig() return true, WiFi.getMode() returns access point, but no access point gets set up and is visible for connection.

We can set the SSID directly and connect to a network , but the initial access point setup to configure the device is impacted.

### Sketch

```cpp
We don't have a simple sketch to demonstrate this.
```

### Debug Message

```plain
Not applicable
```

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lonerzzz, 2023-10-07:
I have now included a sketch to reproduce this:

`
#include "Arduino.h"
#include <WiFiClientSecure.h>

const char *AP_SSID = "TEST_SSID";
const char *SSID = "VALID_SSID";
const char HOST_NAME[] = "MYHOST";

IPAddress ipAddress( 0, 0, 0, 0 );
IPAddress AP_IP_ADDRESS( 192, 168, 1, 1 );
IPAddress AP_GATEWAY_IP_ADDRESS( 192, 168, 1, 1 );
IPAddress AP_SUBNET_ADDRESS( 255, 255, 255, 0 );

void setup()
{
	Serial.begin( 115200 );
	Serial.println( "Initializing system" );

	WiFi.setHostname( HOST_NAME );

	// AAA Code below prevents subsequent AP initialization
	WiFi.mode( WIFI_STA );
	WiFi.begin( SSID, "INVALID_PWD" );

	vTaskDelay( 12000 / portTICK_PERIOD_MS );
	WiFi.disconnect( true );
	WiFi.mode( WIFI_OFF ); // Redundant but attempted anyway

	// vTaskDelay( 1000 / portTICK_PERIOD_MS ); // BBB Delay to allow WiFi shutdown

	// If the above code from the comment AAA above to this comment is commented out, then the following works
	WiFi.mode( WIFI_AP );
	int8_t attempts = 10;
	do
	{
		if (!WiFi.softAP( AP_SSID ))
		{
			Serial.println( "Could not setup access point" );
		}
		else
		{
			Serial.println( "Access point enabled" );
		}
		if (!WiFi.softAPConfig( AP_IP_ADDRESS, AP_GATEWAY_IP_ADDRESS, AP_SUBNET_ADDRESS ))
		{
			Serial.println( "Could not configure access point" );
		}
		else
		{
			Serial.println( "Access point configured" );
		}
		ipAddress = WiFi.softAPIP();
		if (ipAddress == AP_IP_ADDRESS)
		{
			break;
		}
		vTaskDelay( 100 / portTICK_PERIOD_MS );
	}
	while (--attempts > 0);
	if (attempts > 0)
	{
		Serial.printf( "Enabled access point with IP %s\r\n", ipAddress.toString().c_str() );
	}
	else
	{
		Serial.printf( "Failed to enable access point with IP %s\r\n", ipAddress.toString().c_str() );
	}

	Serial.println( "System initialized" );
}

void loop()
{

}
`

I believed that the WiFi.disconnect() should check internally for some condition or wait some amount of time before returning because it is not possible to set up the access point right after turning off the WiFi. Some amount of time is required before this works. I am working around now by adding delays as marked at the BBB.

SuGlider, 2023-10-19:
@lonerzzz - Is the issue starting STA mode and then not being able to successfully switch to AP mode?

SuGlider, 2023-10-23:
I have teste the example code (below) and it works (It shows `TEST_SSID` open network as the first listed.
I can scan WiFi and find the ESP32 AP SSID using another ESP32 running the `WiFiScan.ino` example.

#### WiFiScani.ino outpout:

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
E (255) esp_core_dump_f▒f▒▒▒ Core dump data check failed:
Calculated checksum='fa101780'
Image checksum='ffffffff'
Setup done
Scan start
Scan done
20 networks found
Nr | SSID                             | RSSI | CH | Encryption
 1 | TEST_SSID                        |  -38 |  1 | open
 2 | Office                           |  -48 | 12 | WPA2

Scan start
Scan done
20 networks found
Nr | SSID                             | RSSI | CH | Encryption
 1 | TEST_SSID                        |  -36 |  1 | open
 2 | Office                           |  -46 | 12 | WPA2

Scan start
```

> I have now included a sketch to reproduce this:
> 
> ```cpp
> #include "Arduino.h"
> #include <WiFiClientSecure.h>
> 
> const char *AP_SSID = "TEST_SSID";
> const char *SSID = "VALID_SSID";
> const char HOST_NAME[] = "MYHOST";
> 
> IPAddress ipAddress( 0, 0, 0, 0 );
> IPAddress AP_IP_ADDRESS( 192, 168, 1, 1 );
> IPAddress AP_GATEWAY_IP_ADDRESS( 192, 168, 1, 1 );
> IPAddress AP_SUBNET_ADDRESS( 255, 255, 255, 0 );
> 
> void setup()
> {
> 	Serial.begin( 115200 );
> 	Serial.println( "Initializing system" );
> 	WiFi.setHostname( HOST_NAME );
> 
> 	// AAA Code below prevents subsequent AP initialization
> 	WiFi.mode( WIFI_STA );
> 	WiFi.begin( SSID, "INVALID_PWD" );
> 
> 	vTaskDelay( 12000 / portTICK_PERIOD_MS );
> 	WiFi.disconnect( true );
> 	WiFi.mode( WIFI_OFF ); // Redundant but attempted anyway
> 
> 	// vTaskDelay( 1000 / portTICK_PERIOD_MS ); // BBB Delay to allow WiFi shutdown
> 
> 	// If the above code from the comment AAA above to this comment is commented out, then the following works
> 	WiFi.mode( WIFI_AP );
> 	int8_t attempts = 10;
> 	do
> 	{
> 		if (!WiFi.softAP( AP_SSID ))
> 		{
> 			Serial.println( "Could not setup access point" );
> 		}
> 		else
> 		{
> 			Serial.println( "Access point enabled" );
> 		}
> 		if (!WiFi.softAPConfig( AP_IP_ADDRESS, AP_GATEWAY_IP_ADDRESS, AP_SUBNET_ADDRESS ))
> 		{
> 			Serial.println( "Could not configure access point" );
> 		}
> 		else
> 		{
> 			Serial.println( "Access point configured" );
> 		}
> 		ipAddress = WiFi.softAPIP();
> 		if (ipAddress == AP_IP_ADDRESS)
> 		{
> 			break;
> 		}
> 		vTaskDelay( 100 / portTICK_PERIOD_MS );
> 	}
> 	while (--attempts > 0);
> 	if (attempts > 0)
> 	{
> 		Serial.printf( "Enabled access point with IP %s\r\n", ipAddress.toString().c_str() );
> 	}
> 	else
> 	{
> 		Serial.printf( "Failed to enable access point with IP %s\r\n", ipAddress.toString().c_str() );
> 	}
> 
> 	Serial.println( "System initialized" );
> }
> 
> void loop()
> {
> 
> }
> ```
> 
> I believe that the WiFi.disconnect() should check internally for some condition or wait some predefined amount of time before returning because it is not possible to set up the access point right after turning off the WiFi. Some amount of time is required before this works. I am working around now by adding delays as marked at the BBB comment.

closed 2025-01-15 by Parsaabasi