espressif/arduino-esp32 #8737: Arduino 3.0.0 alpha1 missing entries for `gpio_intr_enable` and `gpio_intr_disable`

tonhuisman, 2023-10-07:
### Board

ESP32 any

### Device Description

Any ESP32 build using Arduino 3.0.0 aplha1 framework that uses `gpio_intr_enable` and/or `gpio_intr_disable` entries.

### Hardware Configuration

d/c

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

default

### PSRAM enabled

yes

### Upload speed

115200

### Description

Trying to update the `IRremoteESP8266` library to use Arduino 3.0/IDF 5.1, but this doesn't compile for missing the entries `gpio_intr_enable` and `gpio_intr_disable`.
These entries are still available in the Arduino 2.0.14 release, and I'd expect them to be in the 3.0 release too. Or a properly documented alternative (though that's less desirable).

It is suggested to use `esp_intr_enable` and `esp_intr_disable` as an alternative, but that has a different signature, so not really usable.

### Sketch

```cpp
#if defined(ESP32)
  gpio_intr_disable((gpio_num_t)params.recvpin);
#endif  // ESP32
```

### Debug Message

```plain
src/IRrecv.cpp:434:3: error: 'gpio_intr_disable' was not declared in this scope; did you mean 'esp_intr_disable'?
  434 |   gpio_intr_disable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~~
      |   esp_intr_disable
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-07:
This is ESP-IDF API and as such we do not have control over it. You are welcome to go complain in the ESP-IDF repo or use actual Arduino API (this is the Arduino repo). There is nothing we can/should do. We have however been providing the current Arduino version in a header, so you can filter by that in your lib and change the calls for IDF 5+

closed 2023-10-07 by me-no-dev

tonhuisman, 2023-10-07:
> change the calls for IDF 5+

Well, the issue is that in ESP-IDF v5.1.x [these calls are available](https://github.com/espressif/esp-idf/blob/release/v5.1/components/driver/gpio/include/driver/gpio.h#L111C1-L126C1), but when using this Arduino 3.0.0 alpha1 release, that AFAICS includes ESP-IDF v5.1.1, they aren't available... 🤔

Jason2866, 2023-10-07:
@me-no-dev Do i understand correctly that the support for which was there in Arduino 2.0.x
is dropped in Arduino 3.0.0? As @tonhuisman already said the API is in IDF 5.1.

me-no-dev, 2023-10-07:
If it's in IDF, then it's in Arduino too. We do not control this. The point here is that we do not provide support for ESP-IDF APIs, we provide our own APIs (Arduino) and accept issues about those. The IDF APIs are for ESP-IDF's repo.

me-no-dev, 2023-10-07:
BTW APIs are in 3.0.0-alpha1 https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32/include/driver/gpio/include/driver/gpio.h#L111-L125

Jason2866, 2023-10-07:
Thx for clarifying!

tonhuisman, 2023-10-07:
As Jason pointed out in the IRremoteESP8266 PR, using `#include <driver/gpio.h>` fixes the compilation error. 
Thanks @Jason2866 ! 🦸