espressif/arduino-esp32 #8738: ESP32-S3 can scan wifi but can't connect

vincentj87, 2023-10-07:
### Board

custom board

### Device Description

custom board esp32-s3 based on s3fn8

### Hardware Configuration

[VRN.pdf](https://github.com/espressif/arduino-esp32/files/12838723/VRN.pdf)

### Version

v2.0.13

### IDE Name

Arduino IDE

### Operating System

Windows 10 

### Flash frequency

40mhZ

### PSRAM enabled

yes

### Upload speed

921600

### Description

my board can scan the wifi and get decent result using the wifi scan example. But when tried to connect to the wifi using [Adafuirt wifi test example](https://learn.adafruit.com/adafruit-magtag/wifi-test), it always failed.

### Sketch

```cpp
// SPDX-FileCopyrightText: 2020 Brent Rubell for Adafruit Industries
//
// SPDX-License-Identifier: MIT

/*
//  Web client
//
// This sketch connects to a website (wifitest.adafruit.com/testwifi/index.html)
// using the WiFi module.
//
// This example is written for a network using WPA encryption. For
// WEP or WPA, change the Wifi.begin() call accordingly.
//
// This example is written for a network using WPA encryption. For
// WEP or WPA, change the Wifi.begin() call accordingly.
//
// created 13 July 2010
// by dlf (Metodo2 srl)
// modified 31 May 2012
// by Tom Igoe
// */
#if ARDUINO_USB_MODE
#warning This sketch should be used when USB is in OTG mode
void setup(){}
void loop(){}
#else
#include "USB.h"

#if ARDUINO_USB_CDC_ON_BOOT
#define HWSerial Serial0
#define USBSerial Serial
#else
#define HWSerial Serial
USBCDC USBSerial;
#endif
#include <WiFi.h>

// Enter your WiFi SSID and password
char ssid[] = "homewifi";             // your network SSID (name)
char pass[] = "mypassword";    // your network password (use for WPA, or use as key for WEP)

int keyIndex = 0;                      // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
//IPAddress server(74,125,232,128);  // numeric IP for Google (no DNS)

char server[] = "wifitest.adafruit.com";    // name address for adafruit test
char path[]   = "/testwifi/index.html";

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
WiFiClient client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin();
  Serial.println("STARTING");
 

  // attempt to connect to Wifi network:
  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
//WiFi.useStaticBuffers(true);
//  WiFi.setMinSecurity(WIFI_AUTH_WPA2_PSK); 
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
//WiFi.setTxPower(WIFI_POWER_MINUS_1dBm);
  delay(1000);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
      
  }

  Serial.println("");
  Serial.println("Connected to WiFi");
  printWifiStatus();

  Serial.println("\nStarting connection to server...");
  // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.println("connected to server");
    // Make a HTTP request:
    client.print("GET "); client.print(path); client.println(" HTTP/1.1");
    client.print("Host: "); client.println(server);
    client.println("Connection: close");
    client.println();
  }
}

void loop() {
  // if there are incoming bytes available
  // from the server, read them and print them:
  while (client.available()) {
    char c = client.read();
    Serial.write(c);
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting from server.");
    client.stop();

    // do nothing forevermore:
    while (true) {
      delay(100);
    }
  }
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
#endif
```

### Debug Message

```plain
_eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
_eventCallback(): Reason:2 - AUTH_EXPIRED
_eventCallback(): WiFi AutoReconnect Running
set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW:
..Guru Meditation Error: Core 0 panic'edd (Interupt wdt timeout on CPU0).
```

### Other Steps to Reproduce

i tried prebuild board based on s3 wroom module, using the program it works perfeclty

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

vincentj87, 2023-10-07:
[image]

vincentj87, 2023-10-08:
[Vrn32.zip](https://github.com/espressif/arduino-esp32/files/12840474/Vrn32.zip)
here is the kicad project file

UnexpectedMaker, 2023-10-08:
You need to read up on how to lay out proper RF - your layout is ...  well, it's not usable. This is not a SW thing, it's a HW thing on your design.

Read the S3 HW design guide to learn about the minimum requirements for RF layout for the S3, and then read lots of books or blog posts about how to layout a proper RF solution and all the things you cannot do (you've done most off them) and what the minimum requirements are.

Also read the antenna data sheet to see their required implementation for the antenna to actually work at 2.4Ghz.

Then, once you've re-laid out your RF correctly, get a hold of a VNA and do proper tuning for the LNA pin and the antenna frequency.

Oopas, 2023-10-09:
I have the same problem as you. Is there any solution? I use esp32-s3r8

vincentj87, 2023-10-09:
@Oopas until now i didn't have other solution except fixing the rf layout of my pcb as @UnexpectedMaker said. i can agree more that rf lay out is such a mess.

TD-er, 2023-10-09:
On other boards (ESP8266/ESP32-classic) which did have really badly tuned WiFi antennas, I could get some WiFi connection when reducing the TX power.

So maybe you can try something like that and see if that would result in some connection?
If so, then you have very likely a badly tuned WiFi antenna.

vincentj87, 2023-10-09:
> -lower down the wifi transmit power

@TD-er as i mentioned before i have try that, doesnt seem to fix it

TD-er, 2023-10-09:
Sorry, missed that part, I indeed found you mentioning it.

UnexpectedMaker, 2023-10-09:
Lowering the TX power can't fix it - there is NO TX in your design. It's not bad tuning, it's a completely wrong layout.
You'll have close to 100% reflection, and likely be damaging your ESP32-S3 every time you try to use it.  There is no software fix for something that is 100% a HW issue.

vincentj87, 2023-10-09:
no wonder my esp32-s3 get hot really fast only when using WiFi

closed 2023-10-09 by vincentj87