espressif/arduino-esp32 #8739: When moving from v2.0.11 to v2.0.12 board startup so slow (up to 5s) (also occur on v2.0.13, v2.0.14)

quyen194, 2023-10-08:
### Board

ESP32C3 Dev Module

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.14

### IDE Name

Arduino IDE v2

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When moving from v2.0.11 to v2.0.12, board startup so slow (up to 5s)
I also tested on v2.0.13, v2.0.14 and got same issue.

### Sketch

```cpp
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);

  Serial.println(F("System Initializing ..."));

  Serial.println(F("System Started"));
}
// -----------------------------------------------------------------------------

// the loop function runs over and over again until power down or reset
void loop() {
}
// -----------------------------------------------------------------------------
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
VMDPV_1|1_VMDPV
VMDPR_System Initializing ...
System Started
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-08:
I see this too when the USB-JTAG is used and not connected to a terminal. @SuGlider PTAL

SuGlider, 2023-10-09:
I have used Arduino Core 2.0.14 with Windows IDE 1.8.19.
ESP32-C3 with "USB CDC On Boot: Disabled" --> therefore, Serial is UART0 as the issue here seems to be.
I tried the C3 with USB JTAG connected and also not connected. Same result.

Sketch used for testing:
``` cpp
void setup() {
  Serial.begin(115200);
  Serial.println("STARTED!!");
}

void loop() {
  Serial.println("Looping...");
  delay(3000);
}
```

Arduino IDE Serial Monitor with "Show timestamp" checked:
```
14:52:52.221 -> ESP-ROM:esp32c3-20200918
14:52:52.221 -> Build:Sep 18 2020
14:52:52.221 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
14:52:52.221 -> SPIWP:0xee
14:52:52.221 -> mode:DIO, clock div:1
14:52:52.268 -> load:0x3fcd5820,len:0x458
14:52:52.268 -> load:0x403cc710,len:0x814
14:52:52.268 -> load:0x403ce710,len:0x2848
14:52:52.268 -> entry 0x403cc710
14:52:52.315 -> STARTED!!
14:52:52.315 -> Looping...
14:52:55.335 -> Looping...
14:52:58.305 -> Looping...
14:53:01.324 -> Looping...
```
Boot ends in 14:52:52:268 and the sktech setup prints at 14:52:52:315, after just 35ms.
### I can't reproduce such issue when USB-JTAG is connected or not.

SuGlider, 2023-10-09:
@quyen194 - Please provide me with more information about how to reproduce the issue.
What are these lines in the Serial Ouput?
```
VMDPV_1|1_VMDPV
VMDPR_System 
```

SuGlider, 2023-10-10:
I have tried something else with USB CDC JTAG.
ESP32-C3 with "USB CDC On Boot: Disabled" --> therefore, Serial is USB CDC and Serial0 is UART0.

```cpp
#define UART Serial0
#define CDC  Serial
void setup() {
  CDC.begin();
  UART.begin(115200);

  delay(100);
  UART.println("UART Started.");
  CDC.println("CDC Started.");
}

void loop() {
  UART.println("UART Looping...");
  CDC.println("CDC Looping...");
  delay(3000);
}
```

UART works fine, no delay.
CDC takes a couple seconds to enumerate. In the meantime, the data sent to the CDC is queued.
Right after this time, all the queue is flushed into the CDC and the Serial Monitor gives the impression that is taking some time to start and then spits all the data out.
But this is just the time that Windows takes to enumerate the CDC after uploading a new sketch, for instance.

This is the only way I found to reproduce something similar to the issue. 
Anyway,it is not an Arduino issue, which by the way makes it work fine by enqueuing CDC data.

quyen194, 2023-10-12:
Hi, I tried to cleanup my PC and issue gone.

closed 2023-10-12 by VojtechBartoska