espressif/arduino-esp32 #8740: Failed to resolve component 'mdns' (IDFGH-11204)

piti-diablotin, 2023-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I followed the guide on how to integrate arduino-esp32 in idf.
In the documentation it is mention we can install the component in global component directory so this is what I did.
In `~esp-idf/components` I did
```bash
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive 
```

Then in a fresh new project, when I run `idf.py menuconfig` I expect the command to display menuconfig with arduino option as describe in documentation

### What is the actual behavior?

Instead I got this error when running `idf.py menuconfig`
```
Executing action: menuconfig
Running cmake in directory /home/diablotin/Codes/ESP32/toto/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/diablotin/.espressif/python_env/idf5.1_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/diablotin/Codes/ESP32/toto"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.42.0") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/diablotin/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/diablotin/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/diablotin/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal : ni ceci ni aucun de ses répertoires parents n'est un dépôt git : .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
Updating lock file at /home/diablotin/Codes/ESP32/toto/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.1.1)
CMake Error at /home/diablotin/Codes/esp-idf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'mdns'.
Call Stack (most recent call first):
  /home/diablotin/Codes/esp-idf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /home/diablotin/Codes/esp-idf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  /home/diablotin/Codes/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
```

### Steps to reproduce.

* git clone https://github.com/espressif/esp-idf.git tags/v5.1.1
* `cd  ~esp-idf/components`
*
```bash
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive 
```
* `idf.py create-project toto && cd toto && idf.py menuconfig`

### Build or installation Logs.

```plain
Running on Manjaro
```

### More Information.

I tried to add an extra directory for component but did not change a thing.
I am for sure missing something but cannot figure out what

igrr, 2023-10-09:
Also, we really should have a CI job to verify that arduino-esp32 can be successfully used as a component in an plain IDF project (without the lib-builder).

me-no-dev, 2023-10-09:
> Also, we really should have a CI job to verify that arduino-esp32 can be successfully used as a component in an plain IDF project (without the lib-builder).

We have it on 4.4 and was turned off here a while ago because there was no way to do it at the time (5.1 as master). Will turn it on again and make sure dependencies are properly declared.

closed 2023-10-09 by me-no-dev

piti-diablotin, 2023-10-10:
Thanks for the fast fix !

I still have one question. When I follow procedure, I have an error about FREERTOS frequency being 100 instead of 1000.
I changed it by hand in `sdkconfig` but I was wondering if there is something better to do ?

Otherwise it works fine ! :+1:

me-no-dev, 2023-10-11:
you can have a `sdkconfig.defaults` file in the root of your project with contents `CONFIG_FREERTOS_HZ=1000`. This will ensure that you always have it set to the proper value

Wikiwide, 2026-01-06:
I have an ancient C project using esp-idf. It was running fine with esp-idf v4.4.8, but that's way out of date in 2026 (especially since we are adding more functionality, and I do not want to stumble into some ancient bugs with Bluetooth or WiFi or whatever). So, I tried to update the C code to compile with esp-idf v5.5.2 - and after a few modifications (where I tried to preserve backward compatibility with esp-idf v4.4.8) it does compile with esp-idf v5.5.2, but it does not work properly. So I am trying to compile the code with esp-idf v4.4.8, to see why the malfunction happens. And now it does not compile with esp-idf v4.4.8 (apparently, because of modifications done to make the code compile with esp-idf v5.5.2).

The error I get is (yes, I am now compiling the project inside Docker, because switching back and forth between different versions of esp-idf on the same computer can be a nightmare otherwise):
/project/managed_components/espressif__esp-modbus/modbus/mb_ports/tcp/port_tcp_utils.c: In function 'port_start_mdns_service':
/project/managed_components/espressif__esp-modbus/modbus/mb_ports/tcp/port_tcp_utils.c:581:11: error: implicit declaration of function 'mdns_hostname_get'; did you mean 'mdns_hostname_set'? [-Werror=implicit-function-declaration]
     err = mdns_hostname_get(temp_str);
           ^~~~~~~~~~~~~~~~~
           mdns_hostname_set
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.

esp-modbus pulls in mdns as dependency, and I don't like that. I don't need Modbus TCP, I am only using Modbus serial.

Yes, some people use Modbus TCP, and would probably welcome esp-modbus/modbus/mb_ports/tcp/port_tcp_utils.c.

Either way, it needs to work, or at least, not break the project. One thing I can try is... add mdns as dependency for the project. In CMakeLists.txt first, and if that does not help, then in idf_component.yml (such duplication is inconvenient; what is idf_component.yml designed for, if CMakeLists.txt exists?)

Yes, that comment probably needs to be moved elsewhere, because it's not related to arduino-esp32?