espressif/arduino-esp32 #8741: Memory fail with BLE/WiFi on latest esp update for PlatformIO Arduino

kgobes, 2023-10-09:
### Board

ESP32

### Device Description

ESP32 connecting to Wi-FI/cloud and an edge device via BLE 

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

PlatformIO Arduino

### Operating System

Mac

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I was having inconsistencies with uploading firmware to my esp32, and I realized it's because I was using an older version of esp32 (espressif32@3.2.1). After updating it (latest), I had a memory fail on the same exact code that works before. This code connects to a device via BLE, then Wi-Fi and the cloud, and sends messages. 

At first I was getting issues that said memory allocation failed. I dug around on forums and made some changes to the platformio.ini file so now it looks like the attached code. Now it connects to both ok, but then when it goes to send a message, the device reboots with the debug message "ELF file SHA256".

Can you take a look at my platformio.ini file and let me know what I might be able to adjust to make my code work again?

### Sketch

```cpp
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 2000000
board_build.f_flash = 80000000L
board_build.flash_mode = qio
;build_unflags = -mfix-esp32-psram-cache-issue
build_flags=
    -DBOARD_HAS_PSRAM
    -DCORE_DEBUG_LEVEL=3
board_build.partitions = partitions_16MB.csv
lib_deps = 
	M5Core2@0.1.3
	knolleary/PubSubClient@^2.8
	ArduinoJson@6.19.4
	arduino-mqtt@2.5.1
	lv_arduino@3.0.1 
	toolchain-xtensa32@~2.80400.0
```

### Debug Message

```plain
ELF file SHA256: 88694f10d75b4b6e
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-10-09:
Don't provide toolchains manually.

kgobes, 2023-10-09:
Thank you for the suggestion. With that, I instead get an error that I have gotten sometimes in the past, which I believe indicates a memory fail:

```
send: a5 02 07 50 00 ff 40 00 0b b8 c3 send: a5 02 07 51 00 ff c0 00 0b b8 db send: a5 02 05 53 02 07 00 05 07 send: a5 02 05 55 04 09 00 05 71 send: a5 02 07 56 00 80 00 ff 0b b8 5c send: a5 02 05 58 02 0c 00 03 43 send: a5 02 05 5a 04 0e 00 03 51 
[ 13287][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 13290][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
[ 13497][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 13500][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
```

Jason2866, 2023-10-09:
Probably you have to adjust "something" in your code. That's clearly auth failure.
What may worked in the past, is not guaranteed to work with latest software.

Jason2866, 2023-10-09:
Looks like you are running out of free heap. BLE and TLS needs a lot. You have to find a way to reduce.
This is not an Arduino bug. This github is the wrong place to ask for coding help.

kgobes, 2023-10-11:
Sorry for putting in the wrong github. Please let me know if there is another one you suggest.

In case there is anyone here who may have ideas... 

I've updated the code to run with a Task and am getting irregular disconnecting with this error:
`[ 53079][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)`

and then, at different times, crashes:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

ELF file SHA256: 441d812b0c977c64

E (14290) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

Here is main code, with some simplification to make it easier to read.

```
#include <M5Core2.h>
#include "ScanScreen.hpp"//graphics for scanning for device
#include <NimBLEDevice.h>
#include <ArduinoJson.h>
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <MQTTClient.h>
#include "secrets.h" //certificate file
#include "SetGestureLed.hpp"
#include "Common.hpp"

using namespace abc;

// ------------------------------- Variables --------------------------------

const char *wiFiName = "wifiname"; 
const char *wiFiPassword = "wifipassword";
bool connectToWifiandAWS = true;

bool startDeviceSetup = false;
WiFiClientSecure wifiClient;
MQTTClient mqttClient;
NimBLEClient *bleClient;
static BLEUUID    serviceUUID("01000100-0000-1000-8000-009078563412");
static BLEUUID charUUID("02000200-0000-1000-8000-009178563412");
static BLEUUID  writecharUUID("03000300-0000-1000-8000-009278563412");

// ------------------------------- Setup function ---------------------------
void setup() {
    M5.begin();
    Serial.print("beginning of setup heap: ");
    Serial.println(ESP.getFreeHeap());
    heap_caps_malloc_extmem_enable(64);

    M5.Lcd.fillScreen(WHITE); 
    
    xTaskCreatePinnedToCore( wifi_task, "wifi", 1024*10, NULL, 1, NULL, 1);
    
    setupScanScreen();
    firstDeviceButton.addHandler(resPressed, E_TOUCH + E_RELEASE);
    secondDeviceButton.addHandler(resPressed2, E_TOUCH + E_RELEASE);
    bleClient  = NimBLEDevice::createClient();
    bleClient->setClientCallbacks(new MyClientCallback());
}

void wifi_task(void *pvParameter)
{
    (void)pvParameter;
    while(1)
[... 159 lines cut ...]
monitor_speed = 115200
upload_speed = 2000000
board_build.f_flash = 80000000L
board_build.flash_mode = dio ;qio ***
;build_unflags = -mfix-esp32-psram-cache-issue
build_flags=
    -DBOARD_HAS_PSRAM
    -DCORE_DEBUG_LEVEL=3
    -mfix-esp32-psram-cache-issue
    -mfix-esp32-psram-cache-strategy=memw
    ;-DCONFIG_ARDUINO_ISR_IRAM=0
    ;-DCONFIG_SPIRAM_USE_MALLOC=1
board_build.partitions = partitions_16MB.csv

 
lib_deps =
      M5Core2@0.1.3
      ;knolleary/PubSubClient@^2.8
      ArduinoJson@6.19.4
      arduino-mqtt@2.5.1
      lv_arduino@3.0.1
      h2zero/NimBLE-Arduino@^1.4.0
      ;toolchain-xtensa32@~2.80400.0
	
```

VojtechBartoska, 2023-12-12:
Hi, closing as there is nothing we can do. For help with your project specific code, you can try to ask on Community Arduino ESP32 channel [here](https://app.gitter.im/#/room/#espressif_arduino-esp32:gitter.im)

closed 2023-12-12 by VojtechBartoska