espressif/arduino-esp32 #8742: Ok, about the architecture of generating the bin format

hzhh110, 2023-10-09:
### Related area

nodemcu-32s

### Hardware specification

esp32

### Is your feature request related to a problem?

[env:nodemcu-32s]
platform = espressif32 @ ~3.5.0
board = nodemcu-32s
framework = arduino
Okay, the bin file schema that I generated before is
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x401e8  0x3f400020  0x00000018  DROM
      2  0x04218  0x3ffbdb60  0x00040208  BYTE_ACCESSIBLE, DRAM
      3  0x0bbe0  0x40080000  0x00044428  IRAM
      4  0x114770  0x400d0018  0x00050010  IROM
      5  0x0ac90  0x4008bbe0  0x00164788  IRAM

Now the same configuration generates a really different architecture.

Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x3e834  0x3f400020  0x00000018  DROM
      2  0x017b4  0x3ffbdb60  0x0003e854  BYTE_ACCESSIBLE, DRAM
      3  0x116048  0x400d0018  0x00040010  IROM
      4  0x029d8  0x3ffbf314  0x00156060  BYTE_ACCESSIBLE, DRAM
      5  0x16870  0x40080000  0x00158a40  IRAM

### Describe the solution you'd like

*****Mac-2:~ *****$ esptool.py --chip esp32 image_info -v 2 /Users/*****/Desktop/ESP32-SerialPort-V1.3.6.bin 
esptool.py v4.7.dev1
File size: 1504336 (bytes)

ESP32 image header
==================
Image version: 1
Entry point: 0x40084d3c
Segments: 5
Flash size: 4MB
Flash freq: 40m
Flash mode: DIO

ESP32 extended image header
===========================
WP pin: 0xee (disabled)
Flash pins drive settings: clk_drv: 0x0, q_drv: 0x0, d_drv: 0x0, cs0_drv: 0x0, hd_drv: 0x0, wp_drv: 0x0
Chip ID: 0 (ESP32)
Minimal chip revision: v0.0, (legacy min_rev = 0)
Maximal chip revision: v0.0

Segments information
====================
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x401e8  0x3f400020  0x00000018  DROM
      2  0x04218  0x3ffbdb60  0x00040208  BYTE_ACCESSIBLE, DRAM
      3  0x0bbe0  0x40080000  0x00044428  IRAM
      4  0x114770  0x400d0018  0x00050010  IROM
      5  0x0ac90  0x4008bbe0  0x00164788  IRAM

ESP32 image footer
==================
Checksum: 0xb2 (valid)
Validation hash: 533a1ba309087f1fc3da16e244fc541c3a0a05affe4d81861b146c9ffec729e5 (valid)

Application information
=======================
Project name: esp32-arduino-lib-builder
App version: 9036908-dirty
Compile time: Mar 26 2021 06:09:28
ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
ESP-IDF: v3.3.5-1-g85c43024c
Secure version: 0

*****Mac-2:~ *****$ esptool.py --chip esp32 image_info -v 2 /Users/*****/Downloads/iDoor_P107-3.3.5-V3.0.6.bin 
esptool.py v4.7.dev1
File size: 1503968 (bytes)

ESP32 image header
==================
Image version: 1
Entry point: 0x40084d3c
Segments: 5
Flash size: 4MB
Flash freq: 40m
Flash mode: DIO

ESP32 extended image header
===========================
WP pin: 0xee (disabled)
Flash pins drive settings: clk_drv: 0x0, q_drv: 0x0, d_drv: 0x0, cs0_drv: 0x0, hd_drv: 0x0, wp_drv: 0x0
Chip ID: 0 (ESP32)
Minimal chip revision: v0.0, (legacy min_rev = 0)
Maximal chip revision: v0.0

Segments information
====================
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x3e834  0x3f400020  0x00000018  DROM
      2  0x017b4  0x3ffbdb60  0x0003e854  BYTE_ACCESSIBLE, DRAM
      3  0x116048  0x400d0018  0x00040010  IROM
      4  0x029d8  0x3ffbf314  0x00156060  BYTE_ACCESSIBLE, DRAM
      5  0x16870  0x40080000  0x00158a40  IRAM

ESP32 image footer
==================
Checksum: 0x99 (valid)
Validation hash: ef72bc8b6d59416c663429b5485f3aa4936f6469d1a5d86cf853c02156d9a730 (valid)

Application information
=======================
Project name: esp32-arduino-lib-builder
App version: 9036908-dirty
Compile time: Mar 26 2021 06:09:28
ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
ESP-IDF: v3.3.5-1-g85c43024c
Secure version: 0

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-10-09:
it's not at all clear what versions you used for each test. Please test with ArduinoIDE (as the supported environment) and state the versions of the core you are comparing.

hzhh110, 2023-10-09:
I am using vscode+platformio environment, using
[env:nodemcu-32s]
platform = espressif32@ ~3.5.0
board = nodemcu-32s
framework = arduino

I found that after I uninstalled the Espressif IDF, I changed back
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x0c54c  0x3f400020  0x00000018  DROM
      2  0x022e8  0x3ffbdb60  0x0000c56c  BYTE_ACCESSIBLE, DRAM
      3  0x017ac  0x40080000  0x0000e85c  IRAM
      4  0x19810  0x400d0018  0x00010010  IROM
      5  0x09784  0x400817ac  0x00029828  IRAM

hzhh110, 2023-10-09:
[image]

[image]

The two configurations are the same, but the results are different. Weird

hzhh110, 2023-10-09:
[image]

me-no-dev, 2023-10-09:
In both cases you are using Arduino ESP32 Core v1.0.6 which has not been supported in a long time. You need to use later/latest version (2.0.14)

Jason2866, 2023-10-09:
Try this
```
[env:nodemcu-32s]
platform = espressif32 @ ^6.4.0
platform_packages = framework-arduinoespressif32 @  https://github.com/espressif/arduino-esp32/releases/download/2.0.14/esp32-2.0.14.zip
board = nodemcu-32s
framework = arduino
```

VojtechBartoska, 2023-11-29:
Hello, closing this as expired, if needed, you can reopen.

closed 2023-11-29 by VojtechBartoska