espressif/arduino-esp32 #8745: Failed to execute script esptool

cuvner, 2023-10-09:
### Board

esp32 dev module

### Device Description

esp32 devkit v1

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

arduino

### Operating System

macOs sonoma 14

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Ide wont compile

Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1032, in _main
  File "esptool/__init__.py", line 674, in main
  File "esptool/__init__.py", line 920, in get_default_connected_device
  File "esptool/loader.py", line 302, in __init__
  File "serial/__init__.py", line 90, in serial_for_url
  File "serial/serialposix.py", line 332, in open
  File "serial/serialposix.py", line 517, in _reconfigure_port
termios.error: (22, 'Invalid argument')
[12469] Failed to execute script 'esptool' due to unhandled exception!
esptool.py v4.5.1

### Sketch

```cpp
Blink sketch wont upload
```

### Debug Message

```plain
FQBN: esp32:esp32:alksesp32:UploadSpeed=115200
Using board 'alksesp32' from platform in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14
Using core 'esp32' from platform in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14

esptool.py v4.5.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
python3 /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/gen_esp32part.py -q /private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6/partitions.csv /private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6/sketch_oct9a.ino.partitions.bin
bash -c "[ ! -d \"/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6\"/libraries/Insights ] || python3 \"/Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14\"/tools/gen_insights_package.py /private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6 sketch_oct9a.ino \"/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/.arduinoIDE-unsaved202399-10779-e3figs.eh0me/sketch_oct9a\""
bash -c "[ 0 -eq 0 ] || cp -f \"/Users/chris/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20230419/share/openocd/scripts/\"board/esp32-wrover-kit-3.3v.cfg \"/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/.arduinoIDE-unsaved202399-10779-e3figs.eh0me/sketch_oct9a\"/debug.cfg"
bash -c "[ 0 -eq 0 ] || cp -f \"/Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14\"/tools/ide-debug/esp32.json \"/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/.arduinoIDE-unsaved202399-10779-e3figs.eh0me/sketch_oct9a\"/debug_custom.json"
bash -c "[ 0 -eq 0 ] || cp -f \"/Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14\"/tools/ide-debug/svd/esp32.svd \"/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/.arduinoIDE-unsaved202399-10779-e3figs.eh0me/sketch_oct9a\"/debug.svd"

/Users/chris/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-size -A /private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6/sketch_oct9a.ino.elf
Sketch uses 230865 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20968 bytes (6%) of dynamic memory, leaving 306712 bytes for local variables. Maximum is 327680 bytes.
"/Users/chris/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool" --chip esp32 --port "/dev/cu.usbserial-0001" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 "/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6/sketch_oct9a.ino.bootloader.bin" 0x8000 "/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6/sketch_oct9a.ino.partitions.bin" 0xe000 "/Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/2.0.14/tools/partitions/boot_app0.bin" 0x10000 "/private/var/folders/d_/f1bcbyw57tn6bch8rmml1zfr0000gp/T/arduino/sketches/24C966B1FCBC00A45C2D08619EF7A5A6/sketch_oct9a.ino.bin" 
Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1032, in _main
  File "esptool/__init__.py", line 674, in main
  File "esptool/__init__.py", line 920, in get_default_connected_device
  File "esptool/loader.py", line 302, in __init__
  File "serial/__init__.py", line 90, in serial_for_url
  File "serial/serialposix.py", line 332, in open
  File "serial/serialposix.py", line 517, in _reconfigure_port
termios.error: (22, 'Invalid argument')
[12657] Failed to execute script 'esptool' due to unhandled exception!
esptool.py v4.5.1
Serial port /dev/cu.usbserial-0001
Failed uploading: uploading error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-10-09:
What is your Arduino IDE version number?

cuvner, 2023-10-09:
Turns out it was a cable issue. Im running a mac silicone chip and there has been issues with cable performance. Replaced cable and it worked.

closed 2023-10-09 by cuvner

mrengineer7777, 2023-10-09:
Thank you for the update!