espressif/arduino-esp32 #8746: SDK is missing on 3.0.0-alpha1

orkungedik, 2023-10-09:
### Board

ESP32C313

### Device Description

DevkitC

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hi,

As far as I am not wrong SDK is not exits under \AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha1\tools so that compile is generating error, below;

c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncEventSource.cpp:188:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  188 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp: In constructor 'AsyncWebSocketResponse::AsyncWebSocketResponse(const String&, AsyncWebSocket*)':
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:1262:3: error: 'mbedtls_sha1_starts_ret' was not declared in this scope; did you mean 'mbedtls_sha1_starts'?
 1262 |   mbedtls_sha1_starts_ret(&ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_starts
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:1263:3: error: 'mbedtls_sha1_update_ret' was not declared in this scope; did you mean 'mbedtls_sha1_update'?
 1263 |   mbedtls_sha1_update_ret(&ctx, (const unsigned char*)key.c_str(), key.length());
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_update
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:1264:3: error: 'mbedtls_sha1_finish_ret' was not declared in this scope; did you mean 'mbedtls_sha1_finish'?
 1264 |   mbedtls_sha1_finish_ret(&ctx, hash);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_finish

Best regards,

Orkun Gedik

### Sketch

```cpp
#include "WebAuthentication.h"
#include <libb64/cencode.h>
#ifdef ESP32
#include "mbedtls/md5.h"
#else
#include "md5.h"
#endif

static bool getMD5(uint8_t * data, uint16_t len, char * output){//33 bytes or more
#ifdef ESP32
    mbedtls_md5_context _ctx;
#else
    md5_context_t _ctx;
#endif
  uint8_t i;
  uint8_t * _buf = (uint8_t*)malloc(16);
  if(_buf == NULL)
    return false;
  memset(_buf, 0x00, 16);
#ifdef ESP32
  mbedtls_md5_init(&_ctx);
  mbedtls_md5_starts_ret(&_ctx);
  mbedtls_md5_update_ret(&_ctx, data, len);
  mbedtls_md5_finish_ret(&_ctx, _buf);
#else
  MD5Init(&_ctx);
  MD5Update(&_ctx, data, len);
  MD5Final(_buf, &_ctx);
#endif
  for(i = 0; i < 16; i++) {
    sprintf(output + (i * 2), "%02x", _buf[i]);
  }
  free(_buf);
  return true;
}
```

### Debug Message

```plain
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncEventSource.cpp:188:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  188 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp: In constructor 'AsyncWebSocketResponse::AsyncWebSocketResponse(const String&, AsyncWebSocket*)':
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:1262:3: error: 'mbedtls_sha1_starts_ret' was not declared in this scope; did you mean 'mbedtls_sha1_starts'?
 1262 |   mbedtls_sha1_starts_ret(&ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_starts
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:1263:3: error: 'mbedtls_sha1_update_ret' was not declared in this scope; did you mean 'mbedtls_sha1_update'?
 1263 |   mbedtls_sha1_update_ret(&ctx, (const unsigned char*)key.c_str(), key.length());
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_update
c:\Users\02482899\OneDrive - KocSistem\Documents\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:1264:3: error: 'mbedtls_sha1_finish_ret' was not declared in this scope; did you mean 'mbedtls_sha1_finish'?
 1264 |   mbedtls_sha1_finish_ret(&ctx, hash);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_finish
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-09:
sdk is still there. It is renamed to `esp32-arduino-libs` and is installed as package. Did you run `tools/get.py`?

orkungedik, 2023-10-09:
This is an ArduinoIDE 2.1.1 problem, I guess. Thank you.

closed 2023-10-09 by orkungedik

supersebbo, 2024-03-11:
I am still encountering this issue even after running `python3 get.py` 

Is there a workaround?

kklmn, 2024-04-04:
Hi @orkungedik,
I have exactly same error messages. What was your resolution when you 'completed' the issue?
Thanks!

Marczello00, 2024-04-17:
Have you tried changing functions i the source code in files where errors show up to recommended ones? For me it worked but I am using PlatformIO

khawajamechatronics, 2024-05-06:
Hi
I have exactly same error messages. Using Arduino IDE. 
Thanks!

lbernstone, 2024-05-06:
Arduino 3.0.0 uses a new version of the SDK.  There are breaking changes.  The underlying mbedtls API has changed.

Kv603, 2024-05-07:
Not being willing to wait for a new release of ESP_Async_WebServer leveraging the new mbedtls API, I tried Marczello00's idea, which at least got me a compiled binary.

I ran into some other unrelated incompatibility-related PANIC's, long before my program got anywhere into the crypto libraries.

orkungedik, 2024-07-05:
> Hi @orkungedik, I have exactly same error messages. What was your resolution when you 'completed' the issue? Thanks!

Update ESP32 to 3.0.2

orkungedik, 2024-07-05:
> Hi @orkungedik, I have exactly same error messages. What was your resolution when you 'completed' the issue? Thanks!

Update ESP to 3.0.2