espressif/arduino-esp32 #8749: Uart driver not working with invert signal

dzungpv, 2023-10-10:
### Board

ESP32-PICO-V3-02

### Device Description

It is an custom PCB from the Faikin project here: https://github.com/revk/ESP32-Faikin

### Hardware Configuration

GPIO 26 (tx) and 27 (rx)

### Version

v2.0.14

### IDE Name

Arduino, IDF 4.4.6

### Operating System

Macos 13, Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

The original code of the Faikin author working fine, it wrote in IDF 5.x. I port the code to Arduino, it working it non invert signal, but when using original board, the author invert RX and TX signal it only work for few second then data could not be read.

### Sketch

```cpp
_HardSerial->begin(2400,SERIAL_8E2, rx, tx, true);
```
...
Then I write command to serial port: 
```
  for (int i = 0; i < length; i++)
  {
    _HardSerial->write((uint8_t)packet[i]);
  }
```
...
After that read the signal:

```

  if (_HardSerial->available() > 0)
  {
    // read until we get start byte 0x02
    while (_HardSerial->available() > 0 && !foundStart)
    {
      header[0] = _HardSerial->read();
      if (header[0] == STX)
      {
        foundStart = true;
        delay(100); // found that this delay increases accuracy when reading, might not be needed though
      }
    }
}
```
```

### Debug Message

```plain
I (9745) uart: queue free spaces: 20
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dzungpv, 2023-10-10:
[Esp32_faikin.pdf](https://github.com/espressif/arduino-esp32/files/12854702/Esp32_faikin.pdf)
I update the schematic of the board.

SuGlider, 2023-10-10:
@dzungpv -
When Arduino is called using `_HardSerial->begin(2400,SERIAL_8E2, rx, tx, true);` it will run this IDF code:
```cpp
    if (retCode && inverted) {
        // invert signal for both Rx and Tx
        retCode &= ESP_OK == uart_set_line_inverse(uart_nr, UART_SIGNAL_TXD_INV | UART_SIGNAL_RXD_INV);    
    }
```

The ESP32-Faikin uses this code:
https://github.com/revk/ESP32-Faikin/blob/main/ESP32/main/Faikin.c#L2105-L2113
```cpp
      if (!err)
      {
         uint8_t i = 0;
         if (((rx & PORT_INV) ? 1 : 0) ^ ((proto & PROTO_RXINVERT) ? 1 : 0))
            i |= UART_SIGNAL_RXD_INV;
         if (((tx & PORT_INV) ? 1 : 0) ^ ((proto & PROTO_TXINVERT) ? 1 : 0))
            i |= UART_SIGNAL_TXD_INV;
         err = uart_set_line_inverse (uart, i);
      }

```

Therefore, I'm not sure if the ESP32-Faikin code invert both RX and TX, or only RX, or only TX, or even none...
What I can tell you if that Arduino invert both RX and TX.

Maybe it only inverts one of the signals...
My suggetions is to start Arduino with no inversion and then try all possible combinations using IDF UART code.
Example:
```cpp
#include <driver/uart.h>

void setup() {
   uint8_t rx = 27;
   uint8_t tx = 26;
   uint8_t uart_number = 0;  // for UART0

   Serial.begin((2400,SERIAL_8E2, rx, tx);

   // try RX only inversion
   // if (ESP_OK != uart_set_line_inverse(uart_number, UART_SIGNAL_RXD_INV)) log_e("Error inverting RX");
   // try TX only inversion
   // if (ESP_OK != uart_set_line_inverse(uart_number, UART_SIGNAL_TXD_INV)) log_e("Error inverting TX");
   // try both RX and TX inversion
   // if (ESP_OK != uart_set_line_inverse(uart_number, UART_SIGNAL_TXD_INV | UART_SIGNAL_RXD_INV))  log_e("Error inverting RX and TX");
   // try no inversion
   // if (ESP_OK != uart_set_line_inverse(uart_number, 0)) log_e("Error with no inversion");
}
```

SuGlider, 2023-10-10:
> I port the code to Arduino, it working it non invert signal, but when using original board, the author invert RX and TX signal it only work for few second then data could not be read.

@dzungpv - I have the inpression that it start non-inverting and when the protocol start it may invert it.
Therefore, it is not in the same state all the time... it  may change state depending of the protocol.

I think that the porting can't consider using `begin()` with inversion. It must follow the original code and call `uart_set_line_inverse()` exactly as done there.

dzungpv, 2023-10-11:
Original code invert both rx and tx, I use Hardware serial both invert and it working first 10 seconds after boot(in that time I can control the heat pump, send and receive) after ten seconds it stop working, you can see the schematic, two fets invert the signal. The author also pull up the RX, reset 2 pin, I try to bring original IDF init uart to hal_uart but it not working.  You can try using 2 N mosfet in bread board and reproduce the issue. After 10 seconds all data by Serial.read() is -1. About original code he using negative to define invert signal and change it in a loop, with serial 1.

https://github.com/revk/ESP32-Faikin/blob/main/ESP32/main/Faikin.c#L2105-L2113

dzungpv, 2023-10-11:
@SuGlider I add **delay(1000)** after  Serial.begin and it work, it may not change fast enough with invert settings, so it is working now. You can add a note to document.

closed 2023-10-11 by dzungpv