espressif/arduino-esp32 #8751: Crash Memory with Scan BLE if code in an function (  over Version 1.0.6 )

TiSiAi, 2023-10-10:
### Board

ESP32 DEV MODULE

### Device Description

AZ-Delivery module
ESP VROOM 32 ( standalone )

### Hardware Configuration

nothing

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Mac OS 13.5.2

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello
I use BLE with BLEScan and after my Arduino ( ESP32 ) update, I have a crash memory when the scan BLE function  finish ( when returning main loop )
I the Scan code is not in an function and placed in the main loop, no problem.

I return with 1.0.6 version and no problem

### Sketch

```cpp
//BLE
#include <BLEDevice.h>
#include <BLEScan.h>

#define DEVICE_NAME "BLE_TEST_66"

//Scanner
bool bFirstScanBLE=false;
unsigned long timeLastEventScanBLEDevice = 0;
unsigned long  delayToScanBLEDevice = 1000UL;
int scanTime = 5;//In seconds

BLEScan* pBLEScan;
/*
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};
*/

void setup() {
  
  Serial.begin(115200);
  delay(300);
    
  BLEDevice::init(DEVICE_NAME);

  pBLEScan = BLEDevice::getScan(); //create new scan
  //pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {

  if ( ((millis() - timeLastEventScanBLEDevice) > delayToScanBLEDevice) || (bFirstScanBLE) )
  {
    if(bFirstScanBLE)
    {
      bFirstScanBLE=false;
    }

    Serial.println("Start Scan");
    //Scan in function ScanBLE >> CRASH , if code in the main loop >> no CRASH

    ScanBLE();
    Serial.println("End Scan");

    timeLastEventScanBLEDevice=millis();
  } 

}

int ScanBLE()
{
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   
}
```

### Debug Message

```plain
Start Scan
Devices found: 27
Scan done!
CORRUPT HEAP: Bad head at 0x3ffde35c. Expected 0xabba1234 got 0x3ffc6e6c

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x400835d5:0x3ffc8e30 0x40092e21:0x3ffc8e50 0x4009817d:0x3ffc8e70 0x40097d75:0x3ffc8fa0 0x40083a21:0x3ffc8fc0 0x400981ad:0x3ffc8fe0 0x401797b5:0x3ffc9000 0x40179ac8:0x3ffc9020 0x400d1dc9:0x3ffc9040 0x400d1dbe:0x3ffc9060 0x400d1dbe:0x3ffc9080 0x400d1dbe:0x3ffc90a0 0x400d1dbe:0x3ffc90c0 0x400d1e3d:0x3ffc90e0 0x400d1e83:0x3ffc9120 0x400d6a65:0x3ffc9140

ELF file SHA256: d53adcd9377d2652

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
```

### Other Steps to Reproduce

just call the function ScanBLE since loop

if the code is directly in the main loop ( no function call ) no crash and no problem

if I return with Arduino 1.0.6 it's ok

int ScanBLE()
{
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   
}

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-10-11:
Issue confirmed. It is under investigation.

SuGlider, 2023-10-11:
@TiSiAi -Issue found. It is related to object creation, destruction and memory allocation / releasing.
The way it is done today, creates a problem with destroying the BLEScanResult object twice when it is used with a function, instead of doing it inside `loop()`.

I'll work on a fix and send a PR.

TiSiAi, 2023-10-11:
> Issue found. It is related to object creation, destruction and memory allocation / releasing. The way it is done today, creates a problem with destroying the `BLEScanResult` object twice when it is used with a function, instead of doing it inside `loop()`.
> 
> I'll work on a fix and send a PR.

Thank for your feedback and for your job ;)

SuGlider, 2023-10-12:
@TiSiAi - Please check the change proposed in #8759 - It solves the issue. The PR also has some examples with the necessary code changes in the application.

You can apply it directly to your Arduino BLE library installed in the `Arduino15` folder at `/Users/{username}/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/BLE/src/`, to `BLEScan.cpp` and `BLEScan.h` files.

This PR will be merged for Arduino Core 3.0.0 in a month or so.

closed 2023-10-16 by me-no-dev