espressif/arduino-esp32 #8752: Led flickering when not all attached pins have a load?

arjena, 2023-10-10:
### Board

ESP32

### Device Description

ESP32-WROOM-U

### Hardware Configuration

GPIO 25, 26, 27, 12 and 14 used as ledc outputs

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

MacOS 13

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

This is not really an issue (or maybe it is), it's more something I found and might be helpful to others. Maybe it had been addressed before, but I could not find anything about this. Not in the Espressif docs, nor in any forum.
I have a room with 5 simple one color ledstrips, controlled by one esp32 with ledc. The esp is controlled by MQTT, everything fine.
Now I am about to change the switch sending the MQTT. So, in stead of having no light in the room whilst tinkering, I took an identical ESP32 and put the firmware from the original on it (changed some names of course, to avoid conflict with its twin).
Since I don't have a lot of extra led strips and drivers, I attached an LED to one of the output pins. But whatever I tried, the led flickered like crazy, no matter what value in ledcWrite. Also, sometimes when sending an MQTT message, the ESP32 went into panic and restarted itself. I tried changing the frequency, resolution etc, commented everything not about the led strips out, but the flickering continued. After a couple of hours trying and slowly getting desperate, I had the weird idea of turning all channels off except for the one the led is attached to. Lo and behold, no more flickering. And no more panics, no matter what I throw at it with MQTT. 
So, apparently, for ledc to behave as expected, all attached pins must have a load or everything goes haywire. I read somewhere that with ledc, a pin can be input and output at the same time so that might have something to do with it. 
Anyway, for anyone pulling his or her hair out because ledc does not behave as expected, this might be helpful...

### Sketch

```cpp
Not related
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-10-10:
Sounds a lot more like a power issue.  The esp32 is rated for 42mA on each pin, and 150mA total driven.  At those levels, the cpu will get very hot, and may be unpredictable.  The second device you connected may also have a much weaker SMPS, which can't supply enough juice for a stable WiFi connection. You should use MOSFETs to gate a separate power line to the devices rather than directly driving them.

arjena, 2023-10-10:
I know the ESP32 is not made to drive heavy loads ;). The led was connected through a 1K resistor. So at 3.3V that is nowhere near anything that could cause trouble. And in my search for the cause I also attached a piece of leftover led strip through a driver like I use in my working setup (a module with two MOSFETS). The only way to stop the flickering was to disable all not connected channels. There maybe a logical explanation that has nothing to do with wether the pins are connected to something or not, but these are my findings.

closed 2025-01-15 by Parsaabasi