espressif/arduino-esp32 #8753: Some function not exposed on 3.0.0-alpha

frdoni, 2023-10-10:
### Board

ESP32 DEVKIT V1

### Device Description

Trying new 3.0.0-alpha version, my sketch don't compile because some internal function seems not be exposed anymore:

error: 'esp_random' was not declared in this scope; did you mean 'srandom'?

error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?

error: 'ledcSetup' was not declared in this scope

### Hardware Configuration

no special configuration

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE V2

### Operating System

Win11 x64

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can't compile on 3.0.0-alpha

### Sketch

```cpp
int i = esp_random() % 10;

ledcSetup(GPIO_NUM_18, 100, 8);

ledcAttachPin(GPIO_NUM_18, 0);
```

### Debug Message

```plain
exit status 1

Compilation error: 'esp_random' was not declared in this scope; did you mean 'srandom'?
```

### Other Steps to Reproduce

on 2.0.x version it's compile correctly

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-11:
Functions have changed. Maybe wait for the update guide to come out?

P-R-O-C-H-Y, 2023-10-11:
@frdoni LEDC functions have changed, migration guide and documentation is almost ready.
For `esp_random()` function you need add an include  `#include "esp_random.h"`.

I am closing this as there won't  be any action needed.

closed 2023-10-11 by P-R-O-C-H-Y

atchoo78, 2023-10-12:
> @frdoni LEDC functions have changed, migration guide and documentation is almost ready. 

Where can I find it? I'm getting ledc errors all over the place.

me-no-dev, 2023-10-13:
here is the PR with it being worked on: https://github.com/espressif/arduino-esp32/pull/8716

P-R-O-C-H-Y, 2023-10-13:
@frdoni another option to check [LEDC API docs source file](https://github.com/espressif/arduino-esp32/blob/master/docs/source/api/ledc.rst) or [reworked examples](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/AnalogOut).

atchoo78, 2023-10-13:
Thanks, @P-R-O-C-H-Y. I figured it out by looking at the new [AnalogOut LEDC examples](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/AnalogOut) 👍🏼