espressif/arduino-esp32 #8755: Serial1.setPins not working on ESP32-S3

Erklei, 2023-10-11:
### Board

ESP32-S3

### Device Description

Custom board with ESP32-S3

### Hardware Configuration

custom board, hardware checked and good. 

### Version

v2.0.14

### IDE Name

Arduino IDE 2.2.1

### Operating System

Win10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600 through USB CDC

### Description

Serial1 pins can be assigned using begin(rx,tx etc
But I need to assign also RTS pin.
If I use Serial1.setPins, it looses Rx assignment. TX and RTS assignment are kept correct.

I test it by commenting or uncommenting Serial1.setPins function after Serial1.begin 

### Sketch

```cpp
void setup() {
  Serial.begin(9600);
  Serial1.begin(9600,SERIAL_8N1,13,11,false,500,112);//works, but I cannot assign RTS pin
  //Serial1.setPins(13, 11, -1, 12); //Does not work, but I need to assign RTS pin. 

  //Serial1.setMode(MODE_RS485_HALF_DUPLEX);//I need RTS for RS-485 mode
}

void loop() {
  if (Serial1.available()) {
    int bait = Serial1.read();
    Serial.println(bait);
  }
}
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-10-11:
@Erklei 

please use it this way (set the RTS Pins, set Hardware Flow Control to activate CTS/RTS control, set the RS485 Mode)
``` cpp
void setup() {
  Serial.begin(9600);
  
  Serial1.begin(9600,SERIAL_8N1,13,11,false,500,112);//works, but I cannot assign RTS pin
  if (!Serial1.setPins(-1, -1, -1, 12)) Serial.println("Failed setting RTS pin!"); // just  assigns RTS pin. 
  if (!Serial1.setHwFlowCtrlMode()) Serial.println("Failed changing Flow Control from Software to Hardware RTS!"); // use RTS/CTS instead of xon/xoff
  if (!Serial1.setMode(MODE_RS485_HALF_DUPLEX)) Serial.println("Failed setting RS485 Mode"); //I need RTS for RS-485 mode
}

void loop() {
  if (Serial1.available()) {
    int bait = Serial1.read();
    Serial.println(bait);
  }
}
```

https://github.com/espressif/arduino-esp32/blob/d048e211648c21ef169a14d94028412c320b6d0b/cores/esp32/HardwareSerial.h#L162-L168

Erklei, 2023-10-12:
Hi. I couldn't compile it  with   "if (!Serial1.setHwFlowCtrlMode(UART_HW_FLOWCTRL_RTS, 64)) "
I had to use "HW_FLOWCTRL_CTS_RTS" instead.
Then it compiled and uploaded but returned:

E (4383) uart: uart_set_mode(1729): disable hw flowctrl before using RS485 mode
Failed setting RS485 Mode

I think this setPins issue should be listed as bug.

SuGlider, 2023-10-12:
@Erklei - `setPins()` works as expected. I see no issue.

I tried these 2 examples with the ESP32-S3:

### Example 1 - Uses Serial Monitor to send data to Serial, read it and print it back 
(don't forget to set Serial Monitor to 9600 baud)
``` cpp
void setup() {
  Serial.begin(9600, SERIAL_8N1, -1, -1, false, 500, 112);
  Serial.println("Started.");

  Serial.setPins(-1, -1, -1, 5); // sets RTS Pin only
}

void loop() {
  if (Serial.available() > 0) {
    Serial.print((char)Serial.read());
  }
  delay(50);
}
``` 

### Example 2 - Cross connect Serial1 RX pin (13) with Serial TX pin. It executes by itself.
(don't forget to set  Serial Monitor to 115200 baud to see the results)
```cpp
void setup() {
  Serial.begin(115200, SERIAL_8N1);
  // RX 13 -- TX 11
  Serial1.begin(115200,SERIAL_8N1,13,11,false,500,112);//works, but I cannot assign RTS pin

  // Both of these next 2 lines works fine!
  //Serial1.setPins(13, 11, -1, 12); //Does not work, but I need to assign RTS pin. 
  Serial1.setPins(-1, -1, -1, 12); //Does not work, but I need to assign RTS pin. 
  
  //Serial1.setMode(MODE_RS485_HALF_DUPLEX);//I need RTS for RS-485 mode
  delay(100);
  
  Serial.println("Started.");
  // waits Serial to send all its data
  delay(100);
  
  // a lot of data has been sent to Serial1 RX by Serial TX in boot.
  while (Serial1.available()) Serial1.read(); // just ignores the last sent data
}

void loop() {
  if (Serial1.available()) {
    char bait = Serial1.read() + 1; // next letter A->B
    Serial.println(bait); // sends "B\r\n" to Serial Monitor and Serial1
    delay(100);
    // consumes the 3 chars just sent to Serial1 by cross connection
    bool ERR = false;
    char ch = Serial1.read();
    if (ch != 'B') {
      Serial.println();
      Serial.println((int) ch);
      ERR = true;
    }
    ch = Serial1.read();
    if (ch != '\r') {
      Serial.println();
      Serial.println((int) ch);
      ERR = true;
    }
    ch = Serial1.read();
    if (ch != '\n') {
      Serial.println();
      Serial.println((int) ch);
      ERR = true;
    }
    if (ERR) {
      Serial.println("\n===> ERRROR reading Serial1!");
      while(1); // halts execution.
    }
  }

  // send 'A' to be read by Serial1 in the cross connections TX0->RX1
  Serial.print('A');
  delay(500);
}
```

#### Output of the Example2:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Started.
AB
AB
AB
AB
AB
AB
AB
```

SuGlider, 2023-10-12:
Example 2 works also when `Serial1.setMode(MODE_RS485_HALF_DUPLEX);` is uncommented.

SuGlider, 2023-10-12:
The error because your settings are using the USB CDC, not the UART as `Serial`.

Set "USB CDC On Boot" to "Disable" and then `Serial` will use `HardwareSerial` API instead of `HWCDC` API.

Erklei, 2023-10-13:
Ok, got it working correctly after disabling USB.

In summary:
- No information that different API is used if USB CDC is enabled.
- CTRL+clicking on functions still take me to HardwareSerial.cpp file, so I was assuming it is used.
- It would be helpful if ESP on-board USB could be used because it is more convenient than hooking up jumper wires for separate USB-serial adapter.

closed 2023-10-14 by Erklei