espressif/arduino-esp32 #8756: V2.0.14 compiler warnings

s60sc, 2023-10-11:
### Board

not specific

### Device Description

not specific

### Hardware Configuration

not specific

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

not specific

### PSRAM enabled

yes

### Upload speed

not specific

### Description

```
C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c: In function 'uartSetPins':
C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
      ^~~~~~~~~~~
```
```

C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\LittleFS\src\LittleFS.cpp: In member function 'bool fs::LittleFSFS::begin(bool, const char*, uint8_t, const char*)':
C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\LittleFS\src\LittleFS.cpp:86:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::partition' [-Wmissing-field-initializers]
     };
     ^
C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\LittleFS\src\LittleFS.cpp:86:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::read_only' [-Wmissing-field-initializers]
C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\LittleFS\src\LittleFS.cpp:86:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::grow_on_mount' [-Wmissing-field-initializers]

```

### Sketch

```cpp
not specific
```

### Debug Message

```plain
not specific
```

### Other Steps to Reproduce

not specific

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-10-11:
There is a PR to solve it. Unfortunately, it was not included for 2.0.14, but it is merged within the 2.x branch:
https://github.com/espressif/arduino-esp32/tree/release/v2.x
You may want to apply it in your machine to the file C:\Users\b1sk1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c
https://github.com/espressif/arduino-esp32/pull/8644

closed 2023-10-11 by s60sc

Leonleon33, 2023-10-14:
I apply 8644 to get rid of esp32-hal-uart.c warning.

But I still have the LittleFS.cpp:86:5 warnings
missing initializer for member 'esp_vfs_littlefs_conf_t::partition' , 'esp_vfs_littlefs_conf_t::read_only' and 'esp_vfs_littlefs_conf_t::grow_on_mount'

Is there a way to solve ?