espressif/arduino-esp32 #8765: Compiling problem - Is 3.0.0.alpha2 more fussy about the source code configuration?

dagnall53, 2023-10-14:
### Board

ESP32 dev module 

### Device Description

Problem is Compilation errors seen in Programming using Arduino 2.2.1 
**Using 3.0.0 Alpha2** 
** NOTE- Code was compiling with 2.0.13 and 2.0.14** 

"**Arduino ESP32 Boards 2.0.13** and **M5Stack2.0.7** were simultaneously installed - but removing these "boards" from the Arduino IDE made no difference to the compilation errors.*

Code is for a code complex, but physically simple,  data multiplexer with optocouplers. 
The code is written as multiple (historical)  .ino  to separate "functions".. and not as a single ino with libraries  + / .h/.cpp.
I am concerned that this may be compounding the problems. But rewriting in .cpp/.h format is impractical for now. 

### Hardware Configuration

Optocouplers on inputs - but I think that is irrelevant to this issue. 

### Version

latest master (checkout manually)

### IDE Name

2.2.1 

### Operating System

win10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

**Code that originally compiled now presents (A LOT) of NEW compile errors. 

I can probably sort out what is happening if I can initially solve the two first  compilation errors- 

- First is an invalid conversion - that used to work! - the referenced "OnDataRecv" is a void defined in another ino that is part of the overall sketch..
- Second is a declaration that also used to work..** 

`C:\Users\admin\Documents\Arduino\NMEA3\NMEA3.ino: In function 'void Start_ESP_EXT()':
C:\Users\admin\Documents\Arduino\NMEA3\NMEA3.ino:506:28: error: invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]
  506 |   esp_now_register_recv_cb(OnDataRecv);
      |                            ^~~~~~~~~~
      |                            |
      |                            void (*)(const uint8_t*, const uint8_t*, int) {aka void (*)(const unsigned char*, const unsigned char*, int)}
In file included from C:\Users\admin\Documents\Arduino\NMEA3\NMEA3.ino:74:
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-6b1f40b9bf/esp32/include/esp_wifi/include/esp_now.h:156:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
  156 | esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
      |                                    ~~~~~~~~~~~~~~~~~~^~
C:\Users\admin\Documents\Arduino\NMEA3\NMEA3.ino: In function 'void Self_Test()':
C:\Users\admin\Documents\Arduino\NMEA3\NMEA3.ino:2021:3: error: 'gpio_matrix_out' was not declared in this scope
 2021 |   gpio_matrix_out(TX_PIN_3, 0x100, false, false);
      |   ^~~~~~~~~~~~~~~
`

To me this indicates some change in naming between 2.0.14 and 3.0.0alpha2 ? 
[... 149 lines cut ...]
   45 | void IRAM_ATTR onTimer_0() {
      |                          ^
C:\Users\admin\Documents\Arduino\NMEA3\TX_UART_ST.ino:21:16: note: previous declaration here
   21 | void IRAM_ATTR onTimer_0();        // this obliges the onTimer_0() to be placed on RAM by the compiler
      |                ^~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'void (*)(const uint8_t*, const uint8_t*, int)' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]

      |                                    ~~~~~~~~~~~~~~~~~~^~
```

### Other Steps to Reproduce

Compiles OK with 2.0.13 and 2.0.14. 
But presents these errors with 3.0.0.alpha2  (alpha1 not tested) 
I suspected perhaps that the compiler could not find OnDataRecv, which is defined in a secondary .ino in the same folder,
 So I added 
extern void OnDataRecv(const uint8_t* mac, const uint8_t* incomingData, int len);
before the void Start_ESP_EXT() definition.. but the compile error remains.  

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-10-15:
Arduino core 3.0.0 is a major release change. So there are (breaking) changes.

dagnall53, 2023-10-15:
“Breaking changes”..??  Do you mean the compiler is breaking at the moment in alpha? - Or that the way V3.0.0 needs the program code written is significantly different from V2.x ?? I hope its the former and not the latter! CheersDagnallSent from my iPad

Jason2866, 2023-10-15:
@dagnall53 Not the compiler. Code changes needed since some APIs has changed.

lbernstone, 2023-10-15:
There are breaking changes, and there are likely still bugs where upstream APIs have changed.  This is why it is marked as an alpha release.  There will be a migration document for 3.0.0 in the release.

dagnall53, 2023-10-15:
SO, should I assume that (e.g.) the API concerned with the esp_now_register_recv_cb (and a whole load of ohers?) will need to be updated for V3.xx. ??  _
I will leave that for better programers and just stick to V2 for now!

lbernstone, 2023-10-15:
Depends on whether you want to be a guinea pig.  It all needs testing, and noting that espnow has an issue will help the developers pin that down.  Change the issue title to reflect that so it can get tagged properly.

dagnall53, 2023-10-15:
Thanks.  I’d much rather help than just complain!! I will attempt to re title (and explain!) issues tomorrow! But at least now I understand what is happening!! Sent from my iPhone

dagnall53, 2023-10-16:
@me-no-dev Thank your for looking at my comment!. 
In response to @lbernstone and @Jason2866 , I will note that the Alpha2 version appears to have exposed issues with the following: I will attempt to be more precise in a future Issue post, but will close this now..  
**esp_now.h**,  (esp_now_register_recv_cb possibly defined in a different way to how V3 expects it?)
**gpio.h** ( issue with  void gpio_matrix_in(uint32_t gpio, uint32_t signal_idx, bool inv); ?)
**esp32-hal-timer.h** (Seems to have different definitions of timerBegin , timerAlarmWrite  timerAlarmEnable ?)

closed 2023-10-16 by dagnall53

NLoebel, 2023-12-22:
Latest Arduino Version: 2.2.1 / Date: 2023-08-31T14:35:44.802Z / CLI Version: 0.34.0 updated boards but automatically installed the V3.0.0 alpha2 rather than the stable release V2 - 14.  This thread shows V3 has issues with esp_now.h, - see [dagnall53](https://github.com/dagnall53) who mentioned this issue [on Oct 16](https://github.com/espressif/arduino-esp32/issues/8765#ref-issue-1944865538) (esp_now_register_recv_cb possibly defined in a different way to how V3 expects it?).  

The V3 update broke all code written calling ESP_NOW.h specifically at the esp_now_register_recv_cb parameter point.  About 3,000 lines so cannot provide code.  Went to uninstall V3, but **Arduino Version: 2.2.1 only provided visibility to updates up to 2.0.11, not the latest stable release .14** - code resumed working on .11 anyway so leaving it be.

me-no-dev, 2023-12-22:
@NLoebel the problem is that you did not add the proper board manager json. Check our docs and you will get to see the versions. You are currently installing the core from Arduino.cc which works only with Arduino nano32 board